esp8266/Arduino #414: EEPROM broken

isexton, 2015-06-11:
EEPROM functionality seems to have ceased working in a recent release. Garbage is returned after a power cycle.

me-no-dev, 2015-06-12:
you need to EEPROM.commit() or EEPROM.end() to save the changes after write
have you done an of those?

isexton, 2015-06-12:
Yes, I have an earlier version (Arduino 1.6.1) where it works fine, but it doesn't seem to work with the current release.

tzapu, 2015-06-15:
seems this is happening here as well
https://github.com/tzapu/WiFiManager/issues/1

ikin77, 2015-06-18:
I appreciate all the work you've done with ESP8266 Arduino IDE extension. Up to now, I've been using Arduino Pro Mini + ESP8266 controlled by AT commands. I've stopped using Arduino because now I can do the same, more easy, faster and with less electronics. But unfortunately there's something I'm missing from Arduino: EEPROM is not working at all with ESP8266. Many reboots and errors if I use it in my program, that runs flawless with no EEPROM usage.

Best regards,

me-no-dev, 2015-06-18:
How are you using it? And which version of the IDE? I'm using EEPROM in one of my sketches and have no issue as long as I follow a few steps.
here is an example of how I have the AVR eeprom methods translated:
```cpp
#include "EEPROM.h"
//Call EEPROM.begin(4096); in setup()

void eeprom_read_block(uint8_t * data, uint32_t address, size_t len){
  int i;
  for(i=0; i<len; i++){
    data[i] = EEPROM.read(address+i);
  }
}

void eeprom_write_block(uint8_t * data, uint32_t address, size_t len){
  int i;
  for(i=0; i<len; i++){
    EEPROM.write(address+i, data[i]);
  }
  EEPROM.commit();
}
```

ikin77, 2015-06-19:
Hi,  me-no-dev, and thanks for your reply. I'm using IDE 1.6.4, in Ubuntu. In Arduino I had a class for managing my program settings in EEPROM. The same class, adding "EEPROM.commit" and "EEPROM.begin" causes my program to crash. If I stop using the class in my code, the program works flawless.

============
class ConfigManager {
  
  // Reparto de espacio en EEPRON
  //X(1)-SSID(33)+CHANNEL(1)+PASSWORD(65)+IP(16)+TIMETABLE(4)+HISTORIC_WATER(48)+HISTORIC_OUTDOOR(48)
  public: static const byte EEPROM_WRITED = 'Z';  
  public: static const byte EEPROM_POINTER_SSID             = 1;
  public: static const byte EEPROM_POINTER_CHANNEL          = EEPROM_POINTER_SSID + 33;
  public: static const byte EEPROM_POINTER_PASSWORD         = EEPROM_POINTER_CHANNEL + 1;
  public: static const byte EEPROM_POINTER_IP               = EEPROM_POINTER_PASSWORD + 65;
  public: static const byte EEPROM_POINTER_TIMETABLE        = EEPROM_POINTER_IP + 16;
  public: static const byte EEPROM_POINTER_HISTORIC_WATER   = EEPROM_POINTER_TIMETABLE +  4;  
  public: static const byte EEPROM_POINTER_HISTORIC_OUTDOOR = EEPROM_POINTER_HISTORIC_WATER + 48;  
  
  
  //     
  private: char* ssid     = (char*)malloc(33);
  private: byte  channel;
  private: char* password = (char*)malloc(65);  
  private: char* ip       = (char*)malloc(16); //IP 15 caracteres mas fin str   
  
  
  public: ConfigManager() {

    EEPROM.begin(512);
    
    if(EEPROM.read(0)  == EEPROM_WRITED) {
      
      // Leemos ssid
      for(byte i=0; i+EEPROM_POINTER_SSID < EEPROM_POINTER_CHANNEL; i++) {
        this->ssid[i] = EEPROM.read(i+EEPROM_POINTER_SSID);
      }

      // Leemos channel      
      this->channel = EEPROM.read(EEPROM_POINTER_CHANNEL);

      // Leemos password
      for(byte i=0; i+EEPROM_POINTER_PASSWORD < EEPROM_POINTER_IP; i++) {
        this->password[i] = EEPROM.read(i+EEPROM_POINTER_PASSWORD);
      }

      // Leemos ip
      for(byte i=0; i < 16; i++) { 
        this->ip[i] = EEPROM.read(i+EEPROM_POINTER_IP);
      }
      
    }
    else { // Ponemos todo como cadenas vacias
      this->ssid[0] = 0;
      this->channel = 0;
      this->password[0] = 0;
      this->ip[0] = 0;  
      
      // Ponemos a 0 historico de datos
      for(byte i=0;i<24;i++) {
        this->updateInt(EEPROM_POINTER_HISTORIC_WATER + 2*i, 0); 
        this->updateInt(EEPROM_POINTER_HISTORIC_OUTDOOR + 2*i, 0); 
      }

      // Ponemos horarios por defecto
      this->updateByte(EEPROM_POINTER_TIMETABLE,7);
      this->updateByte(EEPROM_POINTER_TIMETABLE+1,10);
      this->updateByte(EEPROM_POINTER_TIMETABLE+2,20);
      this->updateByte(EEPROM_POINTER_TIMETABLE+3,22);
      
      EEPROM.commit();
    }
    
  };
  
  public: void setWifiSettings(char* ssid, byte channel, char* passwd, char* ip) {
    
    // Guardamos ssid
    byte length = strlen(ssid);
    for(byte i=0;i<=length;i++) {  // Ponemos menor o igual para incorporar el fin de cadena '0'
      this->updateByte(EEPROM_POINTER_SSID + i,ssid[i]);  
    }

    // Guardamos channel    
    this->updateByte(EEPROM_POINTER_CHANNEL,channel);

    // Guardamos passwd
    length = strlen(passwd);
    for(byte i=0;i<=length;i++) {  // Ponemos menor o igual para incorporar el fin de cadena '0'
      this->updateByte(EEPROM_POINTER_PASSWORD + i,passwd[i]);  
    }
    
    // Guardamos IP
    length = strlen(ip);
    for(byte i=0;i<=length;i++) {  // Ponemos menor o igual para incorporar el fin de cadena '0'
      this->updateByte(EEPROM_POINTER_IP + i,ip[i]);  
    }
    
    this->updateByte(0,ConfigManager::EEPROM_WRITED);

    EEPROM.commit();
  }
  
  public: char* getSSID()     {return this->ssid;}  
  public: byte  getChannel()  {return this->channel;}  
  public: char* getPassword() {return this->password;}  
  public: char* getIP()       {return this->ip;}
  
  public: void getTimetable(byte* timetable) {
    timetable[0] = constrain(EEPROM.read(EEPROM_POINTER_TIMETABLE),0,12);
    timetable[1] = constrain(EEPROM.read(EEPROM_POINTER_TIMETABLE+1),timetable[0],12);
    timetable[2] = constrain(EEPROM.read(EEPROM_POINTER_TIMETABLE+2),12,24);
    timetable[3] = constrain(EEPROM.read(EEPROM_POINTER_TIMETABLE+3),timetable[2],24);;
  }

  // Almacenamos en EEPROM horarios de funcionamiento resistencia electrica
  public: void setTimetable(byte timetable[]) {
    for(byte i=0;i<4;i++) {
      this->updateByte(EEPROM_POINTER_TIMETABLE+i,timetable[i]);
    }
    EEPROM.commit();
  }
  
  
  // Almacenamos en EEPROM medidas de temperatura de una hora dada
  public: void setHistoric(byte hour, int waterTemp, int outdoorTemp) {
    this->updateInt(EEPROM_POINTER_HISTORIC_WATER   + 2*hour, waterTemp);
    this->updateInt(EEPROM_POINTER_HISTORIC_OUTDOOR + 2*hour, outdoorTemp);
    EEPROM.commit();
  }

  // Leemos de EEPROM medidas de temperatura de 24 horas
  public: void getHistoric(int waterTemp[], int outdoorTemp[]) {
    
    for(byte i=0;i<24;i++) {
      waterTemp[i] = this->readInt(EEPROM_POINTER_HISTORIC_WATER + 2*i); 
      outdoorTemp[i] = this->readInt(EEPROM_POINTER_HISTORIC_OUTDOOR + 2*i); 
    }
    
  }
  
  
  private: void updateInt(int position, word value) {
    this->updateByte(position, highByte(value));
    this->updateByte(position+1, lowByte(value));
  }

  private: void updateByte(int position, byte value) {
    if(EEPROM.read(position) != value) {
      EEPROM.write(position,value);
    }
  }

  private: int readInt(int position) {  
    return word(EEPROM.read(position),EEPROM.read(position+1));
  }

  
};

============

ikin77, 2015-06-19:
Another question: when you call "EEPROM.commit()", what is written in EEPROM?

1 - all bytes of the in-memory buffer
2 - just bytes modified in in-memory buffer (like EEPROM.write)
3 - bytes located in buffer that differs from actual EEPROM (like EEPROM.update)

Thanks again in advance and best regards

P.S.

  I'm working with ESP-01, yesterday I received my first ESP-12, I'll try EEPROM in it :o/

me-no-dev, 2015-06-19:
At first glance, you are beginning and committing in the class constructor. Try to move this whole block to Class method begin() or init() that you call in the beginning of the sketch's setup(), so you do not touch the EEPROM before the Arduino is ready.

For the second question:
You can write only pages to the flash and you can only erase a whole block, so the EEPROM works like this:
You have up to 4096 bytes of EEPROM (one flash block/sector)
When you call EEPROM.begin(size up to 4k); That block is read and put into memory. From then on unless you call commit or EEPROM.end(); you are always working with that memory block and not the actual content of the flash block. When you call end() or commit(), the memory block is checked for changes and if it has any, the flash block is erased and the memory block is dumped to the flash to save the data for next time.
Flash memory has a particular number of erase/write cycles on its bits (100000 i think), so commit() calls need to be optimized to prevent wearing the flash too soon.

ikin77, 2015-06-19:
Hi again and thanks for the fast reply :) :)

I'm creating my class intance in the setup() method, so, the EEPROM.begin() is always done before any EEPROM
access. Anyway, I'll move EEPROM.begin() to the setup() and I'll let you know.

For the second question, you mean that if I change value in position 7 of the buffer and I do "commit", all block is written. Later, if I change byte in position 8 and do commit again, all block is written again and I have wasted 2 erase/write cycles. Is that ok?

In arduino I was writting in EEPROM once an hour, but changing position (initposition+hour). The erase/write cycles are 100.000 per EEPROM position, in ESP8266 instead this limit is for the entire EEPROM. Is like that? Just to now and adapt my program :o/

Thanks again!!! I love this ESP8266 project, no more Arduinos :oP

me-no-dev, 2015-06-19:
yes, every time you call commit and the data is changed, you will overwrite the whole block and waste a cycle. AVR arduinos have a dedicated EEPROM memory that allows read/write on byte level. Here we do not have such memory and instead we use the onboard flash to store that data (like having a write access to PROGMEM), but the FLASH chip itself does not allow read/write on byte level and only allows what I mentioned above. That is the reason it's done this way :) watch out for too many commits. External EEPROM or FRAM is also an option ;)

ikin77, 2015-06-22:
Hi and thanks for the detailed information. I think I'll stop using EEPROM in ESP8266 in order not to run out of EEPROM cycles in a few weeks :smile:

me-no-dev, 2015-06-22:
Keep in mind :) AVR's EEPROM also has 100K write/erase cycles live :) so if you change the same data, will end up in the same situation

ikin77, 2015-06-23:
I don't think the situation is the same with AVR.

Let's assume this code in ARDUINO:

/////////////////
for(int i=0;i<1024;i++) {
    EEPROM.update(i, instance->getTemperature());
}
///////////////

You can run it 100K times. In ESP8266, you have to modify:

/////////////////
EEPROM.begin(1024);
for(int i=0;i<1024;i++) {
    EEPROM.update(i, instance->getTemperature());
    EEPROM.commit();
}
///////////////

You can run it just 100 times. Of course, it has no sense and you might do:

/////////////////
EEPROM.begin(1024);
for(int i=0;i<1024;i++) {
    EEPROM.update(i, instance->getTemperature());
}
EEPROM.commit();
///////////////

Now, again, you can run it 100K times. But imagine now that you have to write data one position an hour.....

/////////////////
EEPROM.begin(1024);
for(int i=0;i<1024;i++) {
    EEPROM.update(i, instance->getTemperature());
    delay(1000 * 60 * 60); // 1 hour
}
EEPROM.commit();
///////////////

With that code, you are writting to EEPROM each 1024 hours. If you want to have data updated in EEPROM, again, EEPROM.write must go inside the loop:

/////////////////
EEPROM.begin(1024);
for(int i=0;i<1024;i++) {
   EEPROM.update(i, instance->getTemperature());
   EEPROM.commit(); 
   delay(1000 * 60 * 60); // 1 hour
}
///////////////

Again, you just can run it 100 times. That makes 11 years of EEPROM life. In arduino: 

/////////////////
for(int i=0;i<1024;i++) {
   EEPROM.update(i, instance->getTemperature()); 
   delay(1000 * 60 * 60); // 1 hour
}
///////////////

In Arduino, EEPROM lasts for 274 years (enough).

================================

Anyway, thanks for your help!!!!!! :+1: :+1: :+1: :+1: :+1: :+1:

tzapu, 2015-07-15:
hi guys, this has been reported working fine with newer stage releases

closed 2015-07-30 by igrr

Kevin11721, 2016-04-20:
Hi all,

A newbie problem.

I have a sketch. It works great until I add EEPROM.begin(sizeof(EE));  
"EE" is a multi-element typedef struct.  

When I add EEPROM.begin() all the EEPROM reading code works, but the sketch crashes later on, while I'm doing some looping, manipulating the elements in the typedef. The typedef has 15 occurrences EE[1].something EE[2].somethingelse  etc.   I don't think it has anything to do with the code I'm executing at the time of the crash.  I *suspect* the problem is related to the wdtimer. I'm thinking the chip sets itself up for a resets once EPROM.begin() is used. Subsequent and immediate reading of the EEPROM works. Following code works, but by the time I get about 5 seconds into the sketch, it crashes. 

When I comment out the EEPROM.begin() the code runs without crashing but of course all the EEPROM reading fails. 

Is there some known problem and work around regarding the WDtimer and EEPROM operations?

Thanks in advance for your help.

Kevin11721, 2016-04-20:
P.S. I'm using the SparkFun Thing. I'm also reading serial data. I have not only USB power but a Lion battery connected to the "Thing", so there should be plenty of power. I added some caps to the power rails too.