esp8266/Arduino #415: Hardware SPI not working

tripzero, 2015-06-11:
I'm using adafruits dotstar library with a dotstar strip.  Using hardware SPI doesn't seem to work with their example: https://github.com/adafruit/Adafruit_DotStar/blob/master/examples/strandtest/strandtest.ino

To test, change example data/clock pins to 13 and 14.  run.  Bitbanging SPI should work.  A simple volt meter on pin 13 will read 3.xx volts.  To fail, change:

```
Adafruit_DotStar strip = Adafruit_DotStar(
  NUMPIXELS, DATAPIN, CLOCKPIN, DOTSTAR_BRG);
```
to 

```
Adafruit_DotStar strip = Adafruit_DotStar(
  NUMPIXELS, DOTSTAR_BRG);
```

This will tell the library to use the SPI library instead of bitbanging.

iamthemadz, 2015-06-12:
Try this.
Also, you need a logic level shifter on your data and clock. 

GPIO13 = MOSI
GPIO15 = CLK (I am pretty sure. Try dumping the SCK constant to serial to be certain.);

pin definition
https://github.com/esp8266/Arduino/blob/25ae1dba790660198bf2aca18098efde6a3e0410/hardware/esp8266com/esp8266/variants/wifio/pins_arduino.h

```
#include <Adafruit_DotStar.h>
#include <SPI.h>

#define NUMPIXELS 8

#define DATAPIN    MOSI
#define CLOCKPIN   SCK
Adafruit_DotStar strip = Adafruit_DotStar(NUMPIXELS, DATAPIN, CLOCKPIN);

void setup() {
  strip.begin();
  strip.show();
}
int      head  = 0, tail = -10;
uint32_t color = 0xFF0000;

void loop() {

  strip.setPixelColor(head, color);
  strip.setPixelColor(tail, 0);
  strip.show();
  delay(500);

  if(++head >= NUMPIXELS) {
    head = 0;
    if((color >>= 8) == 0)
      color = 0xFF0000;
  }
  if(++tail >= NUMPIXELS) tail = 0;
}
```

tripzero, 2015-06-12:
I'll try SCK.  I'm using the 74AHCT125 for level shifting.

The sparkfun guide says 14 is spi clk.  It may be an error in their documentation: https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/hardware-overview

iamthemadz, 2015-06-12:
I think it is 14. I cant remember. I dont have it hooked up right now. I had it working using those constants though.

tripzero, 2015-06-12:
@iamthemadz did you use this constructor: 
```
Adafruit_DotStar strip = Adafruit_DotStar( NUMPIXELS, DOTSTAR_BRG);
```
If you use the one in your example, it'll use those pins but bitbang instead of using the SPI library.

iamthemadz, 2015-06-12:
Ah, no. 

I only tried the other constructor.

me-no-dev, 2015-06-12:
I think this line in the CPP file of the Lib is your problem:
```
SPI.setClockDivider((F_CPU + 4000000L) / 8000000L); // 8-ish MHz on Due
```

me-no-dev, 2015-06-12:
replace with:
```cpp
#ifdef ESP8266
SPI.setFrequency(8000000L);
#else
SPI.setClockDivider((F_CPU + 4000000L) / 8000000L); // 8-ish MHz on Due
#endif
```

tripzero, 2015-06-15:
The above change seems to do something... but not what I expect.  The lights turn on, but are extremely dim and after all the lights turn on, only the first LED changes colors (the others turn off).

bfmike, 2015-06-20:
Check your power supply.. I had all kinds of problems with a 0.5A supply that I thought would be plenty enough for a couple of LEDs but it wasn't..

me-no-dev, 2015-06-20:
from Adafruit's site: "Estimate up to 60 milliamps peak for each pixel at full brightness white."
it's really best to supply anything external with separate LDO or whatever, in order to get stable power to the ESP. I have also scratched my bold head way too much, just to find that it was power issue.
Exceptions are only some low power sensors. The ESP on it's own is quite hungry :)

closed 2015-07-23 by igrr

tripzero, 2015-07-23:
Is fixed?

igrr, 2015-07-23:
There was no feedback for one month since the last two suggestions, so I
closed this. Feel free to reopen if these suggestions didn't help.

Yona-Appletree, 2015-10-09:
Thanks everyone for this advice. With the above SPI modification to the DotStar library, everything works great! It actually works fine without a level shifter, APA102s are fairly tolerant when it comes to signal voltage, it would seem (and the first LED shifts the voltage for subsequent ones). To recap, I connect CLK to pin 14 and DATA to pin 13, after applying the above patch to AdaFruit_DotStar.