esp8266/Arduino #418: SDWebserver.ino example shows no file names in the treeview for Firefox

sunnyhighway, 2015-06-12:
The SDWebserver.ino example shows no file names in the treeview when shown in Firefox.

In /SdRoot/edit/index.htm the following two functions are responsible for building a tree-node.
 - function createTreeLeaf(path, name, size)
 - function createTreeBranch(path, name, disabled)

The problem lies in this line:
 - label.innerText = name.toLowerCase();

Firefox can't deal with that, and the result is that the name of the file or directory is not shown.

The following replacement fixes that problem:
 - label.textContent = name.toLowerCase();

This fix is tested to work for the most recent version of the following browsers:
 - Firefox
 - Safari
 - Chrome
 - Internet Explorer

me-no-dev, 2015-06-12:
thanks :) will fix it. I wrote it while debugging on WebKit based browsers, so never caught the error.

me-no-dev, 2015-06-12:
https://github.com/esp8266/Arduino/pull/420

closed 2015-07-23 by igrr