esp8266/Arduino #421: Support for NodeMCU R2 boards

Lenbok, 2015-06-12:
I have just got a couple of R2 boards, and uploading to them from the IDE does not work. 

According to http://www.electrodragon.com/w/ESP8266_NodeMCU_Dev_Board#NodeMCU_1.0_.28ESP-12E.29 flashing needs a couple of parameter changes compared to the r1 (SPI SPEED choose 40MHz, SPI MODE select DIO, FLASH SIZE select 32M)

For my lua code, when I flash nodemcu firmware I need to supply esptool.py with -fm dio -fs 32m parameters to get things working. 

If the arduino IDE could add a new board entry that set the equivalent parameters when uploading it would be great!

me-no-dev, 2015-06-13:
it's already in the trunk. Here is the config for you to add it if it's still missing from what you have
```
##############################################################
nodemcuv2.name=NodeMCU 1.0 (ESP-12E Module)
 
nodemcuv2.upload.tool=esptool
nodemcuv2.upload.speed=115200
nodemcuv2.upload.resetmethod=ck
nodemcuv2.upload.maximum_size=1044464
nodemcuv2.upload.maximum_data_size=81920
nodemcuv2.upload.wait_for_upload_port=true
nodemcuv2.serial.disableDTR=true
nodemcuv2.serial.disableRTS=true
 
nodemcuv2.build.mcu=esp8266
nodemcuv2.build.f_cpu=80000000L
nodemcuv2.build.board=ESP8266_ESP12
nodemcuv2.build.core=esp8266
nodemcuv2.build.variant=nodemcu
nodemcuv2.build.flash_mode=dio
nodemcuv2.build.flash_size=4M
nodemcuv2.build.flash_freq=40
nodemcuv2.build.flash_ld=eagle.flash.4m.ld
nodemcuv2.build.spiffs_start=0x100000
nodemcuv2.build.spiffs_end=0x3FB000
nodemcuv2.build.spiffs_pagesize=256
nodemcuv2.build.spiffs_blocksize=8192
 
nodemcuv2.menu.CpuFrequency.80=80 MHz
nodemcuv2.menu.CpuFrequency.80.build.f_cpu=80000000L
nodemcuv2.menu.CpuFrequency.160=160 MHz
nodemcuv2.menu.CpuFrequency.160.build.f_cpu=160000000L
 
nodemcuv2.menu.UploadSpeed.115200=115200
nodemcuv2.menu.UploadSpeed.115200.upload.speed=115200
nodemcuv2.menu.UploadSpeed.9600=9600
nodemcuv2.menu.UploadSpeed.9600.upload.speed=9600
nodemcuv2.menu.UploadSpeed.57600=57600
nodemcuv2.menu.UploadSpeed.57600.upload.speed=57600
nodemcuv2.menu.UploadSpeed.256000.windows=256000
nodemcuv2.menu.UploadSpeed.256000.upload.speed=256000
nodemcuv2.menu.UploadSpeed.230400.linux=230400
nodemcuv2.menu.UploadSpeed.230400.macosx=230400
nodemcuv2.menu.UploadSpeed.230400.macosx=230400
nodemcuv2.menu.UploadSpeed.230400.upload.speed=230400
nodemcuv2.menu.UploadSpeed.460800.linux=460800
nodemcuv2.menu.UploadSpeed.460800.macosx=460800
nodemcuv2.menu.UploadSpeed.460800.upload.speed=460800
nodemcuv2.menu.UploadSpeed.512000.windows=512000
nodemcuv2.menu.UploadSpeed.512000.upload.speed=512000
nodemcuv2.menu.UploadSpeed.921600=921600
nodemcuv2.menu.UploadSpeed.921600.upload.speed=921600
```

Lenbok, 2015-06-14:
Thanks, I have 1.6.4-673-g8cd3697, via the IDE boards manager, I'm not sure how old that is, so will manually paste that text into the boards file I have.

closed 2015-06-14 by Lenbok