esp8266/Arduino #422: Support for ESP-WROOM-02 module ?

nccchirag, 2015-06-13:
Kindly guide if ESP-WROOM-02 module would be supported. The MCU used in the module is ESP8266EX - with Tensilica L106 32-bit micro controller (MCU).

Thank you

Makuna, 2015-06-14:
http://www.esp8266.com/viewtopic.php?f=13&t=2669

nccchirag, 2015-06-23:
WROOM-02 from espressif loads custom firmware .. have tested the same using firmware built with Arduino IDE

closed 2015-06-23 by nccchirag

ritesh26021989, 2015-11-16:
I am a newbie for ESP8266EX and have one ESP-WROOM-02 Wifi Module. I want to start about how to develop that module as Wifi Router for IoT based applications. Please send me link for SDK and also for Hardware Connection to develop it as Wifi Router.

Regards,
Ritesh Prajapati

asetyde, 2015-11-16:
Use google ? SDK , expressif , etc..

drmpf, 2015-11-16:
This projects 
http://www.forward.com.au/pfod/CheapWifiShield/ESP2866_01_WiFi_Shield/index.html
covers hardware/power supply and code and programming for setting up and ESP01 in 'router' mode for configuration, as well as in normal client mode for telent connections.
Should be useful for ESP-WROOM-02 also.

ritesh26021989, 2015-11-20:
Hi, I have connected my ESP-WROOM-02 Module on Breadboard and also connected FTDI based USB 2 UART cable with RX, TX and GND pin as per pin configuration described in ESP-WROOM-02 Module.

After that, I have tried to download AT firmware on my ESP-WROOM-02 (32MBit SPI Flash) module as per desribed in AT firmware document.

I have kept MTDO: 0, GPIO0: 0, GPIO2: 1 to download firmware using Flash Download Tool v2.4.

But, I am getting "chip sync error" after "Connecting..." log while starting to download AT firmware.
It seems like i am not able to access my Wifi Module. correct?

Please let me know if anyone has idea about this type of issue or any clue to solve this type of issue.

Regards,
Ritesh Prajapati

ritesh26021989, 2015-11-20:
Hi,

Now, I am able to flash AT firmware successfully without any issue. The issue was of Hardware Connection.

But, I am not getting any printf statement on UART0. I have just connected UART RX, TX and GND with GPIO0 is set to 1.

Do I need to connect UART0 RTS and UART0 CTS to work with AT commands?

Please let me know as soon as possible.

Regards,
Ritesh Prajapati

ritesh26021989, 2015-11-25:
Hi,

I have successfully flashed AT firmware on my ESP-WROOM-02 firmware and also started board into boot mode without any issue.

I have also executed some AT Instructions set commands on my ESP WIfi module which works fine without any issue.

Now, I want to connect my FTDI based UART with RTS and CTS Pin and I have applied AT+UART_CUR=115200,8,1,0,3 which return OK.

After that, I have set hardware flow control from serial terminal software and reset it. I have also connected RTS and CTS pin from my UART module to ESP Wifi Module.

But, I am not able to execute any AT command after doing above settings.

Please let me know if anyone has idea or clue to solve it.

Regards,
Ritesh Prajapati

petarstefanov, 2015-11-25:
Hi ritesh26021989. 
Can you please elaborate how did you manage to flash the chip, more specifically which pins have to be pulled up/down and with what value resistors. I have been trying to flash the ESP-WROOM-02 module for days now and it's not working. 
Your help is much appreciated.

ritesh26021989, 2015-11-26:
Hi petarstefanov,

Currently, I am using our FTDI chip based USB 2 UART module to flash firmware into ESP-WROOM-02 Module.

I have did following connections to download firmware into 4 MBytes SPI flash of ESP-WROOM-02 Wifi Module.

                1) Pin 1	-	VCC 3.3V (Applied from Vcc Pin of UART Module)
		2) Pin 2 	- 	Chip Enable (Active High)
		3) Pin 6 	- 	MTD 0 Should be Low
		4) Pin 7	-	GPIO2 Should be High
		5) Pin 8	-	GPIO0 Should be low
		6) Pin 9 	- 	GND
		7) Pin 11 	-	UART0 RXD
		8) Pin 12 	-	UART0 TXD

I have used "Flash Download Tool v 2.4" to flash AT firmware v 0.50 binaries as per SPI Flash Size 4 MByets.

Please let me know if you need any more informations from myside.

Regards,
Ritesh Prajapati

phpbbireland, 2015-11-27:
On one of the boards I played with GPIO15 had to be held low in order to program it...
If you have issues it might be worth a try...
Mike

ritesh26021989, 2015-11-28:
Hi Mike,

GPIO15 and GPIO0 should be low to program firmware into flash memory.

I have flashed AT firmware successfully by setting GPIO15, GPIO0 Low and GPIO2 High into ESP-WROOM-02 Wifi Module.

Regards,
Ritesh Prajapati

petarstefanov, 2015-11-28:
Hi Ritesh,
Thanks for your help. I have successfully flashed the chip as per your instructions butnow when I boot it in normal mode I am not getting anything via serial besides the initial line of gibberish. It seems non-responsive, just as before flashing. Any tips on what am I doing wrong? I tried different baud rates.
Take care,
Petar

ritesh26021989, 2015-11-30:
Hi Petar,

Are you flashing AT firmware in your ESP-WROOM-02 Wifi Module?

if yes than you need to do following settings to boot up your firmware from SPI Flash if firmware has been loaded successfully without any issue.

                1) Pin 1	-	VCC 3.3V
		2) Pin 2 	- 	Chip Enable (Active High)
		3) Pin 6 	- 	MTD 0 Should be Low
		4) Pin 7	-	GPIO2 Should be High
		5) Pin 8	-	GPIO0 Should be High
		6) Pin 9 	- 	GND
		7) Pin 11 	-	UART0 RXD
		8) Pin 12 	-	UART0 TXD

After doing above settings, you will get AT command prompt and can access your Wifi Module using different AT Instruction commands.

let me know if you need any help.

Regards,
Ritesh Prajapati

ritesh26021989, 2015-12-02:
Hi Petar,

did you got any success for your problem?

Regards,
Ritesh Prajapati

petarstefanov, 2015-12-02:
Hi Ritesh,
Thank you again for your input. I has helped me a lot and all of what you said is correct and working. I have uploaded AT firmware successfully as well as other code. My next issue, as many others seem to experience as well, is the module is crashing when trying to connect to Wi-Fi. I can perfectly read serial data until it tries to connect to a network, when it transmits a line of gibberish and restarts itself. Any input on this?
Take care,
Petar

ritesh26021989, 2015-12-02:
Hi Petar,

First of all, Which AT firmware version you have flashed into your Wifi Module? Also send me location of each binary you have flashed into your Wifi Module.

Have you executed basic AT commands after updating your WIfi Module?

send me out or AT+GMR command of your WIfi Module. 

Regards,
Ritesh Prajapati

petarstefanov, 2015-12-02:
I flashed v0.9.2.2 AT firmware (one binary at 0x00000) and executed basic AT commands. To command AT it responds OK. To command AT+GMR it responds 0018000902 OK. I just got an idea. I am not using any pullup/pulldown resistors (I just connected the pins to 3V3/GND respectively). Maybe this setup causes the module to crash because of very high current?

ritesh26021989, 2015-12-02:
Hi Petar,

It may be the cause for it. But, I have connected any pullup/pulldown registors in my Wifi Module and its working fine without any crash issue. I have tested my module in AP, STA and both AT+STA mode which works fine without any issue.

BTW, I am using AT firmware 0.50 which is provided by Espressif Syetem. Please try to use that 0.50 AT firmware binaries by flashing firmware in following order.

eagle.flash.bin				::	0x00000
eagle.irom0text.bin			::	0x40000
esp_init_data_default.bin 	::	0x3fc000 
blank.bin           		        ::	0x7e000 & 0x3fe000

Let me know if you have question.

Regards,
Ritesh Prajapati

petarstefanov, 2015-12-02:
Thanks for your advice Ritesh. I will try your suggestions. What value resistors do you use?
Best,
Petar

ritesh26021989, 2015-12-02:
Hi Peter,

Sorry for Typo Error, I haven't connected any resistors with my Wifi Module.

Regards,
Ritesh Prajapati

ritesh26021989, 2015-12-02:
Hi Petar,

Just try with AT firmware 0.50 and let me know your result for crash issue on your Wifi Module.

Regards,
Ritesh Prajapati

petarstefanov, 2015-12-02:
Hi again,
I flashed AT 0.50 and successfully connected to Wi-Fi with AT commands. Another issue I am getting is when I execute AT+RST the module goes into an endless loop of restarting with this output: 
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 23620, room 16 
tail 4
chksum 0x00
load 0x3ffe8000, len 948, room 4 
tail 0
chksum 0x1c
load 0x3ffe83b8, len 3912, room 8 
tail 0
chksum 0x51
csum 0x51
sdÇCÿ
ready
WIFI CONNECTED

ritesh26021989, 2015-12-02:
Please configure your WIfi Mode as AP mode and let me know result.

Regards,
Ritesh Prajapati

ritesh26021989, 2015-12-02:
Please test it with AP alone, STA alone and both STA+AP mode and let me know result of your Wifi Module.

Regards,
Ritesh Prajapati

petarstefanov, 2015-12-02:
For modes 1 and 2 it won't connect to WiFi. In mode 3 (STA+AP) it connects and starts restarting endlessly. 
Regards,
Petar

petarstefanov, 2015-12-02:
It stops restarting when I quit the AP with AT+CWQAP in between restarts. Seems like a power issue.

ritesh26021989, 2015-12-02:
Ok. So, Now you can flash AT firmware and boot it up successfully without any issue. correct?

Just issue is of restarting module in STA+AP mode due to power issue.

Regards,
Ritesh Prajapati

petarstefanov, 2015-12-02:
Correct. More specifically, when trying to connect to an AP. I tried adding resistors with no luck. The module still keeps restarting and reconnecting to Wi-Fi until I execute AP+CWQAP.

ritesh26021989, 2015-12-02:
please use digital CRO on VCC Pin and trigger signal to check whether you are getting any spike when board is going to restart.

If you getting any spike in VCC pin it means that problem is due to power issue otherwise problem is of software from where system_restart API will be called frequently.

Regards,
Ritesh Prajapati

ritesh26021989, 2015-12-03:
Hi Petar,

Anything found for Wifi Module restart issue?

Regards,
Ritesh Prajapati

kmakhil90, 2015-12-03:
Hi,

So i think there will not be any default firmware in WROOM-02 for AT commands execution,right..?

Regards,
Akhil.

ritesh26021989, 2015-12-03:
No, Espressif System Provides AT Firmware Binaries Version 0.50 with Boot Mode and Non-Boot Mode and it is working fine without any issue.

I have flashed those binaries into my ESP-WROOM-02 WIfi Module and I am able to start AT Firmware in boot-mode and executed some AT Commands which works fine without any issue.

I have also tried to compile AT Frimware Demo in NONOS SDK and started that firmware sucessfully as well.

Regards,
Ritesh Prajapati

kmakhil90, 2015-12-03:
Hi,

Thank you Ritesh.
Is it possible flashing on this module using arduino due board?
Do you know about this?

Akhil.

ritesh26021989, 2015-12-03:
Hi Akhil,

ESP-WROOM-02 will come with blank firmware in SPI Flash. So, You have to flash firmware first time to start it.

Regards,
Ritesh Prajapati

ritesh26021989, 2015-12-03:
No. I don't have much idea about Arduino IDE and its tool. But, I have flashed AT firmware using "Flash Download Tool" which is best firmware downlaod tool i think.

So, You can use that software and flash firmware as per 4 MBytes SPI Flash Size provided in ESP-WROOM-02 Module.

Regards,
Ritesh Prajapati

kmakhil90, 2015-12-03:
Yes.
But can i do this flashing using arduino due as a bridge in between module and PC?

Regards,
Akhil.

kmakhil90, 2015-12-03:
Hi,

Ok.Thank you.

Regards,
Akhil.

ritesh26021989, 2015-12-03:
Hi Akhil,

Are you doing Software Development stuffs for ESP-WROOM-02 Module in Linux or Windows Platform?

Are you using Arduino IDE for it? if yes then which version are you usinig?

Regards,
Ritesh Prajapati

kmakhil90, 2015-12-04:
Hi,

I just started learning on this module.
Now i have a Arduino Due board and trying to do something on this as a basic learning.
I am using Arduino IDE 1.6.5 in windows right now and installed Board package for ESP 8266.

Thank you.

Regards,
Akhil.

ritesh26021989, 2015-12-04:
ok.

I am also newbie for ESP8266 Module and have studied how to use AT Basic, WIfi and TCP/UDP IP related commands on ESP-WROOM-02 module. I also understood how to flash firmware into WIfi module and how to boot it from SPI Flash.

I also compiled sample AT firmware demo in NONOS SDK and flashed all bianries as boot-mode and non boot-mode. 

Now, I am looking hot update firmware using OTA update in ESP-WROOM-02 Module.

Regards,
Ritesh Prajapati

kmakhil90, 2015-12-04:
Hi,

Oh. great.
I have to start from basic step itself like flashing AT firmware.
May i contact you for any clarifications through mail?

Regards,
Akhil.

ritesh26021989, 2015-12-04:
Yes. Sure. 

My Email ID:- ritesh.prajapati@slscorp.com

Regards,
Ritesh Prajapati

kmakhil90, 2015-12-04:
Hi,

Thank you very much.

Regards,
Akhil.

deepikavira, 2016-08-04:
@ritesh26021989  How did you resolve the issue of chip sync error? 
I am trying to program the ESP wroom-02 for the very first time, made the connections as follows
VCC->VCC
GND->GND
EN->VCC
GPIO0->GND for firmware upload
GPIO15->GND
GPIO2->VCC
Rx Tx connections to USB-Serial converter.

Am I missing something here? I have worked around with ESP12F chip and never had this esp chip sync error before. Any help would be appreciated. Thanks

ufukv, 2017-07-10:
Hi,
I will design according to MSI internal clock 4.2 MHz. It will be LPUART as peripheral. Does it support Esp-wroom-02 for 115200 baud? Thank you

kmakhil90, 2017-07-12:
@deepikavira try gpio 02 as floating.