esp8266/Arduino #423: OneWire Library - parasitic power error

petrd, 2015-06-13:
I used a modified OneWire library ESP8266, but does not work in a parasitic power mode. Sensor returns only 85 ° C, so a default value.
I tried the same version of the library DUE Arduino and works without any problems.

holgerlembke, 2015-06-13:
Could you describe your hardware setup a little bit?

petrd, 2015-06-13:
I use ModeMCU devkit v0.9 with ESP-12. Sensor DS18B20 is connected to GPIO5. I also tried GPIO4, with the same result. I used the same circuit with Lua firmware and works there parasitic mode without any problems.

holgerlembke, 2015-06-13:
It still could be a timing problem and therefor playing with resistor values could help. So, if you can, lower the pullup value.

petrd, 2015-06-13:
I reduced the resistance from 4k7 to 2k35. The result is 127.94 ° C, which is nonsense.
When 4k7 resistance value is 85 ° C. If the sensor is plugged three wires (VCC, DQ, GND), then works with 2k35 resistor correctly.

duncan-a, 2015-06-13:
2K35??? That's a very strange resistor value...

The DS18B20 is supposed to have Vcc, GND and Data connected.

holgerlembke, 2015-06-13:
DS18B20 works very fine with Data+Gnd only (Vdd must be to Gnd then).

I'm out of idea. Other than lowering even more. :-)

petrd, 2015-06-13:
These are two 4k7 resistors in parallel. DS18B20 has GND, DQ and VDD see http://datasheets.maximintegrated.com/en/ds/DS18B20.pdf

holgerlembke, 2015-06-13:
Just adding some other suggestion: reduce the resolution. That will reduce conversion time and that will reduce power usage.

DS18B20 is 3V minimum, ESP is 3.3V, so right on the edge of everything... :-)

petrd, 2015-06-13:
Wiring is not an error. The same circuit functions with Lua firmware.
The library runs on Arduino DUE OK.
Parasitic only involvement with ESP8266, the error is somewhere in the library definition GPIO, but I can not find it.
Thank you for trying to help me.

duncan-a: These are two 4k7 resistors in parallel. DS18B20 has GND, DQ and VDD see http://datasheets.maximintegrated.com/en/ds/DS18B20.pdf

holgerlembke, 2015-06-13:
If you use the sample code: there is a very small difference between the lua code and the arduino code: reading the scratch register does not turn on power....

  ds.write(0xBE);         // Read Scratchpad 

should better be

  ds.write(0xBE,1);         // Read Scratchpad with p.P.

petrd, 2015-06-13:
It also occurred to me, but it does not affect the outcome. I tried it. Arduino DUE to works with ds.write (0xBE). The problem is that the temperature sensor starts to transfer is not performed successfully ds.write (0x44, 1). Therefore, it is always the value of 85 ° C, it is the default value after startup. If carried out, at least one conversion value will be different

holgerlembke, 2015-06-13:
ok, coming from my bench with results...

4k7: random results, works, doesn't even show up, gives random results.
1k3: superfine. Like a charm.

Perhaps LUA bends resistor values in between time and space?

petrd, 2015-06-13:
Yes, I used the 1k and me it works.
Thank you very much.

closed 2015-06-13 by petrd