esp8266/Arduino #424: Insufficient Pin function description in README

tprochazka, 2015-06-13:
Current section "Pin Functions" in the README.md is a little bit confusing.
You mentioned that most usable pins functions is mapped to macro SPECIAL.
So SPECIAL "will switch that pin in the most usable FUNCTION_X"
But what is X?  Which column or color it is?
And what about other function? How I can enabled them?

Understand correctly that pins 6-11 are used to communicate with flash, so it is not possible to use them? It is not mentioned in the description. Or not?

And what about HW PWM support which should be possible only on pins 12-15

And some explanation of terms like oe, wpu would be also great.

plinioseniore, 2015-06-14:
I agree, what is actually there is a bit confusing.

me-no-dev, 2015-06-14:
SPECIAL maps to:
0 CLK_OUT
1 TX0
2 TX1
3 RX0
4 CLK_XTAL
5 CLK_RTC
12 SPI_MISO
13 SPI_MOSI
14 SPI_CLK
15 SPI_SS

me-no-dev, 2015-06-14:
pins 6-11 are unusable yes
PWM support is on all usable pins
oe and wpu are espessif names and should translate to:
oe - OutputEnable (pin is output)
wpu - Wakeup PullUp (not sure how to explain that one)

tprochazka, 2015-06-14:
Thanks for explanation.

But for example here
http://tech.scargill.net/esp8266-pwm-on-one-output/
is mentioned, that only pins 12-15 are able HW PWM, but you probably mean, that SW emulated PWM is possible on any pin, but it is unusable.

igrr, 2015-06-14:
ESP doesn't have hardware PWM. All the implementations available currently are software-based, to some degree. Our current implementation uses timer1, and the latest one from Espressif uses timer1 + NMI for more accurate timing.

me-no-dev, 2015-06-14:
I'm using the PWM to control a quadcopter's motors and had run the signal through my scope, so it should do well for just about anything.

Namphibian, 2015-06-15:
I have to agree here. I come from an Arduino background so the pin descriptions are way to vague. I am trying to read a simple digital read from pin number 4 but cant get it to change even though my multi-meter says otherwise. Using a ESP-07.

Overall this is a great effort and I managed to switch to the ESP in a matter of hours. Thanks guys!!!

The questions is how would be contribute to this and what is needed.

igrr, 2015-06-15:
@Namphibian you may be facing an issue with GPIO4 and GPIO5 slikscreen labels being flipped on many ESP12s and ESP07s. Try changing pin 4 to 5 and see if that works for you.

Regarding contributions, well, if someone has time to draw one of these awesome [mbed-style](https://developer.mbed.org/media/img/boardlogos/lpc1768/pinout.png) [pin maps](https://developer.mbed.org/media/platforms/nRF51-DK_Pinout_4.png.pagespeed.ce.odTOFQQ-rA.png) that would be very much appreciated. Bonus for doing it in SVG so that it can be easily modified/updated.

czopas, 2015-06-15:
wpu --> means "weak pull-up" :)
A low resistor value is called a strong pull-up (more current flows), a high resistor value is called a weak pull-up (less current flows).

me-no-dev, 2015-06-15:
both PWM implementations are quite different, we use a hardware timer interrupt to control the outputs and different logic to calculate the timer values.
@czopas thanks for the clarification! Makes more sense now :)

closed 2015-07-23 by igrr