esp8266/Arduino #427: I2S Support

4refr0nt, 2015-06-14:
Now, we know how work I2S on ESP8266
https://github.com/espressif/esp8266_mp3_decoder/blob/master/mp3/include/i2s_reg.h

alecapu, 2015-09-28:
I've a test app that use I2S TX and RX in loopback using the ESP8266 SDK...
I'm try to migrate it to Arduino, but I've some problem with interrupt calls. Can some one help me ?

okCancel, 2015-10-14:
I'm interested on use esp to record and play audio. Did you had advances?

torntrousers, 2016-02-07:
I'd really love to have I2S from the ESP/Arduino code to play saved audio/sounds. Here's a naive attempt that makes some buzzes and whistles so shows it compiles and runs, if nothing else maybe it might prompt someone who knows what they're doing to write an example that works ...

(I connect an 8ohm headphone directly to gpio3 (i2s output) and its quiet but easily audible. The mp3 demo says thats probably not advisable but it hasn't broken anything for me.)

```
#include <ESP8266WiFi.h>
#include <i2s.h>
#include <i2s_reg.h>

void setup() {
   Serial.begin(115200); Serial.println();
   Serial.println("setup");

   // don't know if WiFi interupts will interfere with the i2s timing so for now switch it off
   Serial.println("WiFi off");
   WiFi.disconnect(true);
   delay(2000);

   // need to do something like i2sInit like i2s_freertos.c at: https://github.com/espressif/ESP8266_MP3_DECODER/blob/master/mp3/driver/i2s_freertos.c
   // i2sInit(); 
   // init doc in ESP I2S Module Description, looks a bit complicated, see http://bbs.espressif.com/download/file.php?id=658

//   i2c_writeReg_Mask_def(i2c_bbpll, i2c_bbpll_en_audio_clock_out, 1);

}

void loop() {
   Serial.println("loop");
   playIt();
   delay(5000);   
}

void playIt() {
   Serial.println("i2s begin");
   i2s_begin();

   Serial.println("i2s set rate");
 //  i2s_set_rate(22050);
  i2s_set_rate(8000);

   // read PCM samples from array or spiffs ...

   Serial.println("i2s write sample");
   for ( int i=0; i<65536; i++) {
      i2s_write_lr(i, i);   // this gives a steady buzzing
//      int sx = sampToI2sPwm(i); // this gives clicks and high pitch tones
//      i2s_write_lr(sx, sx);   
      if ((i % 1000) == 0) yield(); // without this get WDT resets
   }
   
   Serial.println("i2s end");
   i2s_end();
}

// These from the i2s demo look useful  https://github.com/espressif/ESP8266_MP3_DECODER

//Array with 32-bit values which have one bit more set to '1' in every consecutive array index value
const unsigned int ICACHE_RODATA_ATTR fakePwm[]={ 0x00000010, 0x00000410, 0x00400410, 0x00400C10, 0x00500C10, 0x00D00C10, 0x20D00C10, 0x21D00C10, 0x21D80C10, 
  0xA1D80C10, 0xA1D80D10, 0xA1D80D30, 0xA1DC0D30, 0xA1DC8D30, 0xB1DC8D30, 0xB9DC8D30, 0xB9FC8D30, 0xBDFC8D30, 0xBDFE8D30, 
  0xBDFE8D32, 0xBDFE8D33, 0xBDFECD33, 0xFDFECD33, 0xFDFECD73, 0xFDFEDD73, 0xFFFEDD73, 0xFFFEDD7B, 0xFFFEFD7B, 0xFFFFFD7B, 
  0xFFFFFDFB, 0xFFFFFFFB, 0xFFFFFFFF};

int sampToI2sPwm(short s) {
  //Okay, when this is enabled it means a speaker is connected *directly* to the data output. Instead of
  //having a nice PCM signal, we fake a PWM signal here.
  static int err=0;
  int samp=s;
  samp=(samp+32768);  //to unsigned
  samp-=err;      //Add the error we made when rounding the previous sample (error diffusion)
  //clip value
  if (samp>65535) samp=65535;
  if (samp<0) samp=0;
  //send pwm value for sample value
  samp=fakePwm[samp>>11];
  err=(samp&0x7ff); //Save rounding error.
  return samp;
}

//Reformat the 16-bit mono sample to a format we can send to I2S.
static int sampToI2s(char s) {
  //We can send a 32-bit sample to the I2S subsystem and the DAC will neatly split it up in 2
  //16-bit analog values, one for left and one for right.

  //Duplicate 16-bit sample to both the L and R channel
  int samp=s;
  samp=(samp)&0xffff;
  samp=(samp<<16)|samp;
  return samp;
} 
```

bbx10, 2016-09-14:
http://www.esp8266.com/viewtopic.php?f=11&t=11790

Here is an example for playing WAV files via I2S interface.

devyte, 2017-10-20:
Arduino reference for the class is [here](https://www.arduino.cc/en/Reference/I2S).

Protonerd, 2017-11-10:
Is there an easy way to convert mono wav files to I2S format in run time? The referenced Arduino I2S lib can only play stereo files but most of the files I want to play are mono coded.
Or does I2S have a mono operating mode?

tedhsieh1966, 2018-10-14:
Is there any way to play 20, 24, 32 bits audio?

Piruzzolo, 2019-09-09:
Any news on i2s audio input?

tedhsieh1966, 2019-09-09:
I tried but it did not work.

從 Windows 10 的郵件傳送

寄件者: Piruzzolo
傳送時間: 2019年9月9日 下午 08:06
收件者: esp8266/Arduino
副本: tedhsieh1966; Comment
主旨: Re: [esp8266/Arduino] I2S Support (#427)

Any news on i2s audio input?
—

---
Avast 防毒軟體已檢查此封電子郵件的病毒。
https://www.avast.com/antivirus

earlephilhower, 2019-10-15:
Moving to 3.0.  The pure Arduino I2S setup is immensely different than the current setup.  The support for add'l bitrate/bps is not difficult, but the way that blocks are xfered to the app is quite different than the simple roud-robin async process we have now.

artua, 2019-11-05:
Hi,
@earlephilhower can we use i2s input to get data from PDM mics? Any examples?

Thanks!

earlephilhower, 2019-11-05:
There is an i2sinput example in the repo already.  Take a look at it.  I've run out on an i2s microphone taken from the Google AIY cardboard box project, and it seemed to go just fine.

artua, 2019-11-05:
Yes, but what about PDM microphone?

earlephilhower, 2019-11-05:
Ah, didn't see PDM.  That's not I2S, that's pulse density modulation which is an analog (timing/duty cycle) based protocol as far as I understand it.

I2S is **not** the right way to read them, but I suppose with *a lot* of post-processing you could do it by calculating bit-by-bit density and translating into a 0-N code.  It would be horribly CPU inefficient, though.

artua, 2019-11-05:
Yes, I know. It’s delta-sigma and it’s digital (there is 1 bit data). But
can I read it the same way via i2s input if I clock it out by i2s clock at
1.5mHz(or even 3mHz)? On esp32 there is a special mode for pdm signal, I
didn’t see the same for esp8266...

Thanks for your help

earlephilhower, 2019-11-05:
Yes, you could supersample by setting as high a clock frequency as possible (remember the frequency will be 16-bits * the Fs) then use a popcount (I think there's a GCC internal, if not lots of algorithms are available on DSP processing sites).  Filtering this will also be interesting as you will pick up harmonics due to the processing.

There's no code in the 8266 or in my audio libraries to anything like this, though, you will be on your own.  If you do get something working, feel free to make a PR for the core or examples.

artua, 2019-11-05:
Yes, I tried to set i2s rate to 192000 and it gives me around 3MHz clock
output. Mic is working fine with that.
So if I use your example it will give me 16bit samples joined every cycle?
Or should I somehow get bit by bit from i2s?
I think I can write it to flash (maybe with lower clock first) then
download from flash and to test it with some PC player...

earlephilhower, 2019-11-05:
No, it won't give you anything other than the binary waveform.  No integer values.  You need to find a way to postprocess the entire recording from a bitstream into ints of some precision.  It's complicated to do properly.  And slow since you're processing things bit-by-bit on a moving window.

Real I2S microphones just return a nice stream of 16- or even 32-bit samples every clock period and "just work."  No processing/etc, the ADC is done inside the mike and it returns the real PCM sample each period which you can use, plot, etc.

artua, 2019-11-05:
I have sound processing expert nearby, but he’s not familiar with arduino
nor esp8266. Can you point me the way to get a raw pdm bitstream from my
mic? Which function to use to read the bitstream from i2s buffer?

devyte, 2019-11-06:
@artua I suggest looking at the i2s input example.
Also, use of PDM is off-topic for this issue. If you would like to investigate it and contribute an implementation, please open a new issue, and discussion can proceed there.

artua, 2019-11-06:
Thanks guys,
Seems it simpler for me to use true i2s mic now.

artua, 2020-03-24:
hi, should I post it here? 
https://github.com/esp8266/Arduino/issues/7164

PR #7874 (2021-03-07): Add I2S class support
Fixes #427

Adds a basic I2S class based off of the Arduino-SAMD core.  The raw
i2s_xxx functions are still a better  way to use I2S due to their
flexibility, but this will allow basic Arduino sketches to work.
commit e99df4fe (2021-03-07): Add I2S class support (#7874)

Fixes #427

Adds a basic I2S class based off of the Arduino-SAMD core.  The raw
i2s_xxx functions are still a better  way to use I2S due to their
flexibility, but this will allow basic Arduino sketches to work.
closed 2021-03-07 by earlephilhower