esp8266/Arduino #428: WEB Server reset module randomly

luc-github, 2015-06-15:
Hi I am using staging module and I compile using 1.6.4 under windows 7 64bit
I use ESP8266 01 to be a bridge between 3D printer and a web browser to get printer information, for this I do a request every 2 seconds to refresh the content of the web page and this make the module to reset randomly.

it happens even watchdog is disabled ( between 10min to 8h)
I open 2 web pages at once to make it happen faster
if no pages open the module do not reset at least within 48h ( I did not tested longer)
so currently can be String issue (but did not saw any memory leak) or Webserver issue but I do not know what to test 

Is is a know issue ?
I there anything wrong in the code ?
 or anything that should be added to prevent module reset ?

Thanks in advance

 I narrow down the minimal code using helloserver.ino to reproduce the problem (IP is harcoded to limit the usage of any function)

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
 
const char* ssid = "dlink";
const char* password = "****";//I remove my password

ESP8266WebServer server(80);

void handleRoot() {
  String message = "<HTML><BODY><H1>Page static</H1>";
  message += "<IFRAME  width=\"200\" height=\"200\"  NAME=\"frmstatus\"  ID=\"frmstatus\"  SRC=\"http://192.168.0.115/STATUS\"></IFRAME>\n<SCRIPT TYPE=\"text/javascript\">\n";
  message +="setInterval(function(){";
  message +="var ifrm=document.getElementById(\"frmstatus\");var doc=ifrm.contentDocument?ifrm.contentDocument:ifrm.contentWindow.document;";
  message +="doc.location.reload(true);";
  message +="},2000);\n";
  message +="</SCRIPT>\n</BODY></HTML>";
  server.send(200, "text/html", message);
 
}

void handleSubRoot() {
  static long cnta=0;
  static long cntb=0;
  cnta++;
  if(cnta>100000000)
  {
	  cntb++;
	  cnta=0;
  }
  String message = "<HTML>\n<BODY>\n<H1>refreshed page</H1>";
  message += String(cntb) + "/" + String(cnta);
  message +="</BODY>\n</HTML>\n";
  server.send(200, "text/html", message);
 
}
 
void setup(void){
  ESP.wdtDisable(); //enable or disable reset will happen 
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  
  
  server.on("/", handleRoot);
  server.on("/STATUS", handleSubRoot);
  
  server.begin();
  Serial.println("HTTP server started");
}
 
void loop(void){
  server.handleClient();
} 
```

holgerlembke, 2015-06-15:
Do you monitor your heap()?

luc-github, 2015-06-15:
not on this sketch, but on my project Yes, and was always over 20k when resetting, that is why I wrote  I did not saw saw any memory leak

luc-github, 2015-06-15:
if have another way to monitor - I would be more than happy to check

holgerlembke, 2015-06-15:
Even inside those routines?

To be sure perhaps some Serial.println() into ESP8266WebServer::send and related stuff.

When I used web server first time, holding down F5 in IE killed it instantly, see #230.

luc-github, 2015-06-15:
no I did just before server.send nothing inside the function send itself because it gave me always more than 20000 when it reset, so far from an out of memory

luc-github, 2015-06-15:
Using the test skech I provided - heap is 28880 as a constant

luc-github, 2015-06-15:
actually seems memory is fluctuating after several minutes:
```
28880
28880
28880
28880
28880
28864
28880
28864
28880
28864
28880
28864
28880
28880
28880
28864
28880
```
then before to reset:
```
28880
28880
28880
28880
28880
28864
28880
28880
28880
28512
28904
28864
28880
28880
28496
28904

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)
```

I do not know if it help in some way as this is far from out of memory

holgerlembke, 2015-06-15:
as far as I see, nothing about this causes is really documented...

25k looks more than ok. I'm out of ideas on that problem. Other than as wild guesses like "check power supply", the root of all evil.

luc-github, 2015-06-15:
it happen when connect to PC using USB-Serial adapter and when directly connected to printer on 3.3V 
But if it was power supply this should happen also when web server do not display page - which is not the case if I do not query the web server or only time to time, in this case module does not reset

iamthemadz, 2015-06-15:
You shouldn't power this module with the USB port. Its power consumption is pretty considerable when it is transmitting.

holgerlembke, 2015-06-15:
Who knows? I didn't hook up the power usage to an ozsi (perhaps I should). I assume that the wifi-send-stuff needs most of the energy. And whatever might happen at random moments while two stations talking together, a cat walking throu the beam and then your stuff sucking the remaining 1 mA... But again, who knows? I attach a 1A-supply, some caps and forget it.

[image]

luc-github, 2015-06-15:
Ok I have plugged my ESP01 to a 2A power supply and module is still resetting after 1h of web query

So I guess issue is software not power

sticilface, 2015-06-15:
Do you have AP running?

luc-github, 2015-06-15:
no it is station mode - ESP01 is connected to an AP using DHCP

sticilface, 2015-06-15:
I get a lot of resets, for no reason when running in AP mode.  Still to get to the bottom of it...

sticilface, 2015-06-15:
I'm actually having new issues with the web server.  it just stops responding after around 2 minutes of up time.  I've debugged extensively, tried different IDEs, looked at heap...  The ESP is working fine..  
Serial is outputing debug time info, MQTT commands still work just fine.  but the web server is not responding.  might be related.  but I'm not sure how to debug it further...

holgerlembke, 2015-06-15:
perhaps cause relates to

enum rst_reason {
	DEFAULT_RST_FLAG	= 0,
	WDT_RST_FLAG	= 1,
	EXCEPTION_RST_FLAG    = 2,
	SOFT_RST_FLAG   = 3,
	DEEP_SLEEP_AWAKE_FLAG	= 4
};

luc-github, 2015-06-16:
if I comment ESP.wdtDisable(); and add several ESP.wdtFeed(); I got reset after few minutes
there is no command used for DeepSleep 
```
28904
28880
28880
28880
28880
28864
28880
28496
28904

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

luc-github, 2015-06-16:
So if behavior is inconsistent it looks like memory management issue

igrr, 2015-06-16:
Watchdog reset/disable functions don't do anything because watchdog API has not been released by Espressif. We need to replace the WDT handling with our own code and then expose that in the API but I haven't yet started doing that.

luc-github, 2015-06-16:
Ok thanks - I was relying on Readme 
ESP.wdtEnable(), ESP.wdtDisable(), and ESP.wdtFeed() provide some control over the watchdog timer.
 so it is known issue - I will wait for the fix 

Thanks a lot for your great job

luc-github, 2015-06-16:
Waiting for the WD functions implementation - is there any way to disable Watchdog manually ?
So I can verify my issue is actually a WD issue

igrr, 2015-06-16:
You should not need to disable watchdog unless you are running some long timing-critical operations (bitbanging serial addressable LEDs, for instance).
Watchdog resets just shows you there is a bug in the code somewhere.

luc-github, 2015-06-16:
Well so then you means there is a bug in web server code ? 
I just use the helloserver.ino code to reproduce my issue => refresh page every 2s  and  module reset

igrr, 2015-06-16:
Yes, that's very likely the case here.

luc-github, 2015-06-16:
Anything I can do to help to find the root cause ?

igrr, 2015-06-16:
You can try enabling debug info uncommenting the DEBUGV definition in debug.h (that's in core directory) and adding ```Serial.setDebugOutput(true);``` to your setup method.

luc-github, 2015-06-16:
Ok doing modifications you asked I got this:
28752
:ur 2

:ur 1

WS:dis

WS:ac

:ref 1

WS:av

:ref 2

:ur 2

:rn 420

:ref 2

:wr

:sent 173

:rcla

:abort

:ww

28368
:ur 2

:ur 1

WS:dis

WS:ac

:ref 1

WS:av

:ref 2

:ur 2

:rn 420

:ref 2

:wr

:sent 173

:rcla

:abort

:ww

28776
:ur 2

:ur 1

WS:dis

WS:ac

:ref 1

WS:av

:ref 2

:ur 2

:rn 420

:ref 2

Fatal exception (9):
epc1=0x40101752, epc2=0x00000000, epc3=0x00000000, excvaddr=0xdfac51de, depc=0x00000000

 ets Jan  8 2013,rst cause:1, boot mode:(1,7)

igrr, 2015-06-16:
Could you please upload the HelloServer.cpp.elf file somewhere? 

Alternatively, you can run
```xtensa-lx106-elf-objdump -S HelloServer.cpp.elf > dump.S```
and look up the region of the code around 0x40101752.

luc-github, 2015-06-16:
https://drive.google.com/file/d/0B6Vda5GXPtCYS0tUUHAyMXB4Qms/view?usp=sharing

holgerlembke, 2015-06-16:
just for my personal enjoyment I digged a little bit into the heap thing. I'm not sure if everyone is aware of it, there is a problem called heap fragmentation. And with for example the lax usage of String it could likely happen. Next message has my test program that is happily rebooting on my esp (depends a little bit of what it is doing. It seems if mine lost access point and reconnects, there is not enough heap). And as far as I see, the source is perfectly ok...

It also shows that system_get_free_heap_size() is useless, every malloc() has to be checked.

So there is a systematic problem.

holgerlembke, 2015-06-16:
extern "C" {
#include "user_interface.h"
}

void setup() {
  Serial.begin(115200);
  delay(5000);
  Serial.setDebugOutput(true);

  fragmenttest();
}

void loop() {
 delay(1000);
}

void fragmenttest(void) {
  // to the music of https://www.youtube.com/watch?v=_X_1o3Qw4KM
  const int freemem = 400;

  struct list {
    void *mem;
    struct list *next;
  } *list100, *list101, *hlist100 = NULL, *hlist101 = NULL;

  typedef struct list listitem;

  Serial.println(system_get_free_heap_size());
  while (ESP.getFreeHeap() > freemem)  {
    list100 = (listitem *)malloc(sizeof(listitem));
    list100->mem = malloc(100);
    list100->next = hlist100;
    hlist100 = list100;

    list101 = (listitem *)malloc(sizeof(listitem));
    list101->mem = malloc(101);
    list101->next = hlist101;
    hlist101 = list101;

    Serial.print("+");
  }
  Serial.println();
  // now we have no memory and a nicely filled heap.
  Serial.println(system_get_free_heap_size());

  // and now free one half
  list100 = hlist100;
  while (list100) {
    free(list100->mem);
    list100 = list100->next;
    Serial.print("-");
  }
  Serial.println();
  Serial.println(system_get_free_heap_size());

  // so, do we get 10k?
  void *lala;
  lala = malloc(10000);
  if (lala) {
    Serial.println("Good");
  } else {
    Serial.println("morbus fragmentis");
  }
}

luc-github, 2015-06-16:
ok it was running all night and got several like this:
```
:ur 2
:ur 1
WS:dis
:ref 1
WS:av
:ref 2
:ur 2
:ref 2
:wr
:rn 420
:sent 171
:rcla
:abort
:ww
Fatal exception (28):
epc1=0x40101880, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00558000, depc=0x00000000

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
```
and 

```
:ref 2
:ur 2
:rn 420
:ref 2
Fatal exception (9):
epc1=0x40101752, epc2=0x00000000, epc3=0x00000000, excvaddr=0x4994e1b1, depc=0x00000000

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
```

luc-github, 2015-06-19:
Hi so is this issue linked to heap fragmentation problem

holgerlembke, 2015-06-19:
No, I don't think so.

luc-github, 2015-06-19:
Ok thanks, so no clue then ?

holgerlembke, 2015-06-19:
not me.

luc-github, 2015-06-19:
thank you anyway

holgerlembke, 2015-06-19:
Thing is, my setup() starts with

  pinMode(PinBlinkLED, OUTPUT);
  digitalWrite(PinBlinkLED, LOW);
  delay(100);
  digitalWrite(PinBlinkLED, HIGH);
  delay(200);
  digitalWrite(PinBlinkLED, LOW);
  delay(100);

and sometimes I get a wdt-reset even before I see my led blink.

luc-github, 2015-06-20:
ho so you mean it is not a  webserver issue but more deep in core ?

luc-github, 2015-06-20:
hmmm I just used new staged module and now reset happen very quickly compare to before and so more often

igrr, 2015-06-22:
Latest staging version has stack trace output for exceptions and soft WDT resets. Can you please post the output which happens after crash? I will also need your output .elf file to translate addresses to code — it is located in the temporary folder where your sketch is built. Turn on debug output in Arduino preferences to see the path to this folder.

luc-github, 2015-06-22:
Ok will post result and file when back home - thank you for the help

luc-github, 2015-06-22:
here the file : compiled with IDE 1.6.5 and staging 1.6.4-835-g77d77e8 
https://drive.google.com/file/d/0B6Vda5GXPtCYeEpUV0d5UDJHN1E/view?usp=sharing

got several reset:

```
28048
28024

Exception (28):
epc1=0x401018bc epc2=0x00000000 epc3=0x00000000 excvaddr=0x000016d0 depc=0x00000000

ctx: cont 
sp: 3ffead50 end: 3ffeb0b0 offset: 01a0

>>>stack>>>
3ffeaef0:  40207898 3ffeaf84 00000073 00000073  
3ffeaf00:  40207898 00000001 3ffeaf84 402078c6  
3ffeaf10:  40100200 00000001 3ffeaf84 40207c8d  
3ffeaf20:  40100218 00000003 3ffeaf84 3ffe8ef4  
3ffeaf30:  3ffeafb0 3ffe9d8c 3ffeaf84 40207ced  
3ffeaf40:  3ffe8ef4 3ffe9d8c 3ffe9510 40207cc2  
3ffeaf50:  3ffeafb0 3ffe9d10 3ffe9510 40204a33  
3ffeaf60:  3fff4918 00000001 00000001 3fff4f28  
3ffeaf70:  0000001b 0000001b 3fff4918 00000002  
3ffeaf80:  00000002 3fff50c0 00000071 00000071  
3ffeaf90:  3ffe8bb8 3ffeafbc 3ffeafbc 3ffe9544  
3ffeafa0:  3ffe9d10 3ffe9dc4 3ffeafbc 402021dd  
3ffeafb0:  3fff4fa0 0000003a 0000003a 3fff4fa0  
3ffeafc0:  00000005 00000005 3fff4918 00000001  
3ffeafd0:  00000001 3fff4ff8 0000003a 0000003a  
3ffeafe0:  00000000 3fff4e58 3ffe9510 402069b6  
3ffeaff0:  00000000 00000000 00000000 40204e1e  
3ffeb000:  3ffeb020 3ffeb020 3ffe9510 40204c7d  
3ffeb010:  3ffe9528 00000008 00000003 00000003  
3ffeb020:  3ffe8d38 00000002 3fff4ec0 3fff4f90  
3ffeb030:  3fff4e98 00000040 402017fe 3ffea090  
3ffeb040:  402010ae 00000001 3ffe9510 3ffeb0dc  
3ffeb050:  3fffdc20 000003e4 3ffe9510 40204d75  
3ffeb060:  3ffe8e50 00000000 000003e8 0002361e  
3ffeb070:  3ffe9528 3fff4ef0 00000016 40101d45  
3ffeb080:  4020181d 00000000 3ffeb0d4 40202328  
3ffeb090:  3fffdc20 00000000 3ffeb0d4 40201876  
3ffeb0a0:  00000000 00000000 3ffea090 40100450  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
erl
...
Connected to dlink
IP address: 192.168.0.108
HTTP server started
28256
28416
```

and

```
28416
28416
28416
28048
28024

Exception (9):
epc1=0x401018bc epc2=0x00000000 epc3=0x00000000 excvaddr=0x05eb2f07 depc=0x00000000

ctx: cont 
sp: 3ffead50 end: 3ffeb0b0 offset: 01a0

>>>stack>>>
3ffeaef0:  40207898 3ffeaf84 00000073 00000073  
3ffeaf00:  40207898 00000001 3ffeaf84 402078c6  
3ffeaf10:  40100200 00000001 3ffeaf84 40207c8d  
3ffeaf20:  40100218 00000003 3ffeaf84 3ffe8ef4  
3ffeaf30:  3ffeafb0 3ffe9d8c 3ffeaf84 40207ced  
3ffeaf40:  3ffe8ef4 3ffe9d8c 3ffe9510 40207cc2  
3ffeaf50:  3ffeafb0 3ffe9d10 3ffe9510 40204a33  
3ffeaf60:  3fff4918 00000001 00000001 3fff4f28  
3ffeaf70:  0000001b 0000001b 3fff4918 00000002  
3ffeaf80:  00000002 3fff50c0 00000071 00000071  
3ffeaf90:  3ffe8bb8 3ffeafbc 3ffeafbc 3ffe9544  
3ffeafa0:  3ffe9d10 3ffe9dc4 3ffeafbc 402021dd  
3ffeafb0:  3fff4fa0 0000003a 0000003a 3fff4fa0  
3ffeafc0:  00000005 00000005 3fff4918 00000001  
3ffeafd0:  00000001 3fff4ff8 0000003a 0000003a  
3ffeafe0:  00000000 3fff4e58 3ffe9510 402069b6  
3ffeaff0:  00000000 00000000 00000000 40204e1e  
3ffeb000:  3ffeb020 3ffeb020 3ffe9510 40204c7d  
3ffeb010:  3ffe9528 00000008 00000003 00000003  
3ffeb020:  3ffe8d38 00000002 3fff4ec0 3fff4f90  
3ffeb030:  3fff4e98 00000040 402017fe 3ffea090  
3ffeb040:  402010ae 00000001 3ffe9510 3ffeb0dc  
3ffeb050:  3fffdc20 000003e7 3ffe9510 40204d75  
3ffeb060:  3ffe8e50 00000000 000003e8 0000bae4  
3ffeb070:  3ffe9528 3fff4ef0 00000016 40101d45  
3ffeb080:  4020181d 00000000 3ffeb0d4 40202328  
3ffeb090:  3fffdc20 00000000 3ffeb0d4 40201876  
3ffeb0a0:  00000000 00000000 3ffea090 40100450  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
erl
...
Connected to dlink
IP address: 192.168.0.108
HTTP server started
27864
28024
28392
28024
28392
```
and 

```
28416
28392
28392
28416
28016

Exception (9):
epc1=0x40101789 epc2=0x00000000 epc3=0x00000000 excvaddr=0xdea07d2f depc=0x00000000

ctx: cont 
sp: 3ffeacf0 end: 3ffeb0b0 offset: 01a0

>>>stack>>>
3ffeae90:  0000005a 00000010 4020c4a0 3fff5ab8  
3ffeaea0:  00000000 00000000 40001e65 3ffeaef1  
3ffeaeb0:  000000ac 3ffeaed0 4020c6bd 00000020  
3ffeaec0:  00000001 3ffeaf3c 00000020 00000020  
3ffeaed0:  40100000 00000000 3ffeaf3c 402078c6  
3ffeaee0:  00000000 00000000 3fff50e0 3fff52b0  
3ffeaef0:  00000000 00000001 40203450 3fff4ef0  
3ffeaf00:  000000ac 00000000 3ffeaf60 000000ac  
3ffeaf10:  000000ac 000000ac 00000000 00000001  
3ffeaf20:  3ffe9510 00000000 000000ac 40204735  
3ffeaf30:  3ffe9528 000005b4 3ffeaf60 402079f4  
3ffeaf40:  4020784d 3ffe9510 3ffe9510 3ffe8ef4  
3ffeaf50:  3ffeafb0 3ffe9d8c 3ffe9510 40204a5e  
3ffeaf60:  3fff52b0 000000ac 000000ac 3fff4f28  
3ffeaf70:  0000001b 0000001b 3fff4918 00000002  
3ffeaf80:  00000002 3fff50e0 000000ac 000000ac  
3ffeaf90:  3ffe8bb8 3ffeafbc 3ffeafbc 3ffe9544  
3ffeafa0:  3ffe9d10 3ffe9dc4 3ffeafbc 402021dd  
3ffeafb0:  3fff5090 00000039 00000039 3fff4fa0  
3ffeafc0:  00000004 00000004 3fff4918 00000001  
3ffeafd0:  00000001 3fff5040 00000039 00000039  
3ffeafe0:  00000000 3fff4e58 3ffe9510 402069b6  
3ffeaff0:  00000000 00000000 00000000 40204e1e  
3ffeb000:  3ffeb020 3ffeb020 3ffe9510 40204c7d  
3ffeb010:  3ffe9528 00000008 00000003 00000003  
3ffeb020:  3ffe8d38 00000002 3fff4ec0 3fff4f90  
3ffeb030:  3fff4e98 00000040 402017fe 3ffea090  
3ffeb040:  402010ae 00000001 3ffe9510 3ffeb0dc  
3ffeb050:  3fffdc20 000003e7 3ffe9510 40204d75  
3ffeb060:  3ffe8e50 00000000 000003e8 00003e02  
3ffeb070:  3ffe9528 3fff4ef0 00000016 40101d45  
3ffeb080:  4020181d 00000000 3ffeb0d4 40202328  
3ffeb090:  3fffdc20 00000000 3ffeb0d4 40201876  
3ffeb0a0:  00000000 00000000 3ffea090 40100450  
<<<stack<<<

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
erl
...
Connected to dlink
IP address: 192.168.0.108
HTTP server started
28232
28392
28024
28392
```
hope it help

luc-github, 2015-06-22:
just remind the code :
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
extern "C" {
#include "user_interface.h"
}
 
const char* ssid = "dlink";
const char* password = "";//I removed my password

ESP8266WebServer server(80);

void handleRoot() {
	
  String message = "<HTML><BODY><H1>Page static</H1>";
  message += "<IFRAME  width=\"200\" height=\"200\"  NAME=\"frmstatus\"  ID=\"frmstatus\"  SRC=\"http://192.168.0.108/STATUS\"></IFRAME>\n<SCRIPT TYPE=\"text/javascript\">\n";
  message +="setInterval(function(){";
  message +="var ifrm=document.getElementById(\"frmstatus\");var doc=ifrm.contentDocument?ifrm.contentDocument:ifrm.contentWindow.document;";
  message +="doc.location.reload(true);";
  message +="},2000);\n";
  message +="</SCRIPT>\n</BODY></HTML>";
  server.send(200, "text/html", message);
 
}

void handleSubRoot() {
  static long cnta=0;
  static long cntb=0;
  cnta++;
  if(cnta>100000000)
  {
	  cntb++;
	  cnta=0;
  }
  
  String message = "<HTML>\n<BODY>\n<H1>refreshed page</H1>";
  message += String(cntb) + "/" + String(cnta);
  message +="</BODY>\n</HTML>\n";
  server.send(200, "text/html", message);
  Serial.println(String(system_get_free_heap_size()).c_str());
 
}
 
void setup(void){
  Serial.begin(115200);
  //Serial.setDebugOutput(true); 
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  
  
  server.on("/", handleRoot);
  server.on("/STATUS", handleSubRoot);
  
  server.begin();
  Serial.println("HTTP server started");
}
 
void loop(void){
  server.handleClient();
  
} 
```

igrr, 2015-06-22:
The exception happens because some pointers in RAM are corrupted. I suspect this is a power supply/shielding issue since I'm not able to reproduce it. I compiled your sketch and did 100 requests from command line:
```$ for i in $(seq 1 100); do curl -o /dev/null http://192.168.0.201/STATUS ; done```
ESP happily produced the following log:
```
 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
e�怘���{l l��| �d�|�d�c<����{�c�b��og�l'o���#p��${ls$p�n��d��co�|l���c��'o� d��$`�o'l`o;Ǜ�'�l�`{ۓo�d�`��
...........
Connected to testwifi
IP address: 192.168.0.201
HTTP server started
28456
28392
28456
28456
28440
28440
28456
28456
28456
28440
28440
28376
28456
28440
28440
28440
28440
28440
28440
28440
28440
28456
28440
28440
28456
28440
28440
28440
28440
28384
28440
26864
28440
28456
28456
28440
26848
28440
28456
28440
28440
28440
28456
28440
28456
28456
28440
28456
28456
28440
28440
28448
28440
28456
28440
28440
28440
28440
28456
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28440
28448
28440
28440
28448
28440
28456
28440
28440
28440
28440
28456
28440
28440
28448
28440
28440
28440
28440
28440
28440
```

luc-github, 2015-06-22:
100 requests - well is not reseting that fast, currently I am at 1300 without reset sometimes is after 4000 sometimes 500

luc-github, 2015-06-22:
To speed up testing I open 4 web pages at once =  currently I am at 300 loop without reset since last reset

What do you mean by "power supply/shielding issue" ? I have tested on PC - direct plugged to printer (Arduino DUE based board) and on power supply providing 2A and same result

igrr, 2015-06-22:
Oh okay, I'll set it running for a while.

igrr, 2015-06-25:
I was able to reproduce this — although i'm not sure what is the root cause of this issue.
The exception happens because heap gets corrupted. Why this happens is not yet clear — this might be a simple buffer overrun or some concurrency/race condition kind of issue. Wasn't able to solve this on the first try but will definitely more into this.

luc-github, 2015-06-25:
Excellent - thanks a lot for you great job
let me know if I can help / test - I would be happy to do

WoodsterDK, 2015-07-01:
I'm getting the exception also - actually just by having the esp8266 idle and checking for data using parsepacket()
Martin

WoodsterDK, 2015-07-01:
Forgot - I'm using the UDP connection

luc-github, 2015-07-07:
Hi any clue/update on this issue ? it is that linked to this one ? https://github.com/arduino/Arduino/issues/3004#issuecomment-95677470

folny, 2015-07-07:
Hi

I have the same problem with my code, the code at any time by itself restarts and also beats me why it would be great to find a solution to this problem Here is my code.

http://www.esp8266.com/download/file.php?id=1249

bbasil2012, 2015-08-07:
I have the same issue with ESP-07 board.
AdvancedWebServer and HelloServer from examples crashed with Exception(9):
WiFiWebServer from examples not crash, but stopped to response after some itteration...
#!/bin/bash
addr='10.172.0.16'
for ((j=0;j<10000;j++))
    do
	curl http://$addr/status
	echo "Itteration "$j
    done
exit 0

bbasil2012, 2015-08-07:
I did same modification in the example HelloServer:
1. I put "String message" from handleNotFound() to the global variable:
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
String message;

2. I added message.reserve(1024); in the setup section:
void setup(void){
  message.reserve(1024);

And now this scetch run very well without any crashes, WTD, freezing and other artefacts ! :)

EUA, 2015-08-26:
I have similar issues but I backed my MCU with LiPo battery. But it still resets time to time....
I start to believe that modules has some hw issue...

abhishek-dixit, 2015-08-29:
I see my ESP12 resets in a days time. I need to debug it for hardware issue because I have another board which did not reset for last 1 week. But their is something more of an observation. 

The ESP's heap keeps decreasing with every request but once left for more than a minute it recovers. And this should pretty consistent. Running a test request loop will reset it if a request is triggered every 2 seconds. 

Under normal circumstances it will not be required to hit requests so fast at least not in my case where I am running a TV, CHARGER and set top box with a single esp12.

luc-github, 2015-09-07:
I do not know if it is same issue and it can help to reproduce easily but I have a button on my web page,this button send simple query to webserver using a iframe which do not display anything as no need

```<IFRAME width="2" height="2" style="visibility:hidden" ID="frmcmd" NAME="frmcmd" ></IFRAME>   ```       
```<BUTTON TYPE="BUTTON" VALUE="Emergency Stop" Onclick="window.open('http://192.168.0.112/CMD?COM=M112','frmcmd');">Emergency Stop</BUTTON>```

If I press button like a crazy the web server restart after 2 or 3 seconds  usualy it happen 70% of the time.

[76 more replies]

closed 2015-09-29 by luc-github