esp8266/Arduino #434: Webserver stops responding

sticilface, 2015-06-16:
I've made a simple sketch that seems to reproducibly cause the web server to stop responding, but the ESP continues as normal.  I'd like to know if a) I'm doing something wrong b) other people can reproduce it.  load the sketch, the first page will auto refresh every 30 seconds which is fine.  click the link... the second page will load.  Wait a few min (usually 2-5), then click the link to go back to /.  every time i try this the server stops responding.  

```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

const char *ssid = "xxxxx";
const char *password = "xxxxx";

long time = 0; 

WiFiClient wifiClient;
ESP8266WebServer server(80);

 const char * simplepage1 = "\
  <html>\
  <head>\
  	<meta http-equiv='refresh' content='30'/>\
  	<title>ESP8266 Demo</title>\
  	<style> body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; } </style> \
  </head>\
  	<body> \
  		<h1>Hello from ESP8266!</h1> <br> \
  		<a href='/ws2812'>WS2812</a>\
  	</body>\
  </html>";

const char * simplepage2 = "\
<!DOCTYPE html>\
<html>\
<head>\
    <title>Hi there</title>\
</head>\
<body>\
  This is a page a simple page\
  <p> <a href='/'>Home </a>\
</body>\
</html>";

void handle_root() {
  Serial.print("handle root..."); 
  server.send(200, "text/html", simplepage1);
  Serial.println("finshed!");
}

void  handle_WS2812 () { 
  Serial.print("handle ws2812..."); 
  server.send(200, "text/html", simplepage2);
  Serial.println("finshed!");
}

void setup() {
  Serial.begin(115200); 
  delay(10); 
  Serial.println();
  Serial.println("Joining Wifi Network");

  WiFi.begin(ssid, password);

  if(WiFi.waitForConnectResult() != WL_CONNECTED){
    Serial.printf("Could not connect to %s\n", ssid);
    while(1) delay(500);
  }

    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP()); 

  server.on("/", handle_root);
  server.on("/ws2812", handle_WS2812); 

  server.begin();
}

void loop() {
	server.handleClient();
	if (millis() > time+ 1000) {
		Serial.print(".");
		time = millis();
	}
}
```

TimeTravelingOwls, 2015-06-17:
Works fine for me - tried several times, waiting 1, 2 & 5 minutes.

BTW, slipping this line into your ```<head>``` makes life much easier if you're testing with a tiny screen:

```    <meta name='viewport' content='initial-scale=1'> ```

sticilface, 2015-06-22:
right... well i turned everything off... and back on.  router wifi etc.. and the problem has gone!  false alarm.

closed 2015-06-22 by sticilface