esp8266/Arduino #435: csum err when using 1.6.4, everything fine when using 1.6.1

drnokill, 2015-06-16:
Hello,
when upgrading to 1.6.4 my ESP 8266-07 does no longer start.
I'm running Windows 8.1 Pro (64 Bit) with Visual Studio 2013 Professional and Visual Micro.
Using the Adruino IDE has the same effect.

Serial Output after booting:
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)<\r><\n><\r><\n>
load 0x40100000, len 28780, room 16 <\r><\n>
tail 12<\r><\n>
chksum 0x00<\r><\n>
ho 0 tail 12 room 4<\r><\n>
load 0x3ffe8000, len 1580, room 12 <\r><\n>
tail 0<\r><\n>
chksum 0x4c<\r><\n>
load 0x3ffe8630, len 36588, room 8 <\r><\n>
tail 4<\r><\n>
chksum 0xb1<\r><\n>
csum 0xb1<\r><\n>
csum err<\r><\n>
ets_main.c 

When switching back to 1.6.1 (using esptool) it runs.
I'm using an ESP8266-07 Board with a 25Q40BT Flash-Chip.
I already tried all Flash Settings in the Board-Options

Here is the Output of the uploading process:
Binary sketch size: 280.620 bytes (used 7% of a 4.194.304 byte maximum) (59,95 secs)
Uploading to I/O board using 'COM1'
Uploader started for board NodeMCU (ESP8266 ESP-12 Module)
Upload method will be: bootloader
Uploading via Bootloader 
Flushing the serial port
C:\Users\DrNok_000\AppData\Roaming\arduino15\packages\esp8266\tools\esptool\0.4.4\esptool.exe -vv -cd ck -cb 115200 -cp "COM1" -ca 0x00000 -cf "C:\Users\DrNok_000\AppData\Local\V.Micro\Arduino\Builds\Test\nodemcu/Test_00000.bin" -ca 0x10000 -cf "C:\Users\DrNok_000\AppData\Local\V.Micro\Arduino\Builds\Test\nodemcu/Test_10000.bin"
esptool v0.4.4 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM1
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat C:\Users\DrNok_000\AppData\Local\V.Micro\Arduino\Builds\Test\nodemcu/Test_00000.bin success
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_send_command: receiving 2 bytes of data
	espcomm_open
Uploading 66992 bytes from C:\Users\DrNok_000\AppData\Local\V.Micro\Arduino\Builds\Test\nodemcu/Test_00000.bin to flash at 0x00000000
	erasing flash
	size: 0105b0 address: 000000
	first_sector_index: 0
	total_sector_count: 17
	head_sector_count: 16
	adjusted_sector_count: 9
	adjusted_size: 009000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 10000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
..................................................................
	setting address from 0x00000000 to 0x00010000
	espcomm_upload_file
	stat C:\Users\DrNok_000\AppData\Local\V.Micro\Arduino\Builds\Test\nodemcu/Test_10000.bin success
	espcomm_open
Uploading 213672 bytes from C:\Users\DrNok_000\AppData\Local\V.Micro\Arduino\Builds\Test\nodemcu/Test_10000.bin to flash at 0x00010000
	erasing flash
	size: 0342a8 address: 010000
	first_sector_index: 16
	total_sector_count: 53
	head_sector_count: 16
	adjusted_sector_count: 37
	adjusted_size: 025000
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	setting serial port timeouts to 10000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
.................................................................................................................................................................................................................
starting app without reboot
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	espcomm_send_command: receiving 2 bytes of data
closing bootloader
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete

Done uploading

probonopd, 2015-06-20:
I also get 

```
esptool v0.4.2 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.SLAB_USBtoUART
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
...
starting app without reboot
	espcomm_cmd: sending command header
	espcomm_cmd: sending command payload
	espcomm cmd: receiving 2 bytes of data
closing bootloader
```
When I then run the app, I get
...
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 32640, room 16 
tail 0
chksum 0xef
load 0x00000000, len 0, room 8 
tail 0
chksum 0xef
load 0x00000000, len 0, room 0 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c 
```
This can be seen e.g., using ESPlorer on the Mac set to 74880 bits/s and booting without pulling GPIO0 low (which was used for programming).

probonopd, 2015-06-21:
On one of my modules (ESP-12) I always get 
```
tail 8
chksum 0xef
csum 0xef
csum err
ets_main.c 
```
I checked my setup and using an ESP-01 instead I do not get this error. Can this have to do with a different flash chip? Using esptool.py, writing to the ESP-12 and reading back the binary from the ESP-12 results in identical files, so I assume the flash is working properly.

probonopd, 2015-06-21:
I also tried with "NodeMCU 0.9 (ESP-12 Module)" but it still shows the ```csum err```.

probonopd, 2015-06-21:
Using esptool.py I checked the flash chip of my ESP-12:
```
Manufacturer: c8
Device: 4016
```
I found on the net that this is a Winbond W25Q32 and that the W25x series supposedly does not support qio mode. Hence I tried to set the board in the IDE to "NodeMCU 1.0 (ESP-12E Module)" (because I see in the boards.txt that it uses nodemcuv2.build.flash_mode=dio).

This time I get:
```
load 0x4010f000, len 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
e
```
At least no `csum err` this time. So I guess my ESP-12 board is just not supported yet?

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004