esp8266/Arduino #437: Wrong GPIO number on ESP8266-12

anhhuy0501, 2015-06-17:
I use this code:

void setup() {
    pinMode(4, OUTPUT);
}
void loop() {
  digitalWrite(4,0);
  delay(1000);
  digitalWrite(4,1);
  delay(1000);
}

The esp8266-12 should change the state of pin GPIO4 but it changes the state of GPIO5 instead. Is the pin diagram wrong?

The pin diagram of ESP8266-12 look like in this link: https://raw.githubusercontent.com/Links2004/Arduino/esp8266/docs/ESP_improved_stability.png

wizard23, 2015-06-17:
Just guessing: I have a "ESP12" and a "ESP12E" module. They have almost the same pin labels in the silkscreen. The only difference is that on the ESP12E GPIO4 and GPIO5 are swapped compared to the ESP12 module. I think this is just an error in the silkscreen of the ESP12E. The pins are just swapped on the silk but the actual electrical conection on the pcb is still the same as on the ESP12. 

Of course using this logic you could also deduce that the label on the ESP12 is incorrect but since it's consistent with the pin numbering in the arduino environment I think it's save to assume that the ESP12 pin lables are correct ;)

On can actually measure this if you take of the metal cover.

iamthemadz, 2015-06-17:
One member at the esp8266 forum left this post; 

http://www.esp8266.com/viewtopic.php?f=5&t=3163

igrr, 2015-06-17:
All ESP12 modules I have purchased seem to have GPIO4 and GPIO5 labels swapped on the silkscreen. I have verified that by removing metal cap and following the traces to chip pads.

stozk, 2015-06-17:
I also noticed that on my esp12 chips.

duncan-a, 2015-06-17:
Does anybody know if this also applies to the 07s? They look to be the same
as the 12, but with a ceramic antenna and uFL connector replacing the PCB
version.

I have a bag of these on order......

me-no-dev, 2015-06-18:
yes :) they are all wrong. ESP-07, ESP-12 all that are in that same package but with different antennas have them labeled wrong. From some experience with Chinese manufacturing I can say that they probably got the gerber files for the modules from somewhere and have no clue how to change the silk screen to fix that year old issue. Also on ESP-12e, the new broken pins are labeled with the names for pins 12-15... even more havoc :D

Testato, 2015-06-20:
Is it important that only the label is wrong, so in the future you can change a broken Esp without problem.

closed 2015-06-25 by igrr