esp8266/Arduino #440: Library not updating in Preferences

tytower, 2015-06-17:
Initially I entered the json address in preferences and the update happened dated 19/5/15 and then while using the IDE it updated itself on about 28/5/15 . I had a glitch and went back to a restored copy of my OS which only had the 19/5/15 version in it . When I opened the Arduino IDE I expected it to update automatically . It didn't and it hasn't .

I also noticed that previously with the 28/5/15 version the Preferences windo did not show the file box to enter the json addresses any more .

igrr, 2015-06-17:
Please file issues related to the IDE in the Arduino issue tracker, as here we are only developing the esp8266 core. As far as I know, Arduino does not yet have automatic updates of board manager packages.

Testato, 2015-06-18:
maybe my question is related to this,

here https://github.com/esp8266/Arduino/releases
there is the 1.6.5 version of this Core ? Updated at 4 day ago ?
Why the json address is not updated ? So by Board Manager we receive the old "version":"1.6.4-673-g8cd3697"

igrr, 2015-06-18:
This is the tag created upstream, I must have pushed it accidently. It's on the master branch and as such doesn't contain this core.

Testato, 2015-06-18:
Is there a way for updating to the actual version of this core via board manager or we must await for an updated json link ?

igrr, 2015-06-18:
If you want, you can clone the repository and run the build/build_board_manager_package.sh script, it will build the board manager package and start a local python web server. Then you can point your ide to http://localhost/.... and it will grab the build package.

closed 2015-07-23 by igrr