esp8266/Arduino #441: OAuth2

vicatcu, 2015-06-17:
Can we get an example of using OAuth2 together with HTTPS to access an API without going through a proxy server like Temboo? Is that possible/practical with the ESP8266?

cimba007, 2015-06-18:
Did you have a look at this OAuth1 implementation for Twitter+Arduino?

http://www.markkurossi.com/ArduinoTwitter/index.html

vicatcu, 2015-06-18:
@cimba007 yes I had come across that in my google-fu, I'm not sure what the gap is between OAuth1 and OAuth2 though, nor if work has already been done on an implementation to that end.

Links2004, 2015-11-27:
HTTPS is supported now,
you can use ```httpClient``` to make http or https requests

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpClient/examples/BasicHttpClient/BasicHttpClient.ino

closed 2015-11-27 by Links2004