esp8266/Arduino #442: Source code of toolchains and tools

chrwick, 2015-06-18:
Hi,

where do I find the source code that corresponds to the content of the ZIP-files that are referenced here: http://arduino.esp8266.com/package_esp8266com_index.json

I would like to be able to compile them on my own, too.
Is there a well automated build process for this already?

igrr, 2015-06-18:
The toolchain for Linux and Mac is built using crosstool-NG. You may find
tutorials about this method on the esp8266 community wiki.
On Windows I used this script:
https://gist.github.com/fpoussin/7ae55b5a5bd9a28ce21d
possibly with some minor modifications specific to my mingw installation.
There is a build script for automated Linux x64 builds on Travis CI
available, but I don't use it. Potentially you can run automated builds for
Windows as well on Appveyor but I didn't have a need to do that yet. Mac OS
builds are possible on Travis but you need to ask them to enable this
platform for your account, it's not available by default. Also since I'm on
a Mac, I didn't find that necessary so far.

 esptool is available from https://github.com/igrr/esptool-ck, as mentioned
in Readme. It may be built by running make dist on all platforms (check
appveyor.yml for Windows-specific stuff). It has automated build for
Windows and Linux x64 running on Appveyor and Travis-CI respectively. I
build Mac version on my laptop.

http://arduino.esp8266.com/package_esp8266com_index.json

<https://github.com/esp8266/Arduino/issues/442>.<img src="
https://ci5.googleusercontent.com/proxy/iubkub8fGEpOVX5kGLnyMqQNCxoehZD41k6kSeQUZckqteQ3S4-ehupHP9ixSY89jIsW3eMKM34DPShaweG2tZLcUkaPRrz8VcNifxokG8KwR_E04cwxNSgGIder11DzTAAlChDpiCX9wgDbCjg-U026N_24Vw=s0-d-e1-ft#https://github.com/notifications/beacon/AEJcevc7goONr5OUYyFEpWZW-dUyKuNhks5oUlWBgaJpZM4FGAG6.gif
">

chrwick, 2015-06-19:
Ok, thanks for the reply.

Having some officially documented process with scripts would be helpful, but I consider this an enhancement for the future when the more important things have been taken care about.

Closing.

closed 2015-06-19 by chrwick