esp8266/Arduino #444: Unsteady Autoreset-Fail on flash

cimba007, 2015-06-18:
Hello,

I am using ESP03 as a generic board with esptool v0.4.5 with this params (
... esptool.exe -vv -cd ck -cb 512000 -cp COM3 -ca 0x00000 -cf ... esp8266_ssd1306_128x64_i2c.cpp.bin)

I am using one of these cheap ebay USB-Serial-Converteres, grabbing 5V USB-Power line and an LDO to 3.30V .
( e.g. http://www.ebay.de/itm/1Stk-USB-2-0-auf-TTL-UART-6Pin-Module-Serial-Converter-CP2102-STC-Neu-/111630933136?pt=LH_DefaultDomain_77&hash=item19fdb8dc90 )

Around 90% of the time just hitting upload and waiting for auto-reset to kick in is working fine.

But sometimes (mostly after starting arduino ide) I get these guys like 4-5 times in a row

warning: espcomm_send_command: wrong direction/command: 0x01 0x02, expected 0x01 0x03
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x02, expected 0x01 0x04

I suspect (just guessing) a timing issue with the reset-commands. Is there any way to modifie them?

Is anyone else experiencing this rather random misbehaviour?

Greetings from Germany

Testato, 2015-06-19:
Same here on a nodemcu board (esp12+ch340 usb)

probonopd, 2015-06-28:
Try my wiring from https://github.com/esp8266/Arduino/issues/480

Links2004, 2015-06-28:
test to reduce the upload speed,
512000 is on the higher end, and only working by good and short wiring.

Links2004, 2015-11-27:
please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004