esp8266/Arduino #449: EEPROM library - put and get doesn't work well

robertgregor, 2015-06-19:
Hello, 
seems like that put and get doesn't work well.
I have this code:

#define PIR_CONFIG 20 //there is nothing writting after byte 20 in eeprom

typedef struct {
String notificationURL;
String smtp;
String smtpUserName;
String smtpPassword;
String subject;
String message;
} PirSettings;

void loadDataFromEeprom() {
     EEPROM.get(PIR_CONFIG, pirSettings);
}
void setup() {
  Serial.begin(115200);
  Serial.println("Starting...");
  EEPROM.begin(512);
  loadDataFromEeprom();
}

//is called in the webserver method handler
void handle_deviceCfgConfigure() {
  pirSettings.notificationURL = server.arg("pirURL");
  pirSettings.smtp = server.arg("srv");
  pirSettings.smtpUserName = server.arg("usr");
  pirSettings.smtpPassword = server.arg("pwd");
  pirSettings.subject = server.arg("sbj");
  pirSettings.message = server.arg("msg");
  EEPROM.put(PIR_CONFIG, pirSettings);
  EEPROM.commit();
  server.send(200, F("text/html"), getStartPageWithMenu() + F("<p>Device configured.</p>") + getEndPage());
}

When I did this, after power off power on, the data in the pirSettings are completely wrong - it is tee like data.

Am I doing something wrong? Or is there a bug in the EEPROM library???

Testato, 2015-06-19:
This is not a direct answer to your question, but on this soc there is no real Eeprom, it is emulated. So you write always on flash.
Try directly Progmem instead of eeprom.

igrr, 2015-06-20:
Your issue is related to trying to save String objects in a struct.
String actually contains a pointer to the actual memory location where the string data is located. So when you save the struct of strings to EEPROM, you are actually saving pointers to dynamically allocated buffers. After restart, these pointers are loaded from EEPROM, but the actual text data they pointed to is no longer there — hence you get garbage.

One way to resolve this is to have a struct of C arrays of fixed size:
```C++
typedef struct {
char notificationURL[64];
char smtp[64];
char smtpUserName[32];
char smtpPassword[32];
char subject[64];
char message[128];
} PirSettings;
```

Then you can use ```strncpy``` function to copy contents of a String into C array:
```C++
String notificationURL = server.arg("pirURL");
strncpy(pirSettings.notificationUrl, notificationURL.c_str(), sizeof(pirSettings.notificationUrl));
```

And when you need to read, just initialize the string with C array:
```C++
String notificationURL(pirSettings.notificationUrl)
```

robertgregor, 2015-06-20:
Hello, OK, I see. I was thinking, that this is the problem and already rewrote it. I know, about pointers and so on, just I am sure, that the Arduino get and put box/unbox the String and save the content, not the pointer.

closed 2015-06-20 by robertgregor

igrr, 2015-06-21:
We can provide an explicit template instantiation for String type which will do box/unbox inside... This way you will be able to get/put String contents. But there isn't a way to do this if String is a part of the struct — C++ lacks introspection features necessary to do this...