esp8266/Arduino #451: ESP8266-12E not supported by package_esp8266com_index.json

asantaga, 2015-06-20:
All , had problems getting a NodeMCUv2 , and a vanilla ESP8266 working with arduino. After much discussion on http://www.esp8266.com/viewtopic.php?f=26&t=3635 we discovered this to be the JSON boards string not pulling in the NodeMCU2 board details (Slightly different flash)

Workaround
1. Install Arduino
2. Install additional boards
3. Locate addition boards.txt and cut-n-paste nodemcu2 section from https://github.com/esp8266/Arduino/blob/esp8266-sdk-1.0/hardware/esp8266com/esp8266/boards.txt into boards.txt
4. Restart Arduino

Bug
Why does http://arduino.esp8266.com/package_esp8266com_index.json not pull in complete boards.txt list?

igrr, 2015-06-20:
This is because the board manager package was not updated recently. If you check the version, you'll see it contains ```"version":"1.6.4-673-g8cd3697"```, so it was generated from commit 8cd3697, while support for NodeMCU 1.0 was added two days later in f6a78cd.

asantaga, 2015-06-20:
OK, I'll test it on a fresh install , if it works then I'll close it.

closed 2015-06-20 by asantaga