esp8266/Arduino #452: LCD Display not working with wire.begin(0,2)

mvdbro, 2015-06-20:
Have been using this with my I2C LCD display without problems:

#include <Wire.h> 
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27,16,2);

void setup()
{
  Wire.pins(0,2);
  Wire.begin();
  lcd.init();                      // initialize the lcd 
 
  // Print a message to the LCD.
  lcd.backlight();
  lcd.print("Hello, world!");
}

void loop()
{
}

But now wire.pins() is deprecated, so I decided to update the sketch and now it doesn't work anymore:

#include <Wire.h> 
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27,16,2);  // set the LCD address to 0x27 for a 16 chars and 2 line display

void setup()
{
  Wire.begin(0,2);
  lcd.init();                      // initialize the lcd 
 
  // Print a message to the LCD.
  lcd.backlight();
  lcd.print("Hello, world!");
}

void loop()
{
}

Can't figure out why... Display does not init and does not show any text message...
I have to keep working the old way with wire.pins...

me-no-dev, 2015-06-20:
```cpp
void TwoWire::begin(int sda, int scl){
  twi_init(sda, scl);
  flush();
}

void TwoWire::pins(int sda, int scl){
  default_sda_pin = sda;
  default_scl_pin = scl;
}

void TwoWire::begin(void){
  begin(default_sda_pin, default_scl_pin);
}
```

As you can see, Wire.pins(sda,slc) + Wire.begin() actually calls Wire.begin(sda,scl)
So I have no clue why one would work and not the other :(

me-no-dev, 2015-06-20:
actually on another note, if you call Wire.begin(sda,scl) and your library calls internally Wire.begin(), then it will go to crap... will fix and push the update upstream. Use pins+begin for now

Duality4Y, 2015-06-21:
I have this too, i just keep using the deprecated function call.

igrr, 2015-06-21:
@me-no-dev has pushed the fix, and I will remove deprecation warning.

closed 2015-06-24 by mvdbro