esp8266/Arduino #454: ESP8266WebServer doesn't encode/decode correctly the url arguments from form, when get method is used

robertgregor, 2015-06-20:
Hello, I have a following html form:

<form method='GET' action='cwdc'>

When reading the form arguments with method arg(value), which contain i.e. space, I can read the space as +. I think, that the method in case of GET doesn't decode the URL encoded special characters.

luc-github, 2015-06-22:
I use this function in my webserver to decode - I found it using google -  I forget where
```
//URI Decoding function 
//no check if dst buffer is big enough to receive string so 
//use same size as src is a recommendation
void WEBINTERFACE_CLASS::urldecode(char *dst, const char *src)
{
  char a, b,c;
  if (dst==NULL) return;
  while (*src) {
    if ((*src == '%') &&
      ((a = src[1]) && (b = src[2])) &&
      (isxdigit(a) && isxdigit(b))) {
      if (a >= 'a')
        a -= 'a'-'A';
      if (a >= 'A')
        a -= ('A' - 10);
      else
        a -= '0';
      if (b >= 'a')
        b -= 'a'-'A';
      if (b >= 'A')
        b -= ('A' - 10);
      else
        b -= '0';
      *dst++ = 16*a+b;
      src+=3;
    } 
    else {
		c = *src++;
		if(c=='+')c=' ';
      *dst++ = c;
    }
  }
  *dst++ = '\0';
}
```

robertgregor, 2015-06-22:
Yes, I did something also, but it is a bug in ESP8266WebServer and should be I think fixed.

robertgregor, 2015-06-25:
I am using this, thus these are the mandatory escape characters:

String escapeParameter(String param) {
  param.replace("+"," ");
  param.replace("%21","!");
  param.replace("%23","#");
  param.replace("%24","$");
  param.replace("%26","&");
  param.replace("%27","'");
  param.replace("%28","(");
  param.replace("%29",")");
  param.replace("%2A","*");
  param.replace("%2B","+");
  param.replace("%2C",",");
  param.replace("%2F","/");
  param.replace("%3A",":");
  param.replace("%3B",";");
  param.replace("%3D","=");
  param.replace("%3F","?");
  param.replace("%40","@");
  param.replace("%5B","[");
  param.replace("%5D","]");  
  return param;
}

igrr, 2016-03-10:
Actually fixed in 34fcc911 already, and released in 2.1.0.

closed 2016-03-10 by igrr

Baumi1245, 2016-04-29:
Still got a problem with it:

String R1 = server.arg("R1");
i let me show "R1":

http://192.168.125.150/cmd?R1=1
answer = "1"

http://192.168.125.150/cmd?R1%3D1
answer = nothing

http://192.168.125.150/cmd?R1=%3D1
=1

%3d is decodet to "=" but too late.
and i exactly need to run the command which is not working :-(
http://192.168.125.150/cmd?R1%3D1

Greetings from Austria, Tom.