esp8266/Arduino #456: helloMesh sketch fails to compile: 'exchangeInfo' was not declared in this scope

probonopd, 2015-06-21:
The helloMesh sketch fails to compile using 1.6.4-835-g77d77e8 with `error: 'exchangeInfo' was not declared in this scope`.

```
ESP8266WiFiMesh/ESP8266WiFiMesh.cpp.o 
$HOME/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.4-835-g77d77e8/libraries/ESP8266WiFiMesh/src/ESP8266WiFiMesh.cpp:101:6: error: 'WiFiMesh' has not been declared
 void WiFiMesh::connectToNode(String target_ssid, String message)
      ^
$HOME/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.4-835-g77d77e8/libraries/ESP8266WiFiMesh/src/ESP8266WiFiMesh.cpp: In function 'void connectToNode(String, String)':
$HOME/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.4-835-g77d77e8/libraries/ESP8266WiFiMesh/src/ESP8266WiFiMesh.cpp:118:40: error: 'exchangeInfo' was not declared in this scope
  if (!exchangeInfo(message, curr_client))
                                        ^
```

igrr, 2015-06-21:
This should be fixed in https://github.com/reaper7/Arduino/commit/44bde71ac6ab802806c2101ecee897bac9da7ae9

probonopd, 2015-06-21:
I can confirm that it is compiling with https://github.com/reaper7/Arduino/commit/44bde71ac6ab802806c2101ecee897bac9da7ae9 and I get `Setting up mesh node...` when running it, but since I have only one ESP module right now I cannot test for actual functionality.

igrr, 2015-06-26:
Should be fixed in esp8266-1.6.5-804-g2d340c7.

closed 2015-06-26 by igrr