esp8266/Arduino #457: mDNS responder sample does not work

probonopd, 2015-06-21:
I am using the ESP8266 mDNS responder sample sketch on an ESP-01 with 1.6.4-835-g77d77e8.
Putting http://esp8266.local as URL into the Safari address bar does not work (just seems to be working forever), but using the IP address of the ESP-01 does show "Hello from ESP8266 at 192.168.0.xx".
Running the sketch gives
```
.........
Connected to h48
IP address: 192.168.0.16
mDNS responder started
TCP server started
```

wattyka, 2015-06-26:
Same here with esp8266-1.6.4-673-g8cd3697 and ESP-07. Webserver working fine. MDNSResponder init ok. Wireshark shows multicast DNS request from PC to esp8266.local, but ESP doesn't seem to answer. I've set MDNS_DEBUG, but nothing is traced to serial. Looks like it is not able to receive.

Also confusion about MDNSResponder::update() - the call in loop() has been removed from the example, but the docs still say it has to be called. If it must not be called, it should probably be private.

wattyka, 2015-06-27:
My bad.

I learned that Win 7 is not able to resolve by mDNS out of the box. So I installed Bonjour on pc2, and voila - it sends an mDNS query and the ESP responds nicely. I have no idea why pc1 did send mDNS queries, as currently no bonjour is installed (I uninstalled iTunes months ago).

I read that Win 10 will support mDNS and even DNS-SD.

probonopd, 2015-06-28:
I am trying this on OS X 10.9 which has Bonjour support built in. Can anyone confirm that the ESP8266 mDNS responder sample sketch is working with 1.6.4-835-g77d77e8 or later?

probonopd, 2015-06-29:
With the changes from https://github.com/me-no-dev/Arduino/commit/1d51c35db553318a6bc0ef0b432ae84a7e7251eb it is now working for me, I can open http://esp8266.local. successfully (although I cannot see it advertised in Bonjour).

closed 2015-07-23 by igrr

romandubrov, 2015-10-30:
I have the similar issue, After some time since ESP started, webserver is still responding perfectly by IP, but not by esp.local hostname. Got latest libs and it doesn't help. I've uncommented MDNS_DEBUG_* defines and have a debug data when ESP just started, but there is no any debug when responder become unavailable. It looks like Udpcontext stops listening or calling onRx handler after some inactivity, but I didn't debugged it deeply. Any suggestions?

igrr, 2015-10-30:
If there is an intermittent loss of wifi connectivity, all open connections will be closed. However this doesn't explain why you can still access the web server.

luc-github, 2015-10-30:
On my side, I triggered issue to device itself -  4 ESP12E and 2 NodeMCU1.0 same code => on ESP12E devices, mDNS work like a charm, on NODEMCU no way, could be a HW root cause ?

luc-github, 2015-10-30:
seems there are some incompatibility with AP or AP configuration - I just changed AP (from Dlink to tplink) and now both NodeMCU which have no mDNS working, work as normal - unfortunatly I cannot "play" with the dlink AP to change settings as it the internet access / DHCP server for the house and I am not the only one using it.

romandubrov, 2015-11-03:
played a bit with device - when I kick esp from router control panel and make router to issue new IP address to esp - mdns starts responding again without restarting esp. however just kicking esp to make it reconnect and obtain same IP worked for me only in 1 case out of 10. I suspect the problem is in a way how esp handles the connection.

igrr, is there any way to graceful restart udp context listening when connection was changed? I don't see any callback there, and polling status() looks not reliable to me. I guess restarting listener every N minutes may be a quick but effective fix, but it is not elegant. May be you can suggest something?

Hassan-Ali-Hassan, 2017-02-01:
I have the same problem and I can't figure it out. I am using ESP8266-01 with Arduino IDE 1.8.  When I use the IP address of the module it works perfectly, but when I use the name assigned to the MDNSResponder object, it doesn't work. I looked around and as far as I understand that there was a bug in some older version and it was supposed to be fixed. However, it doesn't work now. Any help?

flagtrax, 2017-08-31:
I've just started playing with mDNS, and suffer the same issues. I don't see a real resolution anywhere. I'm using and example program by Pieter P. here [https://tttapa.github.io/ESP8266/Chap08%20-%20mDNS.html](url) and as others state I can ping the IP address fine from Windows 10 Command Prompt. . If I try to ping the mDNS name an erroneous IP address is returned (198.105.244.14) (the actual is 192.168.0.110)with 4 lost packets. I've tried pinging with the example code, then loading the code from the next chapter setting up "hello world" server. In both cases I get a return from the 8266 using IP address, but not mDNS. In the latter case I tried 3 browsers: chrome, edge, and opra. Is there a resolution somewhere I've missed? Thanks

MikeyMoMo, 2017-09-11:
I am on Windows 10 with a good WeMos D1 Mini.  Everything is working well, including the great WiFiMulti (thanks to the author) except for mDNS.  It claims to start but does not respond.  I am wired.  My AP is an ECB350 in the same room.  No problem with signal.  No problem connecting and getting info from the D1.  If you want to see this little jewel in action, go to stats.ILikeTheInternet.com and see how hot it is here.  

nDNS is not vital but it would be nice if it could be made to work.  Thanks to all contributors of code and answers.  Could NOT do it without you!!!

usmanshahid001, 2018-03-09:
yes mDNS not responding

jpascher, 2018-04-22:
Looks is still suffer the same problem, i am on windows 10 and use a WeMos D1

jpascher, 2018-04-26:
avahi-browse --all lists my deceives but i can't ping the names. Using " ARMBIAN 5.38 stable Ubuntu 16.04.4 LTS 3.4.113-sun8i" on orange pi.
From windows 10 with itunes installed and service running:
(Bonjour Service, Dienst "Bonjour", Own process, Running, Auto start, 1308)
also still no way to use the names. 
Wireshark shows mDNS:
Standard query 0x0000 A solar.local, "QM" question AAAA solar.local, "QM" question
Standard query response 0x0000 A, cache flush 192.168.0.2

Frame 44: 77 bytes on wire (616 bits), 77 bytes captured (616 bits) on interface 0
Ethernet II, Src: IntelCor_b0:45:10 (00:22:fb:b0:45:10), Dst: IPv4mcast_fb (01:00:5e:00:00:fb)
User Datagram Protocol, Src Port: 5353, Dst Port: 5353
Multicast Domain Name System (query)
Frame 45: 83 bytes on wire (664 bits), 83 bytes captured (664 bits) on interface 0
Ethernet II, Src: IntelCor_b0:45:10 (00:22:fb:b0:45:10), Dst: IPv4mcast_fb (01:00:5e:00:00:fb)
Internet Protocol Version 4, Src: 192.168.0.2, Dst: 224.0.0.251
User Datagram Protocol, Src Port: 5353, Dst Port: 5353
Multicast Domain Name System (response)

wi3ik, 2019-08-29:
for me, example sketch started to work after I run the following URL
esp8266.local**/**

note that without last "slash" symbol it didn't work for me

apulai, 2020-09-28:
Hello,
For me it started working when I have added to the main loop  these lines:
// Allow MDNS processing
  MDNS.update();
I think this is critical.

I do not know if this is critical, but I have this in my setup() function:
Use Wifi in STA mode (not AP mode):
WiFi.mode(WIFI_STA);
  delay(100);

And I am also registering an http service on port 80:
if (MDNS.begin("myname")) {
    MDNS.addService("http", "tcp", 80);
    Serial.println("\nMDNS responder started");
  }

probonopd, 2020-09-29:
> note that without last "slash" symbol it didn't work for me

Where would this bug have to be fixed?

MohammedNoureldin, 2020-11-08:
I have the same problem, any solution?

bass1959, 2021-03-06:
> apulai commented on 28 Sep 2020
> Hello,
> For me it started working when I have added to the main loop these lines:
> // Allow MDNS processing
> MDNS.update();
> I think this is critical.

Yep - that's it! And logical in a way as well - since all the coding of the example (in my case taken from A-Beginner's-Guide-to-the-ESP8266) takes place in setup() .
Apparently something has to be done in loop() as well, so the solution (in my case) is as simple as

void loop() {
MDNS.update();  // Allow MDNS processing
}

Check it out, and leave apulai a ThumbsUp for his hint from 2020-09-28!!