esp8266/Arduino #458: Implement DNS-SD to be seen as a bonjour service

probonopd, 2015-06-21:
As per https://github.com/esp8266/Arduino/issues/7#issuecomment-89640838 "to be seen as a bonjour service we need to support DNS-SD, not just bare mDNS. DNS-SD may be implemented on top of mDNS".

igrr, 2015-06-21:
This is necessary for OTA integration into the IDE by the way.

probonopd, 2015-06-27:
See https://github.com/TrippyLighting/EthernetBonjour

Bonjour (ZeroConf) Library for Arduino which does mDNS (registering services) and DNS-SD (service discovery). According to the description, the newest revised code replaces all direct hardware calls to the W5100 chip with calls to EthernetUDP methods. This will provide much better adaptability to different Ethernet hardware.

probonopd, 2015-06-29:
Someone seems to have done it at https://github.com/me-no-dev/Arduino/commit/1d51c35db553318a6bc0ef0b432ae84a7e7251eb although it did not work for me testing with the supplied test sketch and OS X 10.9. Safari doesn't show esp8266 in the Bonjour bookmarks. Also, Bonjour Browser doesn't show an entry. But I can successfully open http://esp8266.local. now.

smarg120, 2015-07-19:
with the patch above, Zeroconf Browser on Android could finally see my esp8266 device!

closed 2015-07-30 by igrr

hartas17, 2015-09-21:
Hi, can someone help me? how can i make the Zeroconf Browser see my esp8266?

sticilface, 2015-09-21:
This works for me.  Can add it under the similar line for the OTA service, in the OTA sketch. 
'''
    MDNS.addService("http", "tcp", 80);
'''

hartas17, 2015-09-21:
Thanks a lot... really thanks.