esp8266/Arduino #460: Command to disable WiFi to save power

probonopd, 2015-06-21:
Is there a way to disable WiFi altogether? Normally the ESP-01 consumes about 0.07A but it should be possible to bring power consumption significantly down to 0.01-0.02A by going to "Modem-Sleep" (as it is called by the manual). In "Modem-Sleep", CPU and other peripherals are still running, so the sketch should be able to continue running as usual.

[image]

chadouming, 2015-06-21:
This is not an issue, ask your question in relevant part of the forum : [Forum](http://www.esp8266.com/viewforum.php?f=25&sid=f82e14927b91bb1b964c93878ef5b1c5)

igrr, 2015-06-21:
I've just checked the SDK docs for ```wifi_set_sleep_type``` function and they have this note:
```
Note: Default to be Modem sleep.
```
So modem sleep is enabled by default.

probonopd, 2015-06-21:
I am measuring 0.07A power consumption using a "Charger Doctor" like shown here, regardless of whether I am connected to WiFi or just running a blink sketch. Is that the best we can do?

[image]

duncan-a, 2015-06-21:
Isn't that a 5V device???

probonopd, 2015-06-21:
Yes but I am using a voltage regulator to bring it down to 3.3V

duncan-a, 2015-06-21:
OK, so how much are the voltage regulator, power LED and other things
'wasting'?

Frankly I wouldn't rely on that being accurate anyway - is it's own display
and electronics consumption having an effect?

Are you actually invoking the default Modem Sleep mode?

cimba007, 2015-06-22:
Have a look at the sleep modes like WAKE_RF_DISABLED to disable WLAN after deep sleep. Probably it is possible to reenable with WAKE_RF_DEFAULT and another deepsleep.

// disable RF after deep-sleep wake up, just like modem sleep, there will be the smallest current.

Its not == modem sleep but you are right. There is no documentation about how to enable modem sleep without deepsleep.

probonopd, 2015-06-22:
One could go to ESP.deepSleep(1, WAKE_RF_DISABLED) and then immediately wake up, but that would require GPIO16 to be tied to RST to wake from deepSleep - on the ESP-01 there is no (easy) way to get to GPIO16 physically

iotmodules, 2015-07-13:
I'v been trying to do the same thing and have looked into the approach of going to sleep and immediately waking up. Trouble is is how do you stop doing it again, and again etc. I've thought about writing something to  EEprom however rather than that is there something I can test to see if WiFi is on or off before going back to sleep?

stonemull, 2015-08-04:
I have tried all the assorted ESP.DeepSleep modes and also tried using the blobs directly with 
wifi_set_sleep_type and  system_deep_sleep_set_option and so far no matter what I do it awakens and draws 64mA or so ..

Does anyone have a confirmed way of obtaining the modem sleep current drain ?

stonemull, 2015-08-06:
Turns out this is only an issue with the staging release and not with the stable release. My apologys.

EUA, 2015-08-25:
```
void loop() {extern "C" {
#include "user_interface.h"
}

#include <ESP8266WiFi.h>

const char* ssid = "****";
const char* pass = "****";

void setup() {
  WiFi.begin(ssid, pass);
  uint8_t i = 0;
  while (WiFi.status() != WL_CONNECTED && i++ < 10) delay(250);

  // put your setup code here, to run once:
  wifi_set_sleep_type(LIGHT_SLEEP_T);
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(10000);
}
```
Sketch consuming ~15mA (rapid spikes between 7-37 mA ) with using staging release.
Got similar results without connecting to router.
That means something use CPU at background. What it is? . :-1:  

PS: Consumption I also include linear voltage regulator (which is smd 662K but that doesn't make huge differences.) .

croisez, 2015-12-14:
Is the call to wifi_set_sleep_type() enough to put the unit in light sleep mode?

stevescot, 2016-01-04:
My understanding - though some confirmation needed through tests is:
setting the mode (Modem-Sleep) sets how the wifi chip handles being on - it will switch off the wifii, and wait for a beacon interval, then switch on to receive it.

Calling WiFi.disconnect(); will switch off the wifi, reducing this power consumption - and I assume we can call WiFi.mode(WIFI_STA) then WiFi.connect() afterwards to get back going again. I believe passing ip parameters will make connect use less power, but am off to test this.

stevescot, 2016-01-04:
and I believe light sleep still maintains wifi connection:

http://bbs.espressif.com/viewtopic.php?t=133

②：During Light-Sleep, the CPU may be suspended in applications like Wi-Fi switch. Without data transmission, the Wi-Fi Modem circuit can be turned off and CPU suspended to save power according to the 802.11 standard (U-APSD). 
E.g. in DTIM3, to maintain a sleep 300ms-wake 3ms cycle to receive AP’s Beacon packages, the current is about 0.9mA.

stevescot, 2016-01-04:
so I am guessing if you enable Light-sleep, then wifi.disconnect() and delay(some long period) then you will get less power use.

shirish47, 2016-01-04:
Yes I tried to find anything related to DTIM in this library but there is nothing ... try looking in SDK functions ...

Links2004, 2016-01-04:
latest git and Staging supports:
```cpp
    WiFi.forceSleepBegin(uint32 sleepUs = 0);
    WiFi.forceSleepWake();
```
wit this functions the WiFi can be forced to sleep.

probonopd, 2016-01-04:
@Links2004 Thank you. Can this be used to disable WLAN entirely, altogether?

Links2004, 2016-01-04:
it will force the chip to modem mode, at least that is what the SDK docs say ;)
if you do some measurement like too see it :)

stevescot, 2016-01-04:
using cheap multimeter, my sketch reads 70mA on what is idle
 (and delay() statements, setting wifi mode to WIFI_OFF, made no difference, still read 70.

with WiFi.forceSleepBegin(2000000) - I had 2 seconds of 20mA :) - so it definitely makes a difference...

stevescot, 2016-01-04:
oh, and I have no voltage regulator, running straight off a 3.7V Li-ion battery.

stevescot, 2016-01-04:
fyi i'm using these functions, and the staging release - Links2004 i see you commited these changes? - Thanks!! 

where wakeUp is called when I want to reconnect, 
and goToSleep is called to reduce power for a number of millisseconds where I can use delay.

(note without calling delay after forceSleep - I still get 70mA, and the code continues to run immediately.)

________________________________________________
void wakeUp()
{
  if(sleep)
  {
    WiFi.forceSleepWake();
    wifi_set_sleep_type(MODEM_SLEEP_T);
    if(WiFi.status() != WL_CONNECTED)
    {
      WiFi.mode(WIFI_STA);
      WiFi.begin(esid,epass);
    }
    sleep=false;
  }
}

void goToSleep(unsigned int ms)
{
  if(!sleep)
  {
    sleep = true;
    thisclient.stop();//this is my wifi client connection - needs to be stopped before going to sleep mode.
    delay(1000); //make sure the tcp connection is stopped
    wifi_set_sleep_type(LIGHT_SLEEP_T);
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);    
  }
  Serial.println("sleeping for");
  Serial.println(ms);
  WiFi.forceSleepBegin(ms*1000);
  delay(ms);
}

Light sleep doesn't seem to do anything, 
The Light-Sleep is supposed to require a pin interrupt to switch it on again, 

gpio_pin_wakeup_enable(uint32 i, GPIO_INT_TYPE intr_state)

I can't get that to compile - 

http://bbs.espressif.com/viewtopic.php?t=171

Links2004, 2016-01-05:
forceSleepBegin will only disable the WiFi not the "cpu core",
can you try a call ```yield();``` instead of the ```delay``` may the SDK needs a yield.
Light-Sleep will disable the CPU too, no wake by code, only gpio 0 or 2 not sure which it where.
not sure how the SDK reacts when you try to use a WiFi Funkion affter forceSleepBegin may its reactivate the WiFi.

Links2004, 2016-01-05:
found it:
[image]
[image]

probonopd, 2016-01-05:
70mA is not good, 20mA is what we should be looking for in Modem-Sleep (see the first post in this thread).

stevescot, 2016-01-05:
EUA - could you try calling WiFi.forceSleepBegin(26843455) after setting Light sleep mode - that should put the chip into true light-sleep and will require a pin interrupt to wake it, that should hopefully reduce the power to 0.5mA

Links2004 - thanks for your help - I'm guessing from my results that my wifi chip is not going to sleep between beacon intervals as it only drops to 20mA (modem sleep) when I force sleep - I thought that would happen normally when the wifi chip is waiting between beacon intervals.

stevescot, 2016-01-05:
ps. I should have said - I get 70mA when I haven't called delay() but have forced sleep. - but when I call delay() after having forced sleep - I get 20mA, so modem sleep is working with this code, i'm guessing delay does yield. - I'm going to try Light-Sleep with interrupt tonight.

Links2004, 2016-01-05:
note: "WiFi.forceSleepBegin(0);``` or ```WiFi.forceSleepBegin();``` will end up calling ```wifi_fpm_do_sleep``` with 0xFFFFFFF for easier usage.
yes delay calling yield too, but stops you sketch for a time too, yield only get the SDK tasks some time to work and "returns" then, most much faster then 1ms. so you can still do stuff but in the forceSleepBegin no WiFi stuff.

stevescot, 2016-01-05:
Thanks for that Links2004 - 

To clarify - when I call delay(2000) - I only get 2 seconds of 20mA - then it returns to 70mA, as soon as my sketch continues,  so it seems that only when the chip is idle, not doing any CPU processing, does the sleep actually make a difference to the power use?

Links2004, 2016-01-05:
the Modem sleep disables the WiFi the CPU is still up an running.
may the Modem sleep is exited when you try to use WiFi function (espressif done give a not about what happend if you do),
but in theory the 20mA shut be there also when your sketch is doing non WiFi stuff.
only in LIGHT_SLEEP the CPU is disabled/suspend too.

if you want to use light sleep take a look at the implementation here:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L264-L300

note: wifi_set_sleep_type has noting to do whit the forced sleep.

probonopd, 2016-01-05:
I could imagine many use cases where the sketch is doing CPU work 24/7 but only needs WiFi periodically, e.g., to post some data every hour. I would hope to see no more than 20 mA for the majority of the time.  If this is possible with Modem-Sleep, it would be cool if we could add a sample sketch for it to the collection of sample sketches.

Links2004, 2016-01-05:
yes this is the concept around Modem sleep, in theory that exact what it designed for.
the usage shut be easy.

when you want to sleep:
```cpp
WiFi.forceSleepBegin();
```

and to enable WiFi again:
```cpp
WiFi.forceSleepWake();
```

be sure to not call and WiFi/network functions when you in sleep, and thats it.

but it need testing and measuring, I have implement the functions based on the documentation of espressif, but not done any real tests yet.

stevescot, 2016-01-05:
Tests look good - you are right  - drops to 20mA permanently without a time or time of 0 passed in allowing the wake to be called after. 

Great job links - I've not managed to get light sleep working yet but still trying.

So forced light sleep mode works.

stevescot, 2016-01-05:
Sorry forced modem sleep works. Still trying on light sleep.

antelder, 2016-01-11:
I've been playing with the forced sleep modes but not been able to get forced light sleep to work yet either. 

Using the 2.1.0-rc1 code and a skecth based on the 1.5.1 SDK Guide example on page 91 (http://bbs.espressif.com/download/file.php?id=1044) but it doesn't seem to make any difference between calling wifi_fpm_set_sleep_type with MODEM_SLEEP_T or LIGHT_SLEEP_T and it always just does a modem sleep and so uses 15mA.
```
extern "C" {
  #include "user_interface.h"
}

void setup() {
  Serial.begin(115200);
  Serial.println();
}

void loop() {
  Serial.println("Go to light sleep, 10 secs");
  wifi_station_disconnect();
  wifi_set_opmode(NULL_MODE); 
  wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // LIGHT or MODEM doesn't seem to make a difference
  wifi_fpm_open();
  Serial.print("wifi_fpm_get_sleep_type: "); Serial.println(wifi_fpm_get_sleep_type());

  int resp = wifi_fpm_do_sleep(10000000); // 10 secs
  Serial.print("wifi_fpm_do_sleep resp: "); Serial.println(resp);
  delay(20000);  // 20 secs, so should light sleep for just the first 10 secs then go to full power

  Serial.println("delay 10 secs..."); // Now delay using full power
  delay(10000);
}
```

stevescot, 2016-01-15:
FYI - here's an arduino sketch you can use on an UNO to get power usage at 20ms resolution:
if you use a 10 Ohm resistor in series with the power to the esp8266 then attach ground of the arduino to one side and A1 to the other
then run it and copy/paste the results into excel you can get charts, 

output looks like this:

Internal 1.1V reference
Minimum Current measureable: 0.11mA
ms	I(mA)	mAh
9917	0.75	0.00
9934	14.72	0.00
9952	20.09	0.00
9969	30.72	0.00
9987	35.99	0.00
10004	27.18	0.00
10022	26.43	0.00
10040	28.57	0.00
10057	23.96	0.00
10075	21.27	0.00
10092	20.30	0.00
10109	19.87	0.00
10136	19.44	0.00
10174	18.80	0.00
10195	19.23	0.00
10213	18.80	0.00
11176	19.23	0.01
11210	18.80	0.01
11246	19.98	0.01
11264	19.55	0.01
11282	19.12	0.01

[image]

[currentMeasurement.txt](https://github.com/esp8266/Arduino/files/92382/currentMeasurement.txt)

stevescot, 2016-01-15:
incase the text file doesn't work:

//Capture full defintion data
      
      const int currentPin = 0;                    
      const int voltagePin = 1;
      float referenceVoltage = 5;//50 ohm resistor = 
      float previousCurrent = 0;
      float minCurrentMeasureable = 0;
      unsigned long lastMicros =0;
      int Voltage = 0;
      float resistorOhms = 10;
      unsigned long minimumCurrentmilliseconds = 0;
      
      int preva = -9999999;
      int a;
      
      void setup() {
        // put your setup code here, to run once:
        Serial.begin(250000);
        analogReference(DEFAULT);
        referenceVoltage = 5;
        Voltage = getVoltage(voltagePin);
        analogReference(INTERNAL);
        referenceVoltage = 1.1;
        Serial.println("Internal 1.1V reference");
        minCurrentMeasureable = currentFromVoltsAndOhms(divisionsToVoltage(1), resistorOhms);
        Serial.print("Minimum Current measureable: ");
        Serial.print(minCurrentMeasureable *1000);
        Serial.println("mA");
        Serial.println("ms\tI(mA)\tmAh");
      }
      
      
      void loop() {
        unsigned long start = micros();
        float voltsOnResistor = getVoltage(voltagePin);
        float current = currentFromVoltsAndOhms(voltsOnResistor, resistorOhms);
        if(abs(current-previousCurrent) > (minCurrentMeasureable*4))//where the value has changed by more than 10 divisions. 
        {//print it out.
          Serial.print(millis());
          Serial.print("\t");
          Serial.print(current*1000);
          Serial.print("\t");
          minimumCurrentmilliseconds += current / minCurrentMeasureable * (float)(micros()-lastMicros)/1000;
          Serial.println((float)minimumCurrentmilliseconds * minCurrentMeasureable/60/60);
          previousCurrent= current;
          lastMicros = micros();
        }
        delayMicroseconds(1000-(micros()-start));
      }
      
      float currentFromVoltsAndOhms(float Volts, float Ohms)
      {
        return Volts/Ohms;
      }
      
      float getVoltage(int analogPin)
      {
        int aval = analogRead(analogPin);
        float voltage = divisionsToVoltage(aval);
        return voltage;
      }
      
      float divisionsToVoltage(int divisions)
      {
        return (float)divisions * (referenceVoltage/1024);
      }

probonopd, 2016-01-16:
Maybe I am doing something wrong, but this sketch doesn't seem to save any power for me. Can someone please verify
```
/* 
 File -> Examples -> ESP8266 -> Blink
 Modified to call "WiFi.forceSleepBegin();" in setup()
*/

#include <ESP8266WiFi.h>

int ledState = LOW;     

unsigned long previousMillis = 0;
const long interval = 1000;

void setup() {
  pinMode(BUILTIN_LED, OUTPUT);
  WiFi.forceSleepBegin();
}

void loop()
{
  unsigned long currentMillis = millis();
  if(currentMillis - previousMillis >= interval) {
    previousMillis = currentMillis;   
    if (ledState == LOW)
      ledState = HIGH;  // Note that this switches the LED *off*
    else
      ledState = LOW;   // Note that this switches the LED *on*
    digitalWrite(BUILTIN_LED, ledState);
  }
}
```

stevescot, 2016-02-12:
[image]

this is the circuit used to measure current

SensorsIot, 2016-02-25:
I tested the power consumtion with the following simple code:

`void setup() {

  // initialize digital pin 13 as an output.
  pinMode(13, OUTPUT);
  WiFi.mode(WIFI_AP_STA);
}
// the loop function runs over and over again forever
void loop() {

  digitalWrite(13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);  // wait for a second
  WiFi.forceSleepBegin();
  digitalWrite(13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
  WiFi.forceSleepWake();
}`
It reduces the power consumption to 15 mA for one second and then it increases again to 70mA.

This is all without actively using the WiFi connection.

If I do a similar thing with an existing WiFi connection of the HelloServer example:
`void loop(void) {

  yield();
  Serial.println("Begin");
  server.handleClient();
  WiFi.forceSleepBegin();
  delay(1000);              // wait for a second
  WiFi.forceSleepWake();
  delay(100);
  WiFi.begin(ssid, password);
}`

This example does not work, also without the WiFi.begin command. And sometimes, the whole ESP chip is screwed up and only creates dumps. Reset does not help. I have to run another sketch with the command 

ESP.deepSleep(10000000, WAKE_RF_DEFAULT);

in it. Then, the chip is usable again.
Does anybody have a working WiFi connection using the forceSleep commands?

antonmeyer, 2016-03-13:
it is just a guess. I read something about station and ap mode. 
in AP mode power save does'nt work -(I'm not sure about that)
as there ist mode 3 for both ...
may be that is the difference for power saving working and not.

Links2004, 2016-03-13:
yes in AP mode there is no auto ```modem sleep``` or auto ```light sleep``` possible.
you sill can go manually too ```light sleep``` (disable wifi, cpu hold) or ```deep sleep``` (disable cpu).

danielcbit, 2016-08-24:
Hi @stevescot, regarding your sketch and circuit to mesure the current usage of ESP8266, it is correct to power the ESP8266 on through GND and VIN using a 3.7V battery?

Shouldn't it be through GND 3V3?

Also, how would I could do the wiring using the 5V ou the 3v3 output from UNO to power the ESP?

Thank you!

drmpf, 2016-09-11:
The code I posted in https://github.com/esp8266/Arduino/issues/644 seems to work for me using AP mode.

ukrsms, 2016-12-05:
> I've been playing with the forced sleep modes but not been able to get forced light sleep to work yet either.
>
> Using the 2.1.0-rc1 code and a skecth based on the 1.5.1 SDK Guide example on page 91 [...]
I have tried for a day to launch LightSLEEP on ESP and have concluded that if I specified time of sleep (wifi_fpm_do_sleep(TIME<0xFFFFFFF)) then MCU doesnt suspend as it would be in LIGHT_SLEEP, but if I specified sleep time as 0xFFFFFFF then MCU suspend.
Could anyone tell me if there is a possibility to provide LIGHT_SLEEP for a specified time during which MCU would suspend?

nseidle, 2016-12-08:
I stumbled across this thread because I didn't want to use the normal [RST to XPD](https://learn.sparkfun.com/tutorials/esp8266-thing-development-board-hookup-guide/all#hardware-overview) trick to put my Thing Dev board to sleep.

I cobbled together a working sketch based on this thread. Works well for me. Unit runs at ~70mA with WiFi on, then sleeps at 14.7mA for 6 seconds, then goes back to 70mA (connects to WiFi), then goes to sleep, repeat... 

```c
#include <ESP8266WiFi.h>
#include "credentials.h" //Contains WiFi SSID and password

const int sleepTime = 6; //Power down WiFi for 6 seconds

void setup()
{
  Serial.begin(115200);
}

void loop()
{
  WiFi.begin(ssid, pass); //Connect to local Wifi

  Serial.println();
  Serial.print("Connecting to WiFi");
  while(WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.println("WiFi Connected!");

  //Do your thing.

  //Time to let the WiFi go to sleep!
  Serial.println("Sleeping...");
  WiFi.disconnect();
  WiFi.mode(WIFI_OFF);
  WiFi.forceSleepBegin(sleepTime * 1000000L); //In uS. Must be same length as your delay
  delay(sleepTime * 1000); //Hang out at 15mA for 6 seconds
  WiFi.mode(WIFI_STA);
  Serial.println("Awake!");
}
```

Thing Dev Board is powered with 3.7V LiPo into the onboard 3.3V regulator. Power LED disabled.

I'm happy to have gotten the ESP8266 into Modem-Sleep. Let me know if it works for you or if there are ways to simplify it even more.

ukrsms, 2016-12-09:
Modem sleep with its 15mA is to much for me. Our device is supplied by 2500mAh battery have to work for a week performing some additionals power opertions, so I need Light sleep to lower consumption for at least to 1..3mA. At the moment I can provide Light sleep but only in a case:
`
wifi_fpm_set_sleep_type(LIGHT_SLEEP_T);
wifi_fpm_open();
gpio_pin_wakeup_enable(GPIO_NUM_OPENBUTTON, gpio_input_getMy(GPIO_NUM_ENC_A)? GPIO_PIN_INTR_LOLEVEL: GPIO_PIN_INTR_HILEVEL);	
gpio_pin_wakeup_enable(GPIO_NUM_ENC_A, 		gpio_input_getMy(GPIO_NUM_ENC_A)? GPIO_PIN_INTR_LOLEVEL: GPIO_PIN_INTR_HILEVEL);
wifi_fpm_set_wakeup_cb(System_sleep_wakup_cb);   // Set wakeup callback
wifi_fpm_do_sleep(0xFFFFFFF);
	`

but in such a case device cannot be awaken after some time, and if I try to specifiy time, e.g.
`wifi_fpm_do_sleep(10000000);`
then device goes to ModemSleep, not to Light sleep

nseidle, 2016-12-09:
@ukrsms - You can put the ESP into deep sleep (80uA) with the `system_deep_sleep([uint32_t time_in_us])` command but you need to have XPD connected to RST. 

> After you tell the ESP8266 to sleep, it’ll wait a specified number of microseconds, then trigger the XPD pin to toggle the reset line. When the ESP8266 wakes up, it’ll begin back at the start of the sketch. ([source](https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/example-sketch-goodnight-thing-sleep-mode))

Pros to deep sleep: 
* Very low power

Cons to deep sleep:
* You loose access to pin GPIO16 (aka XPD)
* With GPIO16 connected to RST you also can't load new code via the serial bootloader. You have to mechanically disconnect XPD from RST ([we recommend a jumper](https://cdn.sparkfun.com/r/600-600/assets/learn_tutorials/4/4/5/esp8266-jumper.jpg) but I find this painfully tedious). 
* System does a full reset (doesn't start running the code after system_deep_sleep() command)

ukrsms, 2016-12-13:
Thank you, @nseidle, but I need to keep RAM and MCU ready to start immediately after external Interrupt occurred, so DeepSleep is not the solution

vlast3k, 2016-12-13:
@nseidle just a remark about 

> With GPIO16 connected to RST you also can't load new code via the serial bootloader

In fact if you reset the board not via RST but via CH_PD this is avoided. If i remember correctly this is how it is officially suggested by Esspresif. I do not know how it happened that most of the dev boards nowadays connected RST to GPIO16 (it may be as well be that i missed some potential problem). I am using this in my boards since several months and it works.
Another benefit is that there is an internal pullup on RST pin, in addition to the 0.1uF cap that is usually placed there for stability. So sometimes the "nodemcu" circuit (with the two transistors) has issues in resetting the board. So reseting via CH_PD is more consistent

nseidle, 2016-12-13:
>but I need to keep RAM and MCU ready to start immediately after external Interrupt occurred, so DeepSleep is not the solution

Ah, makes sense.

> In fact if you reset the board not via RST but via CH_PD this is avoided. 

Have you tested this? If not be cautious with statement of facts. I ask because Espressif is notorious for their poor documentation. 

Most recent datasheet for ESP8266 I could find is [here](http://espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf). `CH_PD` is no longer a pin. I believe it changed to `CHIP_EN`, and is now called `CHIP_PU`.

From datasheet: 

> XPD_DCDC I/O Deep-sleep wakeup (needs to be connected to EXT_RSTB); GPIO16

So I *think*, and have tested (with our hardware) that to exit deep sleep GPIO16 (aka XPD_DCDC) needs to be connected to EXT_RSTB (aka RST). 

I like your idea of using the CHIP_PU chip enable pin but I'm not sure what the behavior will be when you enter deep sleep and then use XPD to then pull CHIP_PU high again. If you can test this I'd love to hear what you find.

vlast3k, 2016-12-14:
@nseidle 
Actually i use it quite extensively since half an year as part of this project
https://hackaday.io/project/12063-vesprino-esp8266-dongle
and there are already more than 100 devices online working w/o complaints. (i make some environmental monitors)
In fact i read about this first in this blog https://hallard.me/esp8266-autoreset/, which is owned by @hallard, who actually did all the research (Section: Google around on reset function)
I needed the feature, because i wanted to have a device that would go to deepsleep to avoid heat emission, and would be programmable over serial port, so that customers can flash it on their own in case OTA Update fails.

While the device is in deepsleep it is possible to trigger CHIP_PU down (connected to Serial RTS) and then UP and the device reboots. I also use this quite often, as my device goes to sleep after a minute of operation and does no longer accept serial commands. So i just set RTS to LOW and then HIGH and the device reboots. The RTC memory is also cleared in this case (which doesn't happen always when calling ESP.reset() (or.reboot() forgot the exact name))

ufukv, 2017-10-04:
Hi dear friends,
I just started writing programs. I am using the STM32F0 Discovery DK (STM32F051R8) and communicated with UART. I using KEIL IDE. I made a button-led application with a server control but, the module consumes 116mA  (with Discovery). How can I use the Deep sleep mode of the Wifi module? Arduino has ESP.deepSleep() function in "ESP8266WiFi.h" file. How can I write a similar or same function in the KEIL IDE ?
Thanks for your valuable feedback :)

probonopd, 2017-10-04:
Dear @ufukv, you realize that this might be the wrong forum for your question. STM32F051R8 is a completely different hardware and software from what is discussed here.

devyte, 2017-10-20:
Current API has most of the sleep functionality. Only one that's a bit unclear is light sleep, which is tracked elsewhere.
Closing.

closed 2017-10-20 by devyte

Pablo2048, 2017-10-20:
Yes, remove connection between GPIO16 and RESET and wake-up by pulling RESET. This has been discussed in the forum so please do some search on esp8266.com forum...