esp8266/Arduino #462: How to Download Release 1.6.5?

rei-vilo, 2015-06-21:
There have been important changes ultimately but the available package is 1.6.1-esp8266-1 dated Mar 27, 2015. The [Releases page](https://github.com/esp8266/Arduino/releases) also mentions that there are 1070 commits to esp8266 since this release.

Do you plan to make release 1.6.5 available through the [Releases page](https://github.com/esp8266/Arduino/releases) or is the **Tools > Board > Boards Manager** with `http://arduino.esp8266.com/package_esp8266com_index.json` the new official way of getting the ESP8266 plug-in?

I've downloaded the source code and built my own Arduino for ESP8266 using `ant`. Problem is, other users have different versions of the Arduino for ESP8266, and we don't know how to identify them.

Thank you!

igrr, 2015-06-21:
Boards manager is now the official way to install ESP8266 support into
Arduino.

You can always build from git if you want latest changes, but this is the
way mostly for people who develop the core itself or libraries.

There is some confusion with version numbers because I have accidentally
pushed 1.6.5 tag from upstream Arduino repository here.

scryb3, 2015-06-22:
Would it be possible to do a latest/nightly board manager URL for those
that want to live on the edge? i.e.
http://arduino_dev.esp8266.com/package_esp8266com_index.json

reaper7, 2015-06-22:
+1 for "latest/nightly"

igrr, 2015-06-22:
Related: #324 (there is a somewhat fresher version of the boards manager package).
The reason I don't want to update the staging build every night is that the latest version git is very frequently broken — and I don't want this issue tracker to explode from issue reports related to the nightly builds.

mvdbro, 2015-06-25:
But could we have something like a monthly build? Or what is the currently proposed deployment schedule?

As far as i can see in my Arduino IDE (1.6.4) there have been two packages up until now:

1.6.4-628-g545ffde	dated may 19th
1.6.4-673-g8cd3697 dated may 22nd

Or is my IDE missing something here, should I pull/reset or otherwise do something to get a more recent build?

Other question, the Adafruit json link that i've tried before gave the option to select between different builds. Will that option be there when a newer build is available? Or will the IDE automatically select the most recent version?

Testato, 2015-06-26:
there is only one tagged build for now, so you on Board Manager have only one choice.
When @igrr will tag and push a new version you can update via board manager.

igrr, 2015-07-23:
Stable package has been updated to 1.6.5-947-g39819f0.

closed 2015-07-23 by igrr