esp8266/Arduino #463: library Servo claims to run on [avr, sam] architecture(s)

deshipu, 2015-06-21:
When trying to compile the Sweep example of the Servo library, the IDE shows:

> WARNING: library Servo claims to run on [avr, sam] architecture(s) and may be incompatible with your 
> current board which runs on [esp8266] architecture(s).
> In file included from xs.ino:1:0:
> /home/sheep/Stuff/sketchbook/libraries/Servo/src/Servo.h:67:2: error: #error "This library only supports 
> boards with an AVR or SAM processor."
> #error "This library only supports boards with an AVR or SAM processor."

While the readme claims that the servo library works and supports up to 24 servos.

igrr, 2015-06-21:
I suppose you are using a version which doesn't have that supported yet... See discussion near the end of https://github.com/esp8266/Arduino/issues/8.

deshipu, 2015-06-21:
Well, I'm using the version from the tip of the repository, which claims in its readme that it's supported. Either the readme is wrong, and it's not supported yet, or it supported, but the metadata/whatever is wrong. In either case, it's a bug.

If you look at the current tip, you will see that there is no esp8266 support: https://github.com/esp8266/Arduino/tree/esp8266-sdk-1.0/libraries/Servo

So the readme simply lies.

igrr, 2015-06-21:
The library is there, it's just in the other folder:
https://github.com/esp8266/Arduino/tree/esp8266-sdk-1.0/hardware/esp8266com/esp8266/libraries/Servo

I think the issue is that when we deploy the board manager package, the correct Servo library (under ```hardware/esp8266com/esp8266```) is used by the IDE. But when you build from git, IDE picks the library under ```libraries/``` first. I'm not sure how to better fix this? Remove the original Servo library from libraries/ perhaps?

deshipu, 2015-06-21:
Ah, I'm sorry, seem like I don't have the latest version after all.

closed 2015-06-21 by deshipu