esp8266/Arduino #464: telnet server does not work on its own softapip '192.168.4.1' when i disconnect my wifi modem.telnet server work fine with external wifi modem and also give response on its apIP.

nsshalin, 2015-06-23:


omersiar, 2015-06-25:
WiFiTelnetToSerial example is working as STAtion mode (WiFi Client) and requires a connection to a in working state WiFi AP in order to get an IP address from router. It won't have an IP address unless it connects to an AP. If you want it in AP mode (WiFi Server) you can try below snippet. I'm testing it right now and need some feedback.

https://gist.github.com/omersiar/f67a56e0c0da360b87c7

nsshalin, 2015-06-28:
omersiar,above code is working very well.i have tasted it. thank u for your kind reply.but i have a question that if i use this code "while (WiFi.status() != WL_CONNECTED) { delay(500); Serial.print("."); }" then does not connect and print only "." after every 500 mil second.when i removed it then telnet server started normally.why?

Links2004, 2015-11-27:
```WiFi.status()``` is for STA not for AP mode.

closed 2015-11-27 by Links2004