esp8266/Arduino #466: Error: espcomm_open failed

srishtisingh08, 2015-06-23:
I am using Arduino 1.6.5 along with ESP8266MOD (16 Pin).
Connections are: RX-TX, TX-RX, GND-GND, VCC,GPIO0 and CH_PD- 3.3v arduino..
no error in compilation of the code.
But while uploading a warning and an error occurs:

warning: espcomm_sync failed
error: espcomm_open failed

Tried changing RX and TX, and also GPIO0 while ploading kept it low.

code:
void setup() {
 
Serial.begin(9600);
Serial.begin(9600);
delay(1000);
}

void loop() {
 
if(Serial.available()){
  byte b = Serial.read();
  Serial.write(b);
}
if(Serial.available()) {
  byte c = Serial.read();
  Serial.write(c);
}
}

Please help me out!!!

drmpf, 2015-06-23:
I got that recently on Adafruit HUZZAH ESP8266 module and found I had the TX / RX leads swapped over. I fixed the connections and the problem went away.

srishtisingh08, 2015-06-23:
I tried. But didnt work for me.

Links2004, 2015-06-24:
here is the list what need to be connected and where the pullups shut be:
https://github.com/esp8266/Arduino#minimal-hardware-setup-for-bootloading-and-usage

AmrKafina, 2015-06-25:
The ESP-12 chip exposes GPIO 15. This pin must be pulled to ground for the chip to operate in the flash or UART modes.

VictorLavie, 2015-07-26:
I had the same issues, the same errors with an ESP201 board. I understood it's close to the ESP-12. I connected the GPIO15 to Ground and it began uploading the sketch. After the uploading finished I disconnected the GPIO0 and the GPIO15 and the module functioned OK. So it's depending on the board used...My board in the picture attached.
[image]

VictorLavie, 2015-07-26:
Sorry, I have to correct myself: You have to leave connected the GPIO15 to the Ground to make the board functioning after the upload. Only the GPIO0 is to be disconnected.

albertbooey71, 2015-08-14:
Still having the same issues but with the standard ESP8266. Tried swapping the TX/RX, grounded GPIO0, and even tried updating the firmware but not successful. Here's the link of the test

http://iot-playground.com/2-uncategorised/35-esp8266-firmware-update

Hope for any help!

jrperin, 2015-10-20:
Hi. I was having the same problem.
I was uploading the code by the USB. In this case, my USB hasn't had enough power to provide to ESP8266.
After install a external 3.3v power supplier, my problem has solved.

klimovitsky, 2015-11-10:
From Adafruit's tutorial, using the Huzzah board, did you try:
1. Hold down the GPIO0 button, the red LED will be lit
2. While holding down GPIO0, click the RESET button
3. Release RESET, then release GPIO0
4. When you release the RESET button, the red LED will be lit dimly, this means its ready to bootload

Then in the upload.

https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide
Hope it might help.

stephenmueller, 2015-11-21:
I was having the same issue as well. The way the issue was fixed for me is by connecting GPIO2 to 3.3V and GPIO0 to ground then resetting before uploading. GPIO2 can be disconnected from 3.3V after the upload is finished.

Wifsimster, 2016-01-11:
@stephenmueller You are a life saver man !

hakeemta, 2016-01-30:
I got an ESP8266MOD based WiFi Shield made by SZDOIT and it's meant to be a transparent Wifi-serial bridge. Although, the documentation reveals that the official AT or NodeMCU firmware can be loaded.

So, I have been trying to flash the module on the board with an arduino sketch and I always had this error:

Warning: espcomm_sync failed
Error: espcomm_open failed

I'm using Arduino IDE 1.6.5
esp8266/Arduino 1.6.5

klimovitsky, 2016-02-04:
It seems to be a repeated issue. I believe that all the ESP8266 modules with whatever board they are mounted on - all have to be configured into the "bootload mode". The ESP8266 HAS TO BE in the "bootload mode" before each upload/flash. There is no timeout for bootload mode, so you don't have to rush!
Look in this link for example, ([HUZZAH ESP8266](https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide)) - In order to enter the "bootload mode" it is required to: 
1. Hold down the **GPIO0 **button, the red LED will be lit
2. While holding down **GPIO0**, click the **RESET **button
3. Release **RESET**, then release **GPIO0**
4. When you release the **RESET **button, the red LED will be lit dimly, this means its ready to bootload

Now, reading the [Adafruit HUZZAH ESP8266 Breakout](https://learn.adafruit.com/downloads/pdf/adafruit-huzzah-esp8266-breakout.pdf) document will tell you that:

**GPIO#0**, which does not have an internal pullup, and is also connected to both a mini tactile switch and red LED. This pin is **used by the ESP8266 to determine when to boot into the bootloader**. If the pin is held low during power-up it will start bootloading! That said, you can always use it as an output, and blink the red LED.

And

**RESET **- this is the reset pin for the ESP8266, pulled high by default. When pulled down to
ground momentarily it will **reset the ESP8266 system**. This pin is 5V compliant

Therefore, I am pretty sure that if you want to program/flash/upload to the ESP8266 you will have to hold LOW the GPIO0 pin of the ESP8266 module you have, then hold LOW the RESET pin of the ESP8266 module you have in order to clear the ESP8266. Then set the RESET pin HIGH, and then set the GPIO0 pin HIGH. Following those 4 steps will cause the ESP8266 chip to enter into bootloading mode. At this point you should be able to upload any program/sketch you want. 

I could not find any schematic for the SZDOIT's ESP8266MOD board so I am not sure what is the pinout. But try to follow the above?

hakeemta, 2016-02-04:
Thanks so much, Klimovitsky. I'll try that again and get back.

Also, I want to know if there's a particular baudrate for flashing the chip. Although, in the previous discrete ones I have used 115200 was cool.

klimovitsky, 2016-02-04:
Yeah, that is what I used as well.

akeroyd, 2016-03-02:
I had the same espcomm_sync failed problems with an Adafruit Huzzah, but I think mine is caused by a cheap FTDI adapter with a counterfeit FTDI chip.
When I connected via PuTTy and press reset a few times I see:
NON GENUINE DEVICE FOUND! NON GENUINE DEVICE FOUND!

This is something FTDI put in the newest Windows drivers to block communication with counterfeit chips.
http://hackaday.com/2016/02/01/ftdi-drivers-break-fake-chips-again/

Spackstor, 2016-03-08:
Not having a common logic ground can cause this. If you are running your FTDI from a 5 Volts source and your ESP82 from a separate 3 volt source, connect the grounds.

hzd1977, 2016-03-30:
So to be totally newbie to this.
I have the feather wifi esp8266 from adafruit. I have found a reset switch, but I will have to connect a switch to the GPIO#0 and GND (or 3V???) in order to put it into bootload mode?

krzychb, 2016-03-30:
@hzd1977,

Using Adafruit Feather HUZZAH ESP8266 you do not need to worry about toggling individual GPIO pins in order to put ESP into bootloader mode. This is done for you by the sketch uploader program (esptool) over USB/serial interface.

If not done already just follow Adafruit's guide https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/using-arduino-ide

Krzysztof

hzd1977, 2016-03-30:
Yeah, I have followed the adafruit guide, step by step.
I'll try adding a battery for more power.

I also suspect the usb driver I was told to install. I have the .dmg file and upon double clicking it, nothing seems to happen. Usually there apears a .pkg or something like that.

hzd1977, 2016-03-30:
Now looking closer at my serial ports on the arduino IDE drop down menu, I see no usb ports.

hzd1977, 2016-03-30:
[https://youtu.be/T_vh4H4NdCU](url)

Here is a short video showing what happens when I connect the board. The lights flash and turn of.

hzd1977, 2016-03-30:
https://youtu.be/hgvoErjzIAU

There's a link to a 5 sec showing the lights light up and turn of upon connecting.

hzd1977, 2016-04-01:
I stand before you, with my head bowed in shame.
When executing the .dmg file I was expecting a .pkg file to pop up and did not have the mental resource to look for it in my finder window. I now found seven disk images in finder with each .pkg file.
Every thing works fine.

krzychb, 2016-04-01:
@hzd1977
Thank you for update. I think that everybody following this repository is happy to hear back that the issue is resolved and a reasonable root cause identified :smile:
[image]

tscha70, 2016-04-30:
Remove the LED on GPIO2!!!!

I spent hours to find out that my LED was the issue!
Removing it or better replacing it by a good quality LED will help.

tibfox, 2016-08-04:
Trying a different usb cable works for me .. Lucky i didn't removed anything or added a jumper wire or something else. That site contains some tips for that issue: https://learn.sparkfun.com/tutorials/esp8266-thing-hookup-guide/discuss

Vedant803, 2017-02-25:
ly i can select & using COM3. Which is only it's appearing.)
Sketch uses 522510 bytes (50%) of program storage space. Maximum is 1044464 bytes.
Global variables use 51560 bytes (62%) of dynamic memory, leaving 30360 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_send_command: wrong direction/command: 0x00 0x08, expected 0x01 0x08
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_open failed
I did Everything right just when I reached at this uploading part the problem occurred.
I have this chip -> https://www.amazon.com/HiLetgo-Version-NodeMCU-Internet-Development/dp/B010O1G1ES/ref=sr_1_3?ie=UTF8&qid=1487921045&sr=8-3&keywords=ESP8266
2nd. I'm not a Electrician nor anything and I don't even have stuffs to do Advance check.
I explained My exact problem,
hope you guys will help...

vicnevicne, 2017-02-25:
Hi Vedant803,
You already opened bug #2992 and #2994 about the same topic. Polluting every other issue with a copy/paste of your problem is just wasting developer's time and introducing confusion. It's not going to help you.
This system is meant to track bugs in the Arduino environment for ESP8266, but unless you prove otherwise, the problem is most probably on your side.
If you want quick response from ESP8266 enthusiasts, I advise you to post this question in the user forums at esp8266.com . If not, please at least remain focused on bug #2992 and expose what you have already tried to solve your problem...

arthurmoises, 2017-04-03:
The thing that worked for me was to set baud rate to 115200 (was 921600 by default when installing Arduino,(don't know why, by the way)).

aqualaguna, 2017-05-26:
These are reaaaaally annoying. I got the same problem and guess what?
the connection between esp12 in pin tx is not connected(or loose) to the circuit board. got some lead and solder and its solved. damn it, waste 5 hours for this problem

Gijsrongen, 2017-06-15:
Use this:
[https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/faq/a01-espcomm_sync-failed.md](url)

devyte, 2017-10-20:
Closing per #3655 .

closed 2017-10-20 by devyte

sahanwa, 2017-12-06:
ESP01 Flash config:-->
Flash real size: 1048576 1MB 512SPIFFS
Flash ide speed: 40000000 - 4MHz
Flash ide mode:  DOUT

ESP07 Flash config:-->
Flash real size: 1048576 1MB 512SPIFFS
Flash ide speed: 40000000 - 4MHz
Flash ide mode:  QIO

When programming the ESP8266, connect VCC, ground, TX and RX properly. (FTDI <->ESP)

VCC --> 3.3V 
GND --> GND
TX -->RX
RX --> TX

(Power from USB 2.0 was enough to program ESP01 and ESP07; I used arduino software to program; Nodemcu flasher can also be used to flash)

In order to turn into program the ESP, first both the RESET and the GPIO0 should be connected to the ground.
Then keep it for around 5 seconds
Then remove the RESET BUT do not remove GPIO0.

shoebeee, 2018-05-01:
I'm using ESP8266 Wifi Module ESP-03. 
getting similar error -
```
Sketch uses 260040 bytes (25%) of program storage space. Maximum is 1023984 bytes.
Global variables use 33276 bytes (40%) of dynamic memory, leaving 48644 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

```
Picture of the board is here - [https://cdn.instructables.com/FTH/XRX4/IRHV3YPI/FTHXRX4IRHV3YPI.LARGE.jpg](url)

arock309, 2018-05-13:
> 

> 

-   ##[](url)

bhrigs, 2018-06-12:
Changing the USB cable solved it for me as well. Same thing helped others according to this thread:
https://forum.arduino.cc/index.php?topic=404779.0

guyluz11, 2021-07-09:
Plugging the cable each time before upload work for me