esp8266/Arduino #467: client.read() waits for \r before processing

grtyvr, 2015-06-24:
I am trying to implement a TCP server that listens for a char command.  It seems like the client.read() will only return when it gets a new line.  Here is a code snippet 

void loop() {
  WiFiClient thisClient = server.available();
  while (thisClient) {
    if (!alreadyConnected) {
      alreadyConnected = true;
    }
    if (thisClient.connected()) {
      if (thisClient.available() > 0) {
        char thisChar = thisClient.read();
        if (thisChar == 'Q') {
          // do something usefull
          // but this will not get tested unles there is a newline in the stream
        }
      }
    } 
    else {
      thisClient.stop();
      alreadyConnected = false;
    }
}

Is client.read supposed to wait for a newline?

igrr, 2015-06-24:
It shouldn't, have you verified that your TCP client sends every character
individually (instead of waiting for newline as netcat does, for instance)?

grtyvr, 2015-06-24:
I have not tried that.  Can you suggest an alternative to netcat for testing?

grtyvr, 2015-06-24:
Or it might be that client.read has to time out first.

grtyvr, 2015-06-24:
No.  I think it is an error on what I am sending back to the client and what it is expecting in response.  Sorry to bother you.

igrr, 2015-06-24:
I would suggest you to try a tool like tcpdump — that will save you some
debugging time.

closed 2015-06-24 by grtyvr