esp8266/Arduino #470: Can't read HMC5883l data correctly using I2C

leandrogs, 2015-06-25:
I'm using HMC5883l to read earth's magnetic field. Testing the same library and same code on Arduino Uno it gives correct output, but running on ESP give me wrong values.

For example:
Arduino reading: -123.25 | 9.13 | -35.67
ESP reading: -123.00 | 9.00 | -35.00
I cannot read decimal values on ESP.

Also, sometimes it puts the number 65 at the beginning of negavite numbers like  -65123.00 | 9.00 | -6535.00

Here follows my code:

```
#include <Wire.h>
#include <HMC5883L.h>

HMC5883L magnetometer;
MagnetometerScaled scaledVals;

void setup() {
    Serial.begin(115200);
    Wire.setClock(450000);
    Wire.pins(0, 2);
    Wire.begin();
    magnetometer = HMC5883L();
    magnetometer.SetScale(1.3);
    magnetometer.SetMeasurementMode(Measurement_Continuous);
}

void loop() {
    scaledVals = magnetometer.ReadScaledAxis();
    printScaled(scaledVals);
}

void printScaled(MagnetometerScaled scaled)
{
    Serial.print(scaled.XAxis);
    Serial.print(" | ");
    Serial.print(scaled.YAxis);
    Serial.print(" | ");
    Serial.print(scaled.ZAxis);
    Serial.print("\n");
}
```

Should I post library code?

me-no-dev, 2015-06-25:
yes :) the library is the reason. Point to the code and I'll give you the fix

leandrogs, 2015-07-07:
HMC5883L.cpp:
```
#include <Arduino.h> 
#include "HMC5883L.h"

HMC5883L::HMC5883L()
{
  m_Scale = 1;
}

MagnetometerRaw HMC5883L::ReadRawAxis()
{
  const uint8_t* buffer = Read(DataRegisterBegin, 6);
  MagnetometerRaw raw = MagnetometerRaw();
  raw.XAxis = (buffer[0] << 8) | buffer[1];
  raw.ZAxis = (buffer[2] << 8) | buffer[3];
  raw.YAxis = (buffer[4] << 8) | buffer[5];
  delete [] buffer;
  return raw;
}

MagnetometerScaled HMC5883L::ReadScaledAxis()
{
  MagnetometerRaw raw = ReadRawAxis();
  MagnetometerScaled scaled = MagnetometerScaled();
  scaled.XAxis = raw.XAxis * m_Scale;
  scaled.ZAxis = raw.ZAxis * m_Scale;
  scaled.YAxis = raw.YAxis * m_Scale;
  return scaled;
}

int HMC5883L::SetScale(float gauss)
{
  uint8_t regValue = 0x00;
  if(gauss == 0.88)
  {
    regValue = 0x00;
    m_Scale = 0.73;
  }
  else if(gauss == 1.3)
  {
    regValue = 0x01;
    m_Scale = 0.92;
  }
  else if(gauss == 1.9)
  {
    regValue = 0x02;
    m_Scale = 1.22;
  }
  else if(gauss == 2.5)
  {
    regValue = 0x03;
    m_Scale = 1.52;
  }
  else if(gauss == 4.0)
  {
    regValue = 0x04;
    m_Scale = 2.27;
  }
  else if(gauss == 4.7)
  {
    regValue = 0x05;
    m_Scale = 2.56;
  }
  else if(gauss == 5.6)
  {
    regValue = 0x06;
    m_Scale = 3.03;
  }
  else if(gauss == 8.1)
  {
    regValue = 0x07;
    m_Scale = 4.35;
  }
  else
    return ErrorCode_1_Num;
  
  // Setting is in the top 3 bits of the register.
  regValue = regValue << 5;
  Write(ConfigurationRegisterB, regValue);
}

int HMC5883L::SetMeasurementMode(uint8_t mode)
{
  Write(ModeRegister, mode);
}

void HMC5883L::Write(int address, int data)
{
  Wire.beginTransmission(HMC5883L_Address);
  Wire.write(address);
  Wire.write(data);
  Wire.endTransmission();
}

uint8_t* HMC5883L::Read(int address, int length)
{
  Wire.beginTransmission(HMC5883L_Address);
  Wire.write(address);
  Wire.endTransmission();
  
  Wire.beginTransmission(HMC5883L_Address);
  Wire.requestFrom(HMC5883L_Address, length);

  // uint8_t buffer[length];
  uint8_t *buffer = new uint8_t[length];
  if(Wire.available() == length)
  {
    for(uint8_t i = 0; i < length; i++)
    {
      buffer[i] = Wire.read();
    }
  }
  Wire.endTransmission();

  return buffer;
}

const char* HMC5883L::GetErrorText(int errorCode)
{
    if(ErrorCode_1_Num == 1)
      return ErrorCode_1;
    
    return "Error not defined.";
}
```

HMC5883L.h:
```
#ifndef HMC5883L_h
#define HMC5883L_h

#include <Arduino.h>
#include <Wire.h>

#define HMC5883L_Address 0x1E
#define ConfigurationRegisterA 0x00
#define ConfigurationRegisterB 0x01
#define ModeRegister 0x02
#define DataRegisterBegin 0x03

#define Measurement_Continuous 0x00
#define Measurement_SingleShot 0x01
#define Measurement_Idle 0x03

#define ErrorCode_1 "Entered scale was not valid, valid gauss values are: 0.88, 1.3, 1.9, 2.5, 4.0, 4.7, 5.6, 8.1"
#define ErrorCode_1_Num 1

struct MagnetometerScaled
{
	float XAxis;
	float YAxis;
	float ZAxis;
};

struct MagnetometerRaw
{
	int XAxis;
	int YAxis;
	int ZAxis;
};

class HMC5883L
{
	public:
	  HMC5883L();

	  MagnetometerRaw ReadRawAxis();
	  MagnetometerScaled ReadScaledAxis();
  
	  int SetMeasurementMode(uint8_t mode);
	  int SetScale(float gauss);

	  const char* GetErrorText(int errorCode);

	protected:
	  void Write(int address, int byte);
	  uint8_t* Read(int address, int length);

	private:
	  float m_Scale;
};
#endif
```

zurikus, 2016-03-11:
Hello! 
I have the same issue as leandrogs described. Does anyone have information how this issue can be solved or if any workround exists? 

I'm using this pease of code : 

/*
Analog input 4 I2C SDA
Analog input 5 I2C SCL
*/

#include <Wire.h> //I2C Arduino Library

#define HMC5883L_ADDRESS 0x1E
#define HMC5883L_DEFAULT_ADDRESS 0x1E

void setup(){
//Initialize Serial and I2C communications
Serial.begin(9600);
Wire.begin();
//Put the HMC5883 IC into the correct operating mode
Wire.beginTransmission(HMC5883L_ADDRESS); //open communication with HMC5883
Wire.write(0x02); //select mode register
Wire.write(0x00); //continuous measurement mode
Wire.endTransmission();
}

void loop(){
int x,y,z; //triple axis data

//Tell the HMC5883 where to begin reading data
Wire.beginTransmission(HMC5883L_ADDRESS);
Wire.write(0x03); //select register 3, X MSB register
Wire.endTransmission();

//Read data from each axis, 2 registers per axis
Wire.requestFrom(HMC5883L_ADDRESS, 6);
if(6<=Wire.available()){
x = Wire.read()<<8; //X msb
x |= Wire.read(); //X lsb
z = Wire.read()<<8; //Z msb
z |= Wire.read(); //Z lsb
y = Wire.read()<<8; //Y msb
y |= Wire.read(); //Y lsb
}
//Print out values of each axis
Serial.print("x: ");
Serial.print(x);
Serial.print(" y: ");
Serial.print(y);
Serial.print(" z: ");
Serial.println(z);
delay(250);
}

me-no-dev, 2016-03-21:
```cpp
uint8_t* HMC5883L::Read(int address, int length)
{
  Wire.beginTransmission(HMC5883L_Address);
  Wire.write(address);
  Wire.endTransmission();

  //Wire.beginTransmission(HMC5883L_Address); DO NOT DO THIS! IT'S FOR WRITING ONLY
  Wire.requestFrom(HMC5883L_Address, length);

  // uint8_t buffer[length];
  uint8_t *buffer = new uint8_t[length];
  if(Wire.available() == length)
  {
    for(uint8_t i = 0; i < length; i++)
    {
      buffer[i] = Wire.read();
    }
  }
  //Wire.endTransmission(); SEE ABOVE

  return buffer;
}
```

me-no-dev, 2016-03-21:
```cpp
MagnetometerRaw HMC5883L::ReadRawAxis()
{
  const uint8_t* buffer = Read(DataRegisterBegin, 6);
  MagnetometerRaw raw = MagnetometerRaw();
  raw.XAxis = ((uint16_t)buffer[0] << 8) | buffer[1];
  raw.ZAxis = ((uint16_t)buffer[2] << 8) | buffer[3];
  raw.YAxis = ((uint16_t)buffer[4] << 8) | buffer[5];
  delete [] buffer;
  return raw;
}
```

me-no-dev, 2016-03-21:
```cpp
//define the exact size in bits because int is a different thing depending on the MCU's bitnes
struct MagnetometerRaw
{
    int16_t XAxis;
    int16_t YAxis;
    int16_t ZAxis;
};
```

zenmanenergy, 2016-04-24:
Thanks for the code update me-no-dev, although I can't seem to get it to work. I must have done something wrong. If any of you have tested this, can you post a zip file to your code? I would love to see this working. These hmc compasses are super cheap!

gloverd, 2016-06-30:
I'm still seeing issues as well, and I can't get me-no-dev's update to work.

Has anyone gotten this working?

akale2015, 2016-08-24:
Hello everyone, did anybody get this work please????

leandrogs, 2016-08-24:
I had to change lot of thibgs on the library. I'll post on my github
Tomorrow and update you guys. Sorry for the delay!

akale2015, 2016-08-24:
Cool thank you! 

Sent from Akale Selassie

zenmanenergy, 2016-08-24:
Thank you Leandrogs!

zurikus, 2016-08-24:
Thank you Leandrogs! 
Waiting for your update as well!

leandrogs, 2016-08-24:
The library use is exactly as said on previews answers. here is the repository: https://github.com/leandrogs/HMC5883L

akale2015, 2016-08-24:
Thank you. I am still trying to get it work. Do you know how I can connect the hardware please? Which is I am using ESP8266 ESP-01 and HMC5883L 3-Axis Compass?

zurikus, 2016-08-25:
Thank you Leandrogs for your update! Will try it today's evening!
Regards,
Zurikus.

zurikus, 2016-08-25:
Hello Leandrogs again,
I tried your library out, but I'm still getting suspicious values like:

147.00 | 64917.00 | 240.00
or 
65475.00 | 65422.00 | 466.00

I'm using Nodemcu 8266 v.1.0 dev kit and HMC5883L. 

Connecting: 
board -- Sensor
vin -- vcc
gnd -- gnd
D3 -- SCL
D4 -- SDA

Regards,
Zurikus.

leandrogs, 2016-08-25:
Maybe a bad connection?

steelgoose, 2016-09-11:
Hello @leandrogs ,

I am working on a project with nearly identical components, with almost Identical results. It appears to me that the values for x, y, z, are read from an unsigned 16 bit register instead of a signed register. I am not a coder so I don't know how to fix it, but from what I can see is that, 65475 | 65422 | 466 is actually, -61 | -114 | 466. The number 65535 is a (-1). 
A 16 bit unsigned register has 65536 numbers while a 16 bit signed register has -32768 numbers and +32768 numbers.  

So, I thought I would leave it up to the pros to fine a permanent cure. In the mean time I have come up with a Quick Fix until someone comes up with a better one, and maybe this will help someone resolve this issue. I installed the latest version of everything and the problem is still their... so apparently it hasn't been fixed yet. I don't know if the signed registered thing is really the problem but at least it is a place to start.

Good luck,
```
 //Read data from each axis, 2 registers per axis
  Wire.requestFrom(address, 6);
  if(6<=Wire.available()){
    x = Wire.read()<<8; //X msb
    x |= Wire.read(); //X lsb
    z = Wire.read()<<8; //Z msb
    z |= Wire.read(); //Z lsb
    y = Wire.read()<<8; //Y msb
    y |= Wire.read(); //Y lsb
  }

// Quick Fix
  if (x > 32767)
    x = x - 65536;
  if (y > 32767)
    y = y - 65536;
  if (z > 32767)
    z = z - 65536;
```

zurikus, 2016-09-12:
Thank you steelgoose for fix.
I tried that and looks like sensor giving correct results.

So I'm using this code : 

/*
Analog input 4 I2C SDA
Analog input 5 I2C SCL
*/

#include //I2C Arduino Library

#define HMC5883L_ADDRESS 0x1E
#define HMC5883L_DEFAULT_ADDRESS 0x1E

void setup(){
//Initialize Serial and I2C communications
Serial.begin(9600);
Wire.begin();
//Put the HMC5883 IC into the correct operating mode
Wire.beginTransmission(HMC5883L_ADDRESS); //open communication with HMC5883
Wire.write(0x02); //select mode register
Wire.write(0x00); //continuous measurement mode
Wire.endTransmission();
}

void loop(){
int x,y,z; //triple axis data

//Tell the HMC5883 where to begin reading data
Wire.beginTransmission(HMC5883L_ADDRESS);
Wire.write(0x03); //select register 3, X MSB register
Wire.endTransmission();

//Read data from each axis, 2 registers per axis
Wire.requestFrom(HMC5883L_ADDRESS, 6);
if(6<=Wire.available()){
x = Wire.read()<<8; //X msb
x |= Wire.read(); //X lsb
z = Wire.read()<<8; //Z msb
z |= Wire.read(); //Z lsb
y = Wire.read()<<8; //Y msb
y |= Wire.read(); //Y lsb
}

// Quick Fix from steelgoose
  if (x > 32767)
    x = x - 65536;
  if (y > 32767)
    y = y - 65536;
  if (z > 32767)
    z = z - 65536;

//Print out values of each axis
Serial.print("x: ");
Serial.print(x);
Serial.print(" y: ");
Serial.print(y);
Serial.print(" z: ");
Serial.println(z);
delay(250);
}

I'm using Nodemcu 8266 v.1.0 dev kit and HMC5883L
Connecting:
 board -- Sensor
 vin -- vcc
 gnd -- gnd
 D1 -- SCL
 D2 -- SDA

Good luck.

leandrogs, 2016-09-27:
I'm closing this issue... We already have many answers to the same problem.

closed 2016-09-27 by leandrogs

armaan159, 2016-11-23:
can this library work with esp8266 esp12E 
SDA= GPIO4
SCL= GPIO5
code being same as @leandrogs which is given in the beginning

profjean, 2021-05-25:
https://www.esp8266.com/viewtopic.php?p=68154# here is solution