esp8266/Arduino #471: WiFi.printDiag doesn't support softAP

TimeTravelingOwls, 2015-06-25:
This may be outside of the scope of WiFi.printDiag, but it doesn't seem to support softAP. The Mode gets displayed as AP, but status stays set at 255 regardless of the AP's state:
>Mode: AP
>PHY mode: N
>Channel: 1
>AP id: 0
>Status: 255
>Auto connect: 1
>SSID (0): 
>Passphrase (0): 
>BSSID set: 0

I know that some details of the softAP connection are available, because I can see the client's MAC, join, leave, etc., when I have `Serial.setDebugOutput(true);` set, but I'm not sure where that's coming from - is it somewhere in the lwip code? Is there a direct way to query softAP status within a sketch?

I'm trying to flip a token based on whether a client has disconnected from the softAP - any suggestions for a reliable way to track join/leave?

me-no-dev, 2015-06-25:
yeah, I have a fix for that. Will be in the next pull.

luc-github, 2015-06-26:
I read in SDK : SoftAP only support bg not n - I experienced bad connection in with SoftAP mode and PHY mode to n
do you mean now is it working ?

TimeTravelingOwls, 2015-06-26:
Thanks @me-no-dev - I'll test when it lands.

In the meanwhile, any suggestion for how to determine when a client has connected/disconnected? Seems like it ought to be easy, but I'm just not seeing it.

me-no-dev, 2015-06-26:
```cpp
extern "C" {
  #include "user_interface.h"
  #include "ip_addr.h"
  #include "mem.h"

  void _onWiFiEvent(System_Event_t *event){
    if(event->event == EVENT_SOFTAPMODE_STACONNECTED){
      Event_SoftAPMode_StaConnected_t *data = (Event_SoftAPMode_StaConnected_t *)(&event->event_info.sta_connected);
      Serial.printf("Station Connected: id: %d, mac: " MACSTR "\n", data->aid, MAC2STR(data->mac));
    } else if(event->event == EVENT_SOFTAPMODE_STADISCONNECTED){
      Event_SoftAPMode_StaDisconnected_t *data = (Event_SoftAPMode_StaDisconnected_t *)(&event->event_info.sta_disconnected);
      Serial.printf("Station Disconnected: id: %d, mac: " MACSTR "\n", data->aid, MAC2STR(data->mac));
    }
  }
}

void setup(){
  Serial.begin(115200);
  wifi_set_event_handler_cb(_onWiFiEvent);
}

igrr, 2015-06-26:
Note that this will disable the event handler present in ESP8266Wifi library.

me-no-dev, 2015-06-26:
We should add it there then

TimeTravelingOwls, 2015-06-26:
@me-no-dev: perfect - thanks!

Should I close this issue or wait until it's been pulled through?

igrr, 2015-06-26:
Better leave it open so I don't forget to add this.

devyte, 2017-10-21:
@igrr I think the current wifi event system supersedes this. Is there anything left to do?

igrr, 2017-10-22:
Yes, `onStationModeConnected` and `onStationModeDisconnected` functions in WiFi class provide this functionality.

closed 2017-10-22 by igrr