esp8266/Arduino #472: Pullup resistor required now on RST with latest code?

torntrousers, 2015-06-26:
I find that with the latest code (1.6.4+ board manager) an ESP-12 doesn't work properly with deepsleep going into a reset loop when waking up, but the unchanged hardware works fine with the 1.6.1 release from March 27. However, adding a pullup resistor on the RST pin gets it working fine with the 1.6.4+ code.

Is that expected?

duncan-a, 2015-06-26:
The RST (reset) pin should always be pulled high. It will tend to float
high but that's a very unstable state and cannot be relied upon.

Some proprietary boards may have a pullup built in but, if in doubt, add a
12K...

antelder, 2015-06-26:
I don't know, it feels like there may be more to it than that. This isn't some intermittent thing, it works fine with no pullup with the original 1.6.1 code but mostly doesn't with the newer code, and thats trying with different esp modules and breakout boards. Also there are lots of examples and blog posts about the net on how to do deep sleep with esp's, usually some months old so using older code, and they don't use a pullup and apparently worked anyway.

Could there be something like an internal pullup that is actived with the older code but not now with the new code?

torntrousers, 2015-06-27:
Thanks Links2004, trying the very latest code it does seem to have fixed this.

Unfortunately math functions don't seem to compile in the latest code though...
c:/arduino/esp8266/gitbrns/arduino-1.6.6/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_pow.o):(.literal+0x104): undefined reference to `__ieee754_sqrt'
c:/arduino/esp8266/gitbrns/arduino-1.6.6/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/lib\libm.a(lib_a-e_pow.o): In function `__ieee754_pow':
d:\ivan\projects\arduinoesp\toolchain\dl\esp-newlib\build\xtensa-lx106-elf\newlib\libm\math/../../../../../newlib/libm/math/e_pow.c:164: undefined reference to `__ieee754_sqrt'
collect2.exe: error: ld returned 1 exit status
Error compiling.

torntrousers, 2015-06-27:
Though the math thing is a seperate issue so i'll close this one.

tytower, 2015-07-16:
I'm still wondering why you would merge another issue with this one and then close it . I'll reopen this as the reset, deepSleep() ,continual reset problem is still cropping up frequently and has not been answered to my satisfaction anyway.

I am using a 100nF capacitor and no pullup resistor on my ESP-12 and its working beautifully with the latest Arduino IDE 1.6.5 but I have not used any other modules.

torntrousers, 2015-08-18:
I thought it was fixed when i tried building the latest code back on 27 Jun but this doesn't seem to be fixed after all now. I've tried with the latest staging release and building the latest code from git.

asetyde, 2015-08-18:
Also i wait for a safe all time restart...

torntrousers, 2015-09-18:
I've been trying to debug this some more, its still a problem with the very latest code (Sept 16) but its not quite what i thought before - 

The problem only happens when i'm powering the ESP from fully charged LIPO battery, so about 4.2 volts, and then it will only wake up properly from deepSleep if i'm touching the RST/GPIO16 pin with a finger, using a pullup resistor instead of my finger doesn't work. 

I know 4.2 volts is way out of spec, but its really convenient to be able to use a LIPO cell without a 3.3v regulator, and it seems to work fine (other than this wake up issue) on lots of ESPs i've tried some of which have been running for months, and when using the original 1.6.1 esp arduino release from back in April even the wake up works fine. 

Any ideas? What could holding a finger on the pin be doing?

Juppit, 2015-09-18:
Putting your finger on a pin is like connecting an antenna to it.
Remember that you are running an antenna next to a transmitter.
This can explain many...

On 18.09.2015 at 13:27 wrote torntrousers:

torntrousers, 2015-09-18:
That seems right - a crocodile clip with a foot of wire seems to work as well as my finger. Neither are very convenient though, i was hoping for something like a resistor or capacitor somewhere could fix it.

Links2004, 2015-11-27:
using ```ESP.deepSleep``` is working fine for me.

please retry in last release, 
If it still a problem feel free to reopen.

closed 2015-11-27 by Links2004

torntrousers, 2015-11-27:
I've just tried again with the latest staging code - it does appear to be working now. Fab! Thank you.