esp8266/Arduino #475: dtosrtf doesn't pad to desired width

mdmbc, 2015-06-27:
The dtostrf implementation seems to be missing the leading space padding of the AVR variant. The padding is very helpful when doing things like updating LCDs.

On the 328P the following test code prints  '    17.0105' , and on the ESP8266 it prints '17.0105'.

float input = 17.0105;
char val[12];

void setup() {
  Serial.begin(115200);
  dtostrf(input, 11, 4, val);
  Serial.print("'");
  Serial.print(val);
  Serial.println("'");

}

void loop() {

  }

Makuna, 2015-07-10:
If I read the docs correctly, your example should have output "____17.0105" (four spaces) and not "_17.0105" (one space).

mdmbc, 2015-07-11:
That's correct. There actually are 4 spaces in there, but it doesn't seem to be displaying properly. If I edit the post, there are indeed 4 spaces in there. The ESP8266 implementation doesn't add the leading spaces.

closed 2015-07-13 by igrr