esp8266/Arduino #478: INPUT_PULLDOWN Doesn't work

mc-hamster, 2015-06-27:
Good day,

I have observed that INPUT_PULLDOWN doesn't enable the internal pulldown but rather enables a pull up.

  pinMode ( sensor1, INPUT_PULLDOWN );
  pinMode ( sensor2, INPUT_PULLDOWN );  

where sensor1 and sensor2 are gpio 14 and gpio 12 respectively.

If I set them to INPUT, then the pin behaves like a floating input. Otherwise, INPUT_PULLDOWN and INPUT_PULLUP seem to behave exactly the same.

I've take a peak into core_esp8266_wiring_digital.c and esp8266_peri.h to see if there's an easily identified typo can can see that the registers are being written to with different values depending on the mode selection.

Please advice.

Thanks!

- Jm

Links2004, 2015-06-27:
espressif write:
There are no pull-down functions on GPIO pad now, so we should never use these registers.
http://bbs.espressif.com/viewtopic.php?f=5&t=481

may the change the chip internals at one point,
or simply do knows there one chip....

has anyone ever tested pull down?

mc-hamster, 2015-06-28:
I can confirm that enabling a pulldown doesn't work.

- Jm

me-no-dev, 2015-06-29:
i originally did not implement this at all in digital.c, but someone came along asking for it (for the DHT lib maybe?) and we added it then. To be honest I did not even have the definitions for pulldowns on those pins.
On another note, INPUT_PULLDOWN should work for pin 16, as it was originally defined in the documentation.

mc-hamster, 2015-07-01:
May I advice that to keep others (like myself) from running into this, let's remove the reference to the pull downs from https://github.com/esp8266/Arduino?

mc-hamster, 2015-08-14:
Closing this issue. Thanks!

- Jm

closed 2015-08-14 by mc-hamster

scottchiefbaker, 2020-03-10:
Just so I'm 100% clear on this issue. The ESP8266 hardware itself does not have an option for built-in pull-down resistors correct? This is not a code limitation, but a hardware limitation?

From what I'm reading there **is** a pull-down resistor, but only on pin 16?

devyte, 2020-03-11:
correct.