esp8266/Arduino #479: SDWebserver.ino example Compile error: "multiple definition of `hostname'"

sunnyhighway, 2015-06-27:
Using the esp8266-1.6.5-804-g2d340c7 the SDWebServer.ino won't compile anymore.
It complains the about the variable "hostname" being used elsewhere too.

```javascript
/Users/barthijwegen/Library/Arduino15/packages/esp8266/hardware/esp8266/1.6.5-804-g2d340c7/tools/sdk//lib/libmain.a(eagle_lwip_if.o):(.bss+0x0): multiple definition of `hostname'
/var/folders/lf/dxdg9w953sd0xts2v_13grv00000gn/T/build729210233767314573.tmp/SDWebServer.cpp.o:(.data+0x0): first defined here
collect2: error: ld returned 1 exit status
Error compiling.
```
The SDWebServer.ino does not seem to be at fault.

igrr, 2015-06-28:
```const char* hostname = "esp8266sd";```
This definition in the sketch collides with a global variable defined inside libmain.a.

igrr, 2015-07-23:
Fixed in stable package (1.6.5-947-g39819f0).

closed 2015-07-23 by igrr