esp8266/Arduino #480: Different circuit neeeded for automatic upload using CP210x

probonopd, 2015-06-28:
I am using a CP210x based USB-to-serial adapter called the C231932. I tried setting up automatic upload using the "Minimal hardware Setup for Bootloading and usage" circuit from https://github.com/esp8266/Arduino/blob/esp8266-sdk-1.0/README.md#minimal-hardware-setup-for-bootloading-and-usage but it did not work for me.

No I have changed the circuit slightly and it works now:

| PIN           | Resistor | Serial Adapter |
| ------------- | -------- | -------------- |
| VCC           |          | VCC (3.3V)     |
| GND           |          | GND            |
| TX or GPIO2*  |          | RX             |
| RX            |          | TX             |
| GPIO0         |10K **between** GPIO0 and  DTR  | DTR            |
| Reset*        | 10K **between** Reset and RTS          | RTS            |
| GPIO15*       | PullDown |                |
| CH_PD         | PullUp   |                |

My CP210x adapter has 3.3V on the DTR and RTS lines unless they are pulled down by the software. Currently this happens when I enter the Serial Monitor; see https://github.com/esp8266/Arduino/issues/22#issuecomment-116276423

Links2004, 2015-06-28:
the problem is the Arduino IDE or you Serial terminal.

is the boot loading still working with the 10k?
i don't think its good for the stability of the uploading.

the problem shut be fixed in software (Arduino IDE) so that DTR and RTS is not activate in our case.
i use "real term" for serial debugging there i can control DTR and RTS manually.

probonopd, 2015-06-28:
Yes 921600 is working fine for me.

cimba007, 2015-06-28:
I thing I have the same problem here. What are your values for "pull up and pull down"? I am using 10k and only get ~2,2V on my gpio0, resulting from 3,3V high on DTR as it is not in use and a 10k pulldown but ofc. this is not enough if DTR got some decent amps I guess.

probonopd, 2015-06-28:
10K

cimba007, 2015-06-29:
What are u doing with GPIO0 and RESET when the CP210x  is not connected? What values are you using in this case?

probonopd, 2015-06-29:
Good point. In my case the CP210x is always connected and is also used to provide the 3.3V (which appears to work in my case).

Links2004, 2015-06-29:
https://github.com/esp8266/Arduino/blob/esp8266-sdk-1.0/README.md#minimal-hardware-setup-for-running-only

probonopd, 2015-07-20:
Following advice from igrr, I am now using the circuitry from the NodeMCU 1.0 for my ESP-12Es. See schematics of the NodeMCU 1.0.

Two 2N2222 npn transistors (10K) and 3 resistors, thereof 1 pullup for CH_PD. Working well.This circuit can be used for flashing and running, it does auto-reset and the serial monitor in the IDE can be used.

Possibly the documentation in the README should be updated to reflect this.

JeroenBeemster, 2015-07-23:
Thanks probonopd, This solved my issue aswell.For everyone who find it hard to read electronic schematics. I created a simplified example using a breadboard explanation.
http://www.arduinesp.com/blink

probonopd, 2015-07-23:
Thanks JeroenBeemster, I hope you are not offended if I attach your example for quick reference here.

[image]

closed 2015-07-30 by igrr

ReMiA00, 2015-10-14:
I have tried several methods on resetting the ESP-12E for flashing with Arduino IDE
However none worked great, and all will break the "serial monitor" in the IDE

Since i had a **ATtiny85** lying around i used this to send the reset to the ESP and keep the GPIO0 down for a second after reset. 
- The RTS pulse on my CP2102 lasts about 5 to 8 millis
- When serial monitor is started RTS remains low until the monitor is closed.

```
/*
  Reset ESP-12E with ATTiny85 (8Mhz)
 */

#include <SoftwareSerial.h>
// Definitions
#define rxPin 3  
#define txPin 4

SoftwareSerial mySerial(rxPin, txPin);

const int RTS_pin = 2; // Pin 7 on Attiny
const int RST_pin = 0; // Pin 5 on Attiny
const int GPIO0_pin = 1; // Pin 6 on Attiny

// Hookup 
// ESP     <--> Tiny pin <--> CP2102 or FT232RL
//  GND   <--> 4           <--> GND
//  5V      <--> 8 
//                    7           <--> RTS 
//  RST    <--> 5            
//  GPIO0 <--> 6                  
//  RX <-------------------> TX
//  TX <-------------------> RX

unsigned long StartReset = 0;
unsigned long Puls = 0; 

void setup() {
  mySerial.begin(9600);

// USB to Serial Converter Flow Control
  pinMode(RTS_pin,INPUT);

// ESP Pins
  pinMode(RST_pin,OUTPUT);
  pinMode(GPIO0_pin,OUTPUT);
  digitalWrite(RST_pin, HIGH);
  digitalWrite(GPIO0_pin, HIGH);
}

void loop() {
  int RTS = digitalRead(RTS_pin);
  
  if (RTS == 0 and StartReset == 0 ) { // Start Reset Puls
    StartReset = millis();
    mySerial.println("Reset");
  } 
  if (RTS == 1 and StartReset > 0 ) {  // Eind Reset Puls
    Puls = millis() - StartReset;
    mySerial.print("Puls = ");
    mySerial.println(Puls);
    StartReset = 0;
    
    if (Puls > 0 and Puls < 20) { // puls is around 5 to 8 millis
      digitalWrite(RST_pin, LOW);
      digitalWrite(GPIO0_pin, LOW);
      delay(5);        // length Reset puls
      digitalWrite(RST_pin, HIGH);
      delay(1000);        // Wait for ESP to boot
      digitalWrite(GPIO0_pin, HIGH);
      Puls = 0;
    }  
  }
}
```

I used a Arduino Nano with ArduinoISP sketch to program the ATTiny using IDE 1.6.5 
Add in "Boards Manager" https://raw.githubusercontent.com/damellis/attiny/ide-1.6.x-boards-manager/package_damellis_attiny_index.json

Wire it up:
// Hookup to Program ATTiny85
// NANO   <--> Tiny pin
//  GND   <--> 4
//  5V    <--> 8
//  10    <--> 1
//  11    <--> 5
//  12    <--> 6
//  13    <--> 7

Choose ATTiny85 Board at 8Mhz and flash the Bootloader
Now you can upload the sketch to the ATTiny (use Arduino as ISP as programmer)

Wires for Serial debugging
// NANO   <--> Tiny pin
//  RX(0) <--> 3 serial monitor
//  TX(1) <--> 4 serial monitor

vicatcu, 2015-10-20:
@igrr @probonopd ... reading the README on the [esptool github repository](https://github.com/themadinventor/esptool), and using pin nomenclature from [the wiki](https://github.com/esp8266/esp8266-wiki/wiki/Pin-definition), it says:

> esptool uses the RTS and DTR modem status lines to automatically enter the bootloader. Connect RTS to CH_PD (which is used as active-low reset) and DTR to GPIO0.

I believe CHIP_EN is synonymous with CH_PD (i.e. pin 7 of the ESP8266).. Looking at the whole NodeMCU 0.9 schematic, it would appear that they label this net "EN" and it is simply pulled up to 3.3V. They effectively connect the net named "nRST" which is synonymous with EXT_RSTB (i.e. pin 32 of the ESP8266), to the DTR line, and effectively connect GPIO0 to the RTS line. That's quite a difference to the recommendations of esptool. 

GPIO0, GPIO2, and EXT_RSTB are also passively pulled up to 3.3V, and GPIO15 is passively pulled down to GND.

The difference between CHIP_EN and EXT_RSTB is pretty subtle, so it probably doesn't matter (?) which one gets wiggled for auto-reset/upload support.. But I'm more doubtful where it comes to the differences between RTS and DTR. Is the circuitry in this thread above still sound, or should new hardware wire to DTR in agreement with the esptool  recommendations?

igrr, 2015-10-20:
The circuit in this thread works with esptool-ck when reset method is set
to 'nodemcu'. It might be different with esptool.py which you linked to.

vicatcu, 2015-10-20:
@igrr interesting, didn't realize there was a difference. So does auto-reset not work for "Generic ESP8266 Module" selections, or is the same wiring applicable in that case? I'm asking because I'm putting designing in an ESP8266 module on a board right now and would like to have it be compatible with this project.

hallard, 2015-11-12:
For those who just have FTDI cable with only DTR and no RTS line, I'm using this one (FTDI 5V compatible) with great success on ESP12 and ESP07

[image]

Of course for this to work, don't forget Pull up on GPIO0 and Reset (10K each) and pull down on GPIO15 (10K also) and 10K pullup on CH_PD but you know this one.

If your FTDI is 3V3 don't try to power your board with it (use another source), lot's of FTDI onboard reg can't do 200mA needed sometimes by ESP

hallard, 2015-11-17:
After some investigation, it's better to reset with CH_PT
here it [why](http://hallard.me/esp8266-autoreset/) I've update previous schematic

ReMiA00, 2015-11-17:
This looks promosing !
Will the serial monitor also work in the IDE ?

hallard, 2015-11-17:
@ReMiA00 
yes Serial monitor works fine with this and FTDI like cable, I I use it every day ;-)

tablatronix, 2015-12-22:
I am using the nodemcu circuit and it works great for programing, but my module reboots when I open and close the serial monitor.

My logic probe, says my adapter DTR and RST are both high at rest and LOW when serial monitor is open ( same for putty ). Changes to neither `serial.disable...` settings makes any difference either,

Using IDE 1.6.6, and git version of repo.

tablatronix, 2015-12-22:
FYI, I am using 2n3904 transistors (have a bunch) with 10k base resistors, I have not bothered calculating proper resistor values yet.. Seems to work ok for programming every time so far.

jayseeee, 2016-01-03:
I had to come up with another circuit to do the "autoreset" with the FDTI interface as none of the above mentioned worked for me. I designed the PCB as an interface in between the FDTI an the board to be programmed, with an additional output of RESET and FLASH (GPIO0).

[image]

muke1970, 2016-07-08:
I tried JeroenBeemster's circuit on esp8266-01 and work very well. I tried the same on esp8266-12F but receive warning: espcomm_sync failed error: espcomm_open failed error: espcomm_upload_mem failed. The same with hallard's. Do you have suggestions for me? :-)

muke1970, 2016-07-09:
Sorry I make a mistake to wiring esp8266-12F.
It work like a charm!
Thanks again.
Emiliano.

tablatronix, 2016-07-09:
Make sure you are using nodemcu reset mode not ck

paulwww, 2016-08-02:
ReMiA00 many thanks for the attiny85 solution. It is working well for me. At first I got it to work on an atmega328 since I could easily debug/test there. I then went to the attiny85 and built a circuit between my FTDI and ESP8266, now it is even more fun to develop for the ESP8266.