esp8266/Arduino #482: Enabling UART2 for the NodeMCU?

draput, 2015-06-29:
The NodeMCU V1.0 (the new one) is showing a RDX2 and TDX2 on the pins D7 (GPIO13) and D8 (GPIO15). Any chance to have a Serial2 for them?

Best regards, you are doing a great work!

https://raw.githubusercontent.com/nodemcu/nodemcu-devkit/master/Documents/NODEMCU-DEVKIT-INSTRUCTION-EN.png

igrr, 2015-06-29:
Please check https://github.com/esp8266/Arduino#serial regarding Serial1 support.

draput, 2015-06-29:
The idea was to be able to use two UARTS simultaneously (and the third one (TX1) for debug). 

BTW, how many UARTS does ESP8266 have? Are RDX2 and TDX2 just pin remaps for the same UART which is normally mapped on GPIO1/GPIO3, or are there two UARTS, and what Serial.swap() does is to swap between them?

me-no-dev, 2015-06-29:
it has 2 UARTS only. but pins are swappable, that is why you see such markings.
You can only have one full uart and one just TX (Serial1), but the pins for Serial0 can go on the other pins.

draput, 2015-06-29:
Thx for clarifying this.

closed 2015-06-29 by draput

elixirml, 2016-10-31:
Hello friends I'm new to developing with esp8266, I'm trying to connect both an RFID reader rc522 and screen nextion 2.4 "in a NodeMCU 1.0, using arduino IDE v 1.6.11 to read an RFID tag and display the UID in Nextion 2.4 screen ".

I managed to connect and work with each separately, but I can not make them work together.

when the connections operate separately in NodeMCU v 1.0 are the following:

Nextion 2.4 "||| MCU Node 1.0

5V-------------- 5V
RX ------------ (D7) - GPIO13 - RXD2 - HMOSI
TX ------------ (D8) - GPIO15 - TXD2 - HCS
GND ----------GND

RC522 ||| MCU Node 1.0

3.3V------ 3.3V
RST ------ (D2) - GPIO4
GND------- GND 
IRQ -----
MISO -------(D6) - GPIO12 - HMISO
MOSI -------(D7) - GPIO13 - RXD2 - HMOSI
SCK --------(D5) - GPIO14 - HSCLK
SS ----------(D4) - GPIO2 - TXD1

Both devices need to use the pin D7 (RXD2), I would like to use both at the same time, I read that I use UART but do not understand how to apply.

any help is welcome
thank you very much

asetyde, 2016-11-02:
also i try to work but no success .. i waiting

lrmoreno007, 2016-11-02:
It is explained above.

Serial are mapped to GPIO1 and GPIO3, when you do "Serial.swap(); Serial.begin();" TXD2 and RXD2 are used, mapped to GPIO13 and GPIO15. But ESP8266 always have only one full Serial (Serial or Serial2) and one TX (Serial1).

More info: https://github.com/esp8266/Arduino/blob/master/doc/reference.md#serial

To solve the problem you must use SoftwareSerial Library.

arantius, 2017-08-23:
Google led me here, correct link is now:

https://github.com/esp8266/Arduino/blob/master/doc/reference.rst#serial

ozarchie, 2018-01-09:
Is it possible to use GPIO1 as TxD1?
This would allow debug messages to appear on the serial console of the Arduino IDE.

e.g.
`
Serial.begin(9600);           // Initialise RxD0, TxD0

Serial.swap();                    // Swap to Alternate pins -> EQG on "Serial" (RxD-nn, TxD-nn)

delay(1000);                      // Wait for a second

Serial.println("Test 13, 15"); // Serial string appears on line attached to 13, 15

Serial1.begin(9600, SERIAL_8N1, SERIAL_FULL, 1); // Set "Serial1" as debug, swap to GPIO1 (i.e. output to USB connected to download port)    
`

allgoodtw, 2018-04-24:
Did you wire the Tx to the Rx on the serial device connected to pins GPIO 13 and GPIO 15
GPIO 15 D8 is TXD2 needs to be connected to the RX pin on the connected device depending on if it is considered to be a Terminal or Modem Device.  DCE or DTE  device. Always Out put to Input . If you connect a DTE device to another DTE device the pins are swapped 2->3 and 3>2. Just a thought. IT's the Same DCE to DCE But there is no swap of pins on a DTE to DCE connection. DTE (Data terminal) Equipment and DCE (Data Communication Equipment) Modem. Most all devices not clearly defined are generally considered Terminal or DTE.

gitpeut, 2018-04-24:
Probably a misunderstanding. The swap to pin13 and 15 worked. But I didn't get a swap to GPIO1 to work,
as ozarchie proposed.

allgoodtw, 2018-04-24:
Question: Does the Swap function cause a problem reprogramming the device as if it is swapped the normal IDE pins are no longer connected?

alex-eri, 2023-05-12:
> GPIO4

pull this pin to gnd (software or resistor)