esp8266/Arduino #483: NodeMCU issue with Arduino IDE v1.6.5

kelray, 2015-06-29:
Hello there,

My NodeMCU kit can't be programmed properly with the latest Arduino IDE v1.6.5 (after following the instructions here to install ESP8266  Arduino package in the IDE: https://github.com/esp8266/Arduino )

The WiFiWebServer examples gives nonsense dumb on the terminal like that whenever the board is programmed or reset:
[image]

As for the Baud rate and board connection configuration it is:
[image]

The same happens if I upload a simple sketch (LED blinking sketch) instead, doesn't work, any clue how to fix this issue?

Thanks,
K

igrr, 2015-06-29:
Are you using NodeMCU 0.9 (yellow one) or NodeMCU 1.0 (black one)? NodeMCU 1.0 is not yet supported in the current version of boards manager package.

kelray, 2015-06-29:
I'm using v1.0 (the white one), now it makes sense, thanks igrr.
do you know if support for this version is coming soon or not?.

Thanks.

igrr, 2015-06-29:
The support is ready, but i'm not sure when the release will be updated. You can use this staging package in the meantime: http://arduino.esp8266.com/staging/package_esp8266com_index.json

me-no-dev, 2015-06-29:
the sketch on the picture does not call Serial.begin(115200);
also the upload speed has nothing to do with the Serial speed of the sketch.
And third, it's ok to get garbage in the beginning of the sketch boot.

kelray, 2015-06-30:
The serial.begin() is called in the setup, and I know the speed of the serial setup in the sketch isn't related to the upload speed, but I was asking if the upload speed was the right one or not, beside igrr cleared it out.

closed 2015-06-30 by kelray

leosf19, 2016-09-04:
Could you find any solution to this? Every time i tried to upload any code using arduino IDE, i get that garbage after reset the module.

kelray, 2016-09-04:
The latest Arduino IDE (starting 1.6.9 and newer) support NodeMCU, so upgrade to the newest Arduino IDE and install the package required for NodeMCU from Tools -> Boards -> Board Manager, hopefully things will work.

Cheers,
K