esp8266/Arduino #484: ESP reboot on using PSTR() within sprintf function call

mvdbro, 2015-06-29:
This results into a reboot:

char str[80];
sprintf(str,PSTR("This is a demo: %s for sprintf from flash"),"test123");
Serial.println(str); 

Or should I be using sprint_P() but that seems unavailable on ESP.
How to use progmem strings as param to sprintf() ?

Links2004, 2015-06-29:
you mean ```sprintf_P()```?
https://github.com/esp8266/Arduino/blob/ea302aab05480ad36c076b618abb642b1eb1893b/hardware/arduino/sam/cores/arduino/avr/pgmspace.h#L54

hallard, 2015-07-07:
@Links2004 
Yeah, sprintf_P() would be great but I'm unable to use it (not found at compilation), did I missed one include file ?

Makuna, 2015-07-10:
Links2004 link is incorrect, it should be 
https://github.com/esp8266/Arduino/blob/ea302aab05480ad36c076b618abb642b1eb1893b/hardware/esp8266com/esp8266/cores/esp8266/pgmspace.h

Are you including "avr/pgmspace.h"?  If so that doesn't work.  You need to include just "pgmspace.h".
Further, it looks like sprintf is not listed, but snprintf is.

closed 2015-07-30 by igrr