esp8266/Arduino #485: UART Swap only RX or TX pin seems not possible

hallard, 2015-06-29:
Hi there, 
I had an idea because I needed to swap only RX pin but not TX, so I changed the firmware the swap function to be able to do this like the following

    void uart_swap_tx(uart_t* uart) {
    if(uart == 0)
        return;
    switch(uart->uart_nr) {
        case UART0:
            if(uart->txPin == 1 ) {
                pinMode(15, FUNCTION_4); //TX
                IOSWAP |= (1 << IOSWAPU0);
                pinMode(1, INPUT); //TX
                uart->txPin = 15;
            } else {
                pinMode(1, SPECIAL); //TX
                IOSWAP &= ~(1 << IOSWAPU0);
                pinMode(15, INPUT); //TX
                uart->txPin = 1;
            }
            break;
        case UART1:
            // current no swap possible! see GPIO pins used by UART
            break;
        default:
            break;
        }
    }

    void uart_swap_rx(uart_t* uart) {
    if(uart == 0)
        return;
    switch(uart->uart_nr) {
        case UART0:
            if(uart->rxPin == 3) {
                pinMode(13, FUNCTION_4); //RX
                IOSWAP |= (1 << IOSWAPU0);
                pinMode(3, INPUT); //RX
                uart->rxPin = 13;
            } else {
                pinMode(3, SPECIAL); //RX
                IOSWAP &= ~(1 << IOSWAPU0);
                pinMode(13, INPUT); //RX
                uart->rxPin = 3;
            }
            break;
        case UART1:
            // current no swap possible! see GPIO pins used by UART
            break;
        default:
            break;
        }
    }

    void uart_swap(uart_t* uart) {
    uart_swap_rx(uart);
    uart_swap_tx(uart);
    }

Unfortunately if swap function still working as expected, calling swapRx really swap Rx but makes TX unavailable (on original Tx pin), I read the datasheet in details and I'm pretty sure device hardware can't do this. But it was worth trying

Links2004, 2015-06-29:
the IOSWAP register swap always both internally.
you allways swap like this:
```
TX: 1
RX: 3
CTS: 13
RTS: 15
```
or
```
TX: 15
RX: 13
CTS: 3
RTS: 1
```

hallard, 2015-06-29:
Yeah, exactly what I noticed, not really a problem, I swapped the Serial to avoid conflict with RX/TX upload and I'm using Serial1 for debug, but with your OTA it will resolve all my TX/RX conflict ;-)

Cheers
s

closed 2015-07-30 by igrr