esp8266/Arduino #486: Can't parse nested JSON on the ESP

maaike, 2015-06-29:
I posted this in the forum, but on second thought it probably belongs here (apologies if it doesn't).

I'm using an ESP01 with the Arduino IDE (1.6.4). I'm trying to parse some - valid - JSON with the ArduinoJson library and have had lots of trouble. By now I've narrowed it down to this. I can parse this string:
```
char json[] = "{\"1\":{\"lijn\":\"2 Kraayenstein\",\"min\":5}}";
```
but this one won't parse and returns "Parse object failed":
```
char json[] = "{\"1\":{\"lijn\":\"2 Kraayenstein\",\"min\":5},\"2\":{\"lijn\":\"2 Leyenburg\",\"min\":18}}";
```
When I upload my sketch to an Arduino Mega both strings parse fine. 

Here's the complete sketch:

```

#include <ArduinoJson.h>

void setup() {
  Serial.begin(9600);
  Serial.println("Here we go");

}

void loop() {
  StaticJsonBuffer<200> jsonBuffer;

  char json[] = "{\"1\":{\"lijn\":\"2 Kraayenstein\",\"min\":5},\"2\":{\"lijn\":\"2 Leyenburg\",\"min\":18}}";

  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println("parseObject() failed");
    delay(500);
    return;
  }

  const char* lijn = root["1"]["lijn"];
  int mins = root["1"]["min"];
  const char* lijn2 = root["2"]["lijn"];
  int mins2 = root["2"]["min"];

  Serial.println(lijn);
  Serial.println(mins);
 Serial.println(lijn2);
  Serial.println(mins2);
  delay(5000);
}

```

me-no-dev, 2015-06-30:
your jsonBuffer is shorter than the string

maaike, 2015-06-30:
Thanks. I figured 1 character = 1 byte. I guess I have to read up on data types. 
Still it's funny the JSON parses allright on the Arduino Mega. Anyway, thanks!

closed 2015-07-03 by maaike