esp8266/Arduino #487: Anybody can help with implenemt OLED SSD1306 library from project http://arduino-project.net/oled-chasy-na-arduino/ ?

Lstt2005, 2015-06-29:
Dear sirs. Sorry for my English! Recently i was found cool library for I2c OLED SSD1306 - http://arduino-project.net/oled-chasy-na-arduino/ . But this project is for Arduino. When I try compile sketch, i see :
"C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:90:2: error: 'uint8_t' does not name a type
  uint8_t* font;
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:91:2: error: 'uint8_t' does not name a type
  uint8_t x_size;
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:92:2: error: 'uint8_t' does not name a type
  uint8_t y_size;
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:93:2: error: 'uint8_t' does not name a type
  uint8_t offset;
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:94:2: error: 'uint8_t' does not name a type
  uint8_t numchars;
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:95:2: error: 'uint8_t' does not name a type
  uint8_t inverted;
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:101:17: error: expected ')' before 'data_pin'
   OLED (uint8_t data_pin, uint8_t sclk_pin, uint8_t rst_pin);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:102:17: error: expected ')' before 'data_pin'
   OLED (uint8_t data_pin, uint8_t sclk_pin);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:105:22: error: 'uint8_t' has not been declared
   void setBrightness(uint8_t value);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:109:17: error: 'uint16_t' has not been declared
   void setPixel(uint16_t x, uint16_t y);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:109:29: error: 'uint16_t' has not been declared
   void setPixel(uint16_t x, uint16_t y);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:110:17: error: 'uint16_t' has not been declared
   void clrPixel(uint16_t x, uint16_t y);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:110:29: error: 'uint16_t' has not been declared
   void clrPixel(uint16_t x, uint16_t y);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:111:17: error: 'uint16_t' has not been declared
   void invPixel(uint16_t x, uint16_t y);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:111:29: error: 'uint16_t' has not been declared
   void invPixel(uint16_t x, uint16_t y);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:114:14: error: 'String' has not been declared
   void print(String st, int x, int y);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:116:30: error: 'byte' has not been declared
   void printNumF(double num, byte dec, int x, int y, char divider='.', int length=0, char filler=' ');
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:117:16: error: 'uint8_t' has not been declared
   void setFont(uint8_t* font);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:118:33: error: 'uint8_t' has not been declared
   void drawBitmap(int x, int y, uint8_t* bitmap, int sx, int sy);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:129:3: error: 'uint8_t' does not name a type
   uint8_t   _scl_pin, _sda_pin, _rst_pin;
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:130:3: error: 'boolean' does not name a type
   boolean   _use_hw;
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:132:3: error: 'uint8_t' does not name a type
   uint8_t   scrbuf[1024];
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:135:57: error: 'byte' has not been declared
   void _convert_float(char *buf, double num, int width, byte prec);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:144:24: error: 'uint8_t' has not been declared
   void _sendTWIcommand(uint8_t value);
C:\Users\User\Documents\Arduino\libraries\OLED_I2C/OLED_I2C.h:145:19: error: 'byte' has not been declared
   void _sendStart(byte addr);"
Anybody can help me with iplement this library for ESP8266 Arduino IDE? Thanks.

draput, 2015-06-29:
Look here:

https://github.com/squix78/esp8266-projects/tree/master/arduino-ide/thirsdee

for the ssd1306_i2c.* classes. 

The project's page is here:

http://blog.squix.ch/2015/06/esp8266-meet-thirsdee-intelligent-plant.html

Lstt2005, 2015-06-30:
Thanks for links, but it's very difficult to me, because i am novice in coding )). And In addition, the link above has allready Russian and Ukranian fonts and simple function for print any text with different size on display..

me-no-dev, 2015-06-30:
here is the Adafruit lib made to work
https://dl.dropboxusercontent.com/u/8279480/Adafruit_SSD1306.zip

Lstt2005, 2015-06-30:
Thanks, i try later!

probonopd, 2015-06-30:
@me-no-dev, do you plan to push this upstream? If not, why don't you put it on GitHub?

me-no-dev, 2015-06-30:
https://github.com/me-no-dev/Adafruit_SSD1306

closed 2015-07-11 by Lstt2005

duncan-a, 2015-07-12:
Has anybody got the u8glib library working on the ESP8266 yet?

It seems to have a number of advantages over the Adafruit library,
including some really nice, smooth-edged, large fonts which would be ideal
for temperature display.

hruskozrout, 2015-07-23:
@duncan-a Somebody said that is working now, but for me it stops (stucks) compiling on the same file again and again: u8g_com_arduino_t6963.c.o

drmpf, 2015-07-23:
I had a similar never ending compile on another sketch.  I just closed the IDE and started it again to force a complete compile from scratch.

duncan-a, 2015-07-23:
Roman

Forgive me being dense, but are you meaning that the 'standard' library is
supposed to be working now with the ESP8266 or is there a 'special' brew?