esp8266/Arduino #488: sscanf compilation error

nicjohnston, 2015-06-29:
I recently tried to use sscanf, however the compiler threw an undefined reference error.
Replacing sscanf with os_sprintf as recommended in issue #404 did not solve it, instead it created a watchdog timer reboot.
Here is the code I was using:
```c++
void setup() {
  Serial.begin(115200);
  Serial.println("starting");
  char *data = "1234";
  int i = 0;
  sscanf(data,"%d", &i);
  Serial.println(i);
}
void loop() { }
```
Any suggestions would be appreciated.

chadouming, 2015-06-30:
You need os_scanf and not os_sprintf

nicjohnston, 2015-06-30:
Thanks for the reply, however I just tried os_scanf, and it threw the following error
```
error: 'os_scanf' was not declared in this scope
```

chadouming, 2015-06-30:
what are you trying to do ?
If you want to scanf is intended to read an input from an interaction with the user. If you want to take the content of "i" and put it in the char buffer "data", then you would do : 

void setup() {
  Serial.begin(115200);
  Serial.println("starting");
  char *data = "";
  int i = 0;
  sprintf(data,"%d", i);
  Serial.println(i);
}

Serial output should be : 

Starting
0

Also, don't forget that sprintf on arduino doesnt support float or double. Just in case you wanted to use these types.

nicjohnston, 2015-06-30:
I am attempting to extract an integer value from a char buffer into a variable; I probably should have specified this sooner.
This issue surfaced while attempting to compile the ArduinoIMU example sketch from [this](https://github.com/RTIMULib/RTIMULib-Arduino) library.
Does this esp8266 library have support for sscanf?  If not, could you point me in the right direction to add it myself?
Thanks.

chadouming, 2015-06-30:
Ok, so this might be a dirty hack, but can you try : 

char *buff2 = "24";
  int i = ((String)(buff2)).toInt();

Worked fine for me even tho it made a warning.

Links2004, 2015-06-30:
String is a class not a type.
warning free:
```cpp
char buff2[] = "24";
int i = String(buff2).toInt();
Serial.println(i);
```

nicjohnston, 2015-06-30:
Thanks for reminding me about toInt, however the toInt function doesn't support parsing multiple variables from a char array.
This should have been in my previous post, but a more representative string is "1500,2100,1200,0".  The example code used a single variable string to make debugging easier.
I could loop through the array and find every delimiting character in order to split it up into separate arrays and then convert those to integers using the toInt function, however I use sscanf to parse data in many of the sketches I would like to port to the ESP8266.  Because of this, either a drop in replacement for sscanf or a patch for sscanf would be preferable.

a-andreyev, 2015-08-12:
completely agree with the @nicjohnston, we should implement `siscanf(...)`

Toshik, 2015-10-23:
Any news about sscanf?

torntrousers, 2015-10-25:
I'm looking to parse an HTTP Date header - "Thu, 15 Oct 2015 08:57:03 GMT" - into day, hours, minutes etc - I'd like sscanf too.

lighthousebulb, 2015-12-11:
any ideas for a workaround yet?

a-andreyev, 2015-12-13:
@lighthousebulb, I guess current ugly workaround is to use Arduino String.

Toshik, 2015-12-13:
@a-andreyev How do you suppose to use String instead of sscanf?

andig, 2016-01-13:
+1 for `sscanf` as its available on "standard" Arduino.

nouser2013, 2016-02-01:
Same issue. I'd need `sscanf`, too.

gerardwr, 2016-02-10:
+1.Same issue. I'd need sscanf, too.

Duality4Y, 2016-02-10:
while this triggers wdt resets:
```cpp
void setup() {
  Serial.begin(115200);
  Serial.println("starting");
  char *data = "1234";
  int i = 0;
  sprintf(data,"%d", &i);
  Serial.println(i);
}
void loop() { }
```

this doesn't:
```cpp
void setup()
{
  Serial.begin(115200);
  Serial.println("starting");
  char data[20];
  int i = 0;
  sprintf(data, "%d", &i);
  Serial.println(i);
}

void loop()
{

}
```

Links2004, 2016-02-10:
@Duality4Y "1234" is ```const char``` and can not be modified.
note: your issues has noting to do with ```sscanf```

Duality4Y, 2016-02-10:
i am just saying that sprintf works, but the way nicjohnston was using it (probably) was with a const char * and that doesn't work.

I assumed he just replaced his sscanf with sprintf.
(he mentions this in the first post)

Links2004, 2016-02-10:
```sscanf``` and ```sprintf``` have totally different functions using as a replaced is not possible.
the SDK not provide a ```sscanf``` Implementation, and its not a small functionality.
if some one find code that do ```sscanf``` under LGPL licence (or compatible) we can adapted it to the ESP.

Duality4Y, 2016-02-10:
I see my mistake :) sorry for the confusion!

if I understand it correctly he wants to convert a series of comma seperated values to series of ints.
you could probably do that with atoi and strtok_r.
little bit advanced though :)

but yea sscanf would be nice to have!

Duality4Y, 2016-02-10:
ok here is a thing i copy pasted together (me being way lazy), this works on you case this is not tested at all!
it came from here: http://mirror.fsf.org/pmon2000/2.x/src/lib/libc/
and it is probably tested there.
I can't garantee that it doesn't have bugs. :)

http://pastebin.com/N4d029VQ

igrr, 2016-03-11:
Implemented in #1752

torntrousers, 2016-03-11:
Woohoo, thank you!

igrr, 2016-06-23:
Merged and available in git version.

skorokithakis, 2016-08-14:
I would really like to use this, how can I try it out? I checked out the latest `feature/libc` head to PlatformIO's directory but I just got "undefined reference to sscanf"...

electronicsguy, 2017-04-06:
@igrr  Has sscanf() been implemented in the latest build? I still get `undefined reference to sscanf` error.

jjssoftware, 2017-04-06:
please see https://github.com/esp8266/Arduino/issues/3120#issuecomment-292093594

electronicsguy, 2017-04-06:
@Duality4Y   Thanks Robert! This helped in the meantime. It doesn't work as is (compilation errors due to some strange blank characters in that file). I've cleaned it up at put it here for anyone who needs it: [sscanf.h](https://pastebin.com/W2PsGR7J) & [sscanf.cpp](https://pastebin.com/KCjn1D87). Cheers :+1:

closed 2018-01-02 by igrr

AnishDey27, 2021-07-19:
With arduino IDE while using NODEMCU variables are defined as 32 bit ( only "int" ) may be anyone defined it as int16_ or anything else. I used 32 bit and that solved my problem with sscanf