esp8266/Arduino #490: atan2f causes reset/reboot

KillingJacky, 2015-06-30:
The following code:

```c++
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println("\r\n");
}

void loop() {
  // put your main code here, to run repeatedly:
  float f = -459.00;
  float t = -38.00;
  int i = digitalRead(13);
  t += i * 10.0;  //if comment out this line, it works!
  Serial.println(atan2f(t, f));

  delay(1000);
}
```

The weird thing is, if comment out t+=i*10.0 , it works. Actually, the code above is simplified from my project. In my project, I propagated the operation variable of atan2f from the reading of I2C device.
So the sign is, if the variable for atan2f is polluted by hardware output, atan2f will cause reset.

Another info, if call ets_memcpy/os_memcpy before atan2f, will also make atan2f call cause reboot.

Why? Another bug of cross-toolchain?

KillingJacky, 2015-06-30:
I tried put libm into .text section rather than .irom0.text, no good.
Also I tried using -O1 rather than -Os, no good either.

igrr, 2015-06-30:
see #233, #354
The fix has two parts:
- https://github.com/esp8266/Arduino/commit/81c9edb18ed46068fbdf31f34ef6e072a0de64e9
- on Windows, libm.a was not built with math functions enabled. I have uploaded an updated toolchain but haven't changed the .json files yet. Use libm.a from linux or mac toolchain if you happen to be on windows.

KillingJacky, 2015-07-01:
Fixed.
Thank you.

closed 2015-07-01 by KillingJacky