esp8266/Arduino #491: printf() doesnt produce any output

pastcompute, 2015-06-30:
Using a very simple program `Serial.begin(); Serial.println("Hello World");` works, but `Serial.begin(); printf("Hello, world\n");` produces no output.

I had a look at the code, printf() ends up calling `ets_vprintf` so I assume that that call  is not working either.

I haven't done anything unusual yet, just an ESP201 module.

Is there some (undocumented) call to enable mixing of ets_printf with Serial? I know when using the SDK there is some trickery involved in making a workable printf function on the uart in various cases...

igrr, 2015-06-30:
`Serial.setDebugOutput(true);` should enable ets_printf output through Serial.

pastcompute, 2015-06-30:
Also, the SDK has a function `system_set_os_print()` that would turn on output of Espressif messages from the Wifi layer, etc., is there a way of accessing that?

igrr, 2015-06-30:
```
extern "C" void system_set_os_print(int);

void setup() {
  system_set_os_print(...);
}
```

pastcompute, 2015-06-30:
No worries. thanks that worked. I add a pull request to mention `setDebugOutput` in the readme

closed 2015-06-30 by pastcompute