esp8266/Arduino #494: SoftAP only mode does not work

halherta, 2015-06-30:
SoftAP only mode does not work for me. It seems to only work if the ESP8266 is connected  in station mode from a previously flashed sketch.

For example, softAP will work if in a previously flashed sketch the setup function looks like this:

void setup(void){
  Serial.begin(115200);
  WiFi.begin(stassid,stapassword);
  Serial.println("");
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

then in the current flashed sketch:

void setup(void){
  Serial.begin(115200);
  WiFi.softAP(apssid,appassword);
  Serial.println("");
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

where "stassid"/"stapassword" are the SSID and password of my local router AP that the ESP8266 accesses in station mode and the "apssid"/"appassword" are the ESP8266's SSID and password in it's acting as an access point itself.

if I force the ESP8266 to be in AP only mode with "WiFi.mode(WIFI_AP)",  the sketch fails i.e. I'm unable to properly connect to to the ESP8266's AP and when my laptop seemingly connects to the AP the rest of the sketch fails.  I'm using ESP-12/ESP-12E devices on NodeMCU DevKit 0 and DevKit 1 boards.

Links2004, 2015-06-30:
may calling
```cpp
WiFi.disconnect();
```
before ```WiFi.softAP``` helps.

halherta, 2015-06-30:
Tried using WiFi.disconnect(); it didn't help.
Here's the complete sketch..It's a simple Web based robot controller
```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
 
#define mot1PinA 9
#define mot1PinB 10
#define mot2PinA 3
#define mot2PinB  5

const char* stassid = "routerAP";
const char* stapassword = "0123456789";
const char* apssid = "Robot";
const char* appassword = "0123456789";
ESP8266WebServer server(80);

void handleRoot() {
  server.send ( 200, "text/html","<html>\
<head>\
<title>Robot Control</title>\
<style>\
          html, body {width:85%;margin:0 auto;padding:0;background-color:#cfcbd1;text-align:center;}\
          .f{display:block;}\
	  .rev{display:block;}\
          .l{display:inline;}\
          .r{display:inline;}\
          .s{display:inline;}\
          .fb{vertical-align:middle;width:150px;height:120px;margin:25px;background-color:#45cc5a;font-size:20px}\
	  .revb{vertical-align:middle;width:150px;height:120px;margin:25px;background-color:#cc45b7;font-size:20px}\
          .lb{vertical-align:middle;width:150px;height:120px;margin:25px;background-color:#4574cc;font-size:20px}\
          .rb{vertical-align:middle;width:150px;height:120px;margin:25px;background-color:#cc9d45;font-size:20px}\
          .sb{vertical-align:middle;width:150px;height:120px;margin:25px;font-size:20px}\
</style>\
</head>\
<body>\
    <h1>Carl Robot Control</h1><br><br><br>\
    <form class=\"f\" action=\"Forward\"> <input type=\"submit\" class=\"fb\" value=\"Forward\">  </form> \
    <form class=\"l\" action=\"Left\"> <input type=\"submit\" class=\"lb\" value=\"Left\"> </form>\
    <form class=\"s\" action=\"Stop\"> <input type=\"submit\" class=\"sb\" value=\"Stop\"> </form> \
    <form class=\"r\" action=\"Right\"><input type=\"submit\" class=\"rb\" value=\"Right\"> </form>\
    <form class=\"rev\"action=\"Reverse\"><input type=\"submit\" class=\"revb\" value=\"Reverse\"></form>\
</body>\
</html>");
}

void handleNotFound(){
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void goForward(){ 
    Serial.println("Go Forward");
    digitalWrite(mot1PinA, HIGH);
    digitalWrite(mot1PinB, LOW);
    
    digitalWrite(mot2PinA, LOW);
    digitalWrite(mot2PinB, HIGH);
    handleRoot();
}
  
void goReverse(){ 
    Serial.println("Go Reverse");
    digitalWrite(mot1PinA, LOW);
    digitalWrite(mot1PinB, HIGH);
    
    digitalWrite(mot2PinA, HIGH);
    digitalWrite(mot2PinB, LOW);
    handleRoot();
}
  
void goLeft(){ 
    Serial.println("Go Left");
    digitalWrite(mot1PinA, HIGH);
    digitalWrite(mot1PinB, LOW);
    
    digitalWrite(mot2PinA, HIGH);
    digitalWrite(mot2PinB, LOW);
    handleRoot();
}
  
void goRight(){ 
    Serial.println("Go Right");
    digitalWrite(mot1PinA, LOW);
    digitalWrite(mot1PinB, HIGH);
    
    digitalWrite(mot2PinA, LOW);
    digitalWrite(mot2PinB, HIGH);
    handleRoot();
}
  
void goStop(){ 
    Serial.println("Go Stop");
    digitalWrite(mot1PinA, LOW);
    digitalWrite(mot1PinB, LOW);
    
    digitalWrite(mot2PinA, LOW);
    digitalWrite(mot2PinB, LOW);
    handleRoot();
}

void setup(void){
  delay(1000);
  WiFi.disconnect();
  delay(100);
  Serial.begin(115200);
  WiFi.softAP(apssid,appassword);
  Serial.println("");
  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  
  Serial.println("done");
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
  
  server.on("/", handleRoot);
  server.on("/Forward", goForward);
  server.on("/Reverse", goReverse); 
  server.on("/Left", goLeft);
  server.on("/Right", goRight); 
  server.on("/Stop", goStop);
  server.onNotFound(handleNotFound);
  server.begin();
  Serial.println("HTTP server started");
}
 
void loop(void){
  server.handleClient();
} 

```

igrr, 2015-06-30:
No need to wait for connection loop in softAP mode.

halherta, 2015-06-30:
thanks Ivan, that seemed to have resolved the problem. Is this documented anywhere? if not it might be a good idea to do so.

I based my code on the  "WiFiAccessPoint" example that came with the Arduino ESP8266 library. It also has the wait for connection loop in softAP mode. and it also doesn't work properly. It might be a good idea to fix that.

Thanks again

closed 2015-07-01 by halherta

themindfactory, 2015-07-01:
This code looks familiar :-)

irun4fundotca, 2016-02-06:
I have a loop happening when trying to connect to a new router in a different location
is there a retry string I can add to allow it to stop trying to connect after like 3 tries, so I can send the new credentials without re-uploading the whole sketch again?