esp8266/Arduino #497: Error during flashing - FLASH_DOWNLOAD_DATA

Attil, 2015-07-01:
Hello,
When I'm trying to upload my sketch to arduino, everything works seems to go smoothly, however at the end of the uploading, there is usually an error.

```
Sketch uses 241,952 bytes (55%) of program storage space. Maximum is 434,160 bytes.
Global variables use 46,880 bytes (57%) of dynamic memory, leaving 35,040 bytes for local variables. Maximum is 81,920 bytes.
Uploading 246096 bytes from /tmp/build1487447564622711308.tmp/ESPdriver.cpp.bin to flash at 0x00000000
............................................................................................................................................................................................................................................warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response
```

After this happens, i cannot even attempt to upload sketches, because i immediately get the following error

```
warning: espcomm_sync failed
error: espcomm_open failed
```

Then i have to unplug USB-to-serial from my PC, plug it again and try uploading, which usually results in getting error during FLASH_DOWNLOAD_DATA again.

I should add that sometimes, in about 1 per 20 or so tries i can flash, also monitoring Serial, both sending and receiving data works fine. Every time this error happens, it appears at about the same moment (judging by the amount of dots).

I have tried both Arduino IDE with installed ESP8266 from the Board Manager and compiled one from the git by myself, same error.

I am using ESP8266-01 and MicroFTX to communicate with PC (http://jim.sh/ftx/). ESP8266 is powered by YwRobot. My OS is Ubuntu 14.04.

Because my MicroFTX doesn't support DTR I'm manually resetting and setting GPIO0 low.

Any suggestions on how could I fix this problem?

Thanks

Any sugge

Attil, 2015-07-01:
I have semi-resolved this issue by editing esptool.py to be
```
    ESP_RAM_BLOCK   = 0x180
    #ESP_RAM_BLOCK   = 0x1800
    ESP_FLASH_BLOCK = 0x40
    #ESP_FLASH_BLOCK = 0x400

```
Instead of the commented lines. Granted, the upload is a lot slower now, however it always finishes without any errors. Any insight about why doesn't the default IDE uploader work for me?

Credit: http://www.esp8266.com/viewtopic.php?f=6&t=2791#p16082

cimba007, 2015-07-06:
General question? What upload speed are you using? I had mine set to ~512kbps but hat some errors now and then and set it down to 256kbps .. just asking as you never said a word about it.

Attil, 2015-07-06:
I tried all of them, mostly 115200, however none worked.
Interestingly, no matter which speed i chose, the error always occured at the same % of upload.

drmpf, 2015-07-06:
I also get this occasionally.  When I do I try again, if that does not work I exit the Arduino IDE completely and restart it.  This forces a full compile, not sure if that is significant, but then routinely my upload works.

cimba007, 2015-07-08:
All my problems were gone as soon as I set upload speed lower. I am using the esp03 models from which I own like 5 I think. I guess it would be best to try an other unit if you have one at hand or at least order a backup unit for comparising/debugging reasons. Otherwise I dont have too much insight in the uploading process, can only hope one of da real pros read this thread ;-)

kommancs96, 2015-10-15:
Same issue here. I can't use my esp8266-01  with Arduino IDE (1.6.5.)

drmpf, 2015-10-15:
I am using 1.6.5 with no problems

kommancs96, 2015-10-17:
I tried to modify esptool.py like in the comments above, but I couldn't do that, because in my esptool.py is  only the following:
```python
#!/usr/bin/python
# EASY-INSTALL-SCRIPT: 'esptool==0.1.0','esptool.py'
__requires__ = 'esptool==0.1.0'
import pkg_resources
pkg_resources.run_script('esptool==0.1.0', 'esptool.py')
```
I found this file in `/usr/local/bin` (Mac OS X) should I search anywhere else?

dpechman, 2016-02-26:
I fix this issue increasing the current of the 3V3 power supply. the esp8266 requires more current while in programming.

adamelliot, 2016-03-22:
I sometimes get this as well, and it seems I just need to reboot to fix it. Not really sure what that's about.

devyte, 2017-10-20:
Closing per #3655 .

sp6cpn, 2019-01-30:
I succeeded after setting Serial Monitor baud rate to 115200 (default was 9600)