esp8266/Arduino #499: WiFiClient.println ignore string length

av1024, 2015-07-01:
Looks like a bug in string length calculation at println code in WiFiClient. The same code:
```C
#define OFF "OFF"
void handler(Print &prn) {
  prn.println(OFF);
  //prn.println(String(OFF)); // same result
}
```
produces different result on Serial and WiFiClient:

 * Serial: `OFF<CRLF>`
 * WiFiCient (as telnet server): `OFFw<CRLF>`

Different strings produces different results. As an example:
```C
//...
 const char *_mode[] = {
  "on", "off", "direct", //...
 };
 prn.print("Mode: ");
 prn.println(_mode[i]); // i==2
```
result:
```
Mode: directect
```

igrr, 2015-07-01:
That's weird. ```println``` is implemented in Print, which is a base class of WiFiClient (via Stream and Client), so that's one implementation for both WiFiClient and Serial.
Can you add some debug info into WiFiClient::write? Like
```
ets_printf("write size=%d\r\n", size);
```
and also add Serial.setDebugOutput(true) into the sketch.
This is to verify that WiFiClient is passed correct string size from Print::print(const char*) function.

av1024, 2015-07-01:
Hmm... The passed lengths are ok, but result is not.

As an idea: issue with sending a lot of small packets or in telnet handler?? Looks like the rest of invoked command printed.

Log output:
```
write size=1
write size=1
write size=1
write size=1
write size=1
write size=1
write size=1
write size=1
write size=1
#648769 cmds_tool.h/on_led: print 'LED '
write size=4
#648777 cmds_tool.h/on_led: print 'ON'
write size=2
#648783 cmds_tool.h/on_led: println
write size=2
#648787 cmds_tool.h/on_led: print 'Mode: '
write size=6
#648801 cmds_tool.h/on_led: print 'Direct'
write size=6
#648807 cmds_tool.h/on_led: println
write size=2
```

corresponding telnet output:
```
led show<Enter>
LED OFFw
Mode: WifiOff

```

Another check:
```
#425656 cmds_tool.h/on_led: print 'LED '
write size=4
#425664 cmds_tool.h/on_led: print 'ON'
write size=2
#425670 cmds_tool.h/on_led: println
write size=2
#425674 cmds_tool.h/on_led: print 'Mode: '
write size=6
#425680 cmds_tool.h/on_led: print 'Direct'
write size=6
#425687 cmds_tool.h/on_led: println
```
Output:
```
led show<Enter>
LED ONow
Mode: Direct

```

av1024, 2015-07-01:
Found error.
It is my telnet handler issue (should invoke println() because sent command printed back somewhere)

closed 2015-07-01 by av1024