esp8266/Arduino #500: Static variable

hackscribble, 2015-07-01:
Getting the following errors with the minimal test program below.  Can work around by making variable global, but it would be good to fix the problem with static.

```
C:\Users\Ray\AppData\Local\Temp\build2036106428985831191.tmp\sketch_jul01b.cpp.o: In function `setup':
sketch_jul01b.cpp:(.text+0xc): undefined reference to `__cxa_guard_acquire'
sketch_jul01b.cpp:(.text+0x10): undefined reference to `__cxa_guard_release'
C:\Users\Ray\AppData\Local\Temp\build2036106428985831191.tmp\sketch_jul01b.cpp.o: In function `loop':
sketch_jul01b.cpp:(.text+0x25): undefined reference to `__cxa_guard_acquire'
sketch_jul01b.cpp:(.text+0x37): undefined reference to `__cxa_guard_release'
collect2.exe: error: ld returned 1 exit status
Error compiling.
```

```
void setup() 
{
}

void loop() 
{
	static uint32_t lastMillis = millis();
}
```

reaper7, 2015-07-01:
declaring a static variable in the loop? again and again and ....

I think the proper method is:
```
static uint32_t lastMillis;

void setup() 
{
}

void loop() 
{
    lastMillis = millis();
}
```

hackscribble, 2015-07-01:
Static means it is initialised the first time loop() is executed and it retains its value when loop() is called again.

The advantage is that the variable has local scope in loop() rather than being global.

Code below compiles OK for Arduino Uno etc but not for ESP8266. Maybe it's not supported on ESP8266, in which case note should be added to documentation page.

See www.learncpp.com/cpp-tutorial/43-static-duration-variables/

reaper7, 2015-07-01:
probably You are right :)

Links2004, 2015-07-01:
@hackscribble  what version and OS you use?
I use windows with the latest git and I can use static in the loop or in any other function it is working fine.

hackscribble, 2015-07-01:
Arduino IDE 1.6.5 on Windows 8.

esp8266 version 1.6.4-673-g8cd3697 installed through Boards manager.

Board settings: Generic ESP8266 module, 80MHz, 512K, 115200.

Links2004, 2015-07-01:
ok can reproduce it, only using a function is not working.

no error with:
```cpp
void setup() 
{
}

void loop() 
{
   static uint32_t lastMillis = 0;
   lastMillis = millis();
}
```
strange.

hackscribble, 2015-07-01:
Thank you @Links2004 - that's a good workaround.

Makuna, 2015-07-02:
Actually, the original code is flawed in that when the static is initialed you have no  idea if millis has been initialized, so it should be avoided.  You should use Links2004 code as the correct way to do this.

hackscribble, 2015-07-02:
Thanks, @Makuna. I had not thought about that before, but I see exactly what you mean.  As @Links2004 pointed out, his code is not exactly the same functionality.  So I think I will change to a global variable, set to millis() at the end of setup().

igrr, 2015-07-06:
Actually the original code is fine — C++ standard guarantees that local static initializer will be called exactly once, first time the function is executed. We just need to add __cxa_guard* implementations.

Makuna, 2015-07-06:
@igrr when does the timer/variables get initialized for millis?  Can you guarantee this happens before this static variable is initialised?  This is my point.

igrr, 2015-07-06:
In this case local static variable will be initialized the first time loop() function runs, that's what C++ standard guarantees. millis is certainly initialized before that, so that you can use it in setup()/loop().

tuxedo0801, 2016-01-30:
Is there a chance to get this fixed instead of implementing a workaround? Is there someone out there with the required expert knowledge?

tuxedo0801, 2016-02-05:
*push*

Duality4Y, 2016-02-05:
you don't have to set a value for static variables,
they will always be initialized to zero.

tuxedo0801, 2016-02-05:
The point is: The library (not written by me) I'm talking about works quite well with Atmel/AVR Arduinos. 

But using it with ESP8266 brings up those errors.

Rewriting the library so that it works with ESP8266 (as suggestes by @Links2004), would result in rewriting the whole library (yes, it makes massive use of those static stuff. I'm also not happy with it, but that's the current fact).

That's not the idea of a portable library and an "arduino-style" compatible ESP8266.

So instead of working around the problem, I would really recommend solving this issue at it's root.
Sadly, I'm not that good in C/C++ to do it. So I hope someone else can step up and fix it. I would even donate for this fix...

br,
Alex

Duality4Y, 2016-02-05:
I see your point and I agree.
as iggr said:
> We just need to add __cxa_guard* implementations.

I myself have walked against this error. and solved it by making that variable (which is an object in my case) global to that function.

I was just pointing out though the correctness and convention of such static variable use :)

igrr, 2016-02-05:
https://github.com/esp8266/Arduino/pull/1586 should resolve this.

@tuxedo0801 could you please try this pull request with your library and see if it works? 
Thanks.

tuxedo0801, 2016-02-06:
Thanks a lot. Will test it, but takes until start of next week.. Will report back about the result.

closed 2016-03-11 by igrr

kenji21, 2016-06-22:
Got the same issue with a Feather 32u4 from adafruit in /Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/PluggableUSB.cpp :

````
  101 PluggableUSB_& PluggableUSB()
  102 {
  103         static PluggableUSB_ obj;
  104         return obj;
  105 }
````

Moving the static before the method worked (not fan because it is an Arduino.app provided file). But also adding the flag (like the arduino IDE does) :  -fno-threadsafe-statics make compilation succeed.

vdiallonort, 2016-07-28:
Hi,I am looking for the work-arround you guys find as i am hitting the same issue.Does there is a fork of the library somewhere or little information about how to deal with this issue ?

igrr, 2016-07-28:
This is fixed since 2.2.0.
If you want more info about the fix, feel free to scroll the page up and click on the link which says "#1586 should resolve this" :) There you will find corresponding pull request including all the changes.

vdiallonort, 2016-07-28:
Thanks,just to make sure i got it straight version 2.2 is for the ESP8266 arduino core ? So that mean i got the wrong version somehow and need to upgrade

igrr, 2016-07-28:
The latest version right now is 2.3.0. If you see this happening in 2.3.0, please open a new issue and post detailed steps to reproduce (example sketch, etc.). Thanks.