esp8266/Arduino #503: ESP.deepsleep() causes immediate reboot

SESupply, 2015-07-01:
I am having a problem with the deepsleep command - no matter what period I use for the sleep time, the module reboots immediately.  (I do have GPIO16 tied to RST)
Issueing:
ESP.deepSleep(25000000, RF_WAKE_DEFAULT);   //sleep for 25Secs
results in an immediate reboot.  Anyone come across this?  I have tried on 2 different modules & get the same result (both ESP-07's)
I also have a similar issue trying to use Serial.swap() - the rest of the code works perfectly.  Am I maybe just missing some update or something?

chadouming, 2015-07-01:
Works fine on my esp-07 here. Do you have the new or the old 07 ?

SESupply, 2015-07-03:
Not sure if mine are the "old" or "new" versions - how can I tell?  I do have GPIO16 & RST tied together to a pull-up.  
There also seems to be some confusion regarding the time parameter - in some places it listed as microseconds, but I've also seen it as milliseconds...

chadouming, 2015-07-03:
For deepsleep it's micro seconds. If you have rx and tx together in one
corner, it's the new one with same pinout than esp-12.

SESupply, 2015-07-05:
They are the new ones then - I've updated & now it seems to be working...

closed 2015-09-30 by igrr